dotnet/iot #1737: Oled display, I2C, SH1106, 128x64, 1.3 inch, V2.0, blue for Orange Pi Lite

HyVong007, 2021-11-27:
I want to use oled display with orange pi lite by using Dotnet IOT, as the title said. Details here:
1- Oled display: https://shopee.vn/M%C3%A0n-H%C3%ACnh-Oled-1.3-Inch-Giao-Ti%E1%BA%BFp-I2C-(blue)-i.60387211.1341644922. It uses I2C interface and driver SH1106.

2- Orange Pi Lite: http://www.orangepi.org/orangepilite/

3- Armbian 21.08.6 Focal with Linux 5.10.60-sunxi: https://www.armbian.com/orange-pi-lite/ . I have enabled i2c0 in Hardware  from armbian-config, and check with "i2cdetect" and installed latest .NET Core runtime. See attached images. 

4- I use "Wiring OP" to show the GPIO information table. See attached images. https://github.com/orangepi-xunlong/wiringOP

[image]

[image]

[image]

Thank you so much !

raffaeler, 2021-11-27:
Sorry, I don't understand what is your request.
Could you please tell us exactly what do you need or does not work in your case?

HyVong007, 2021-11-27:
> 

I cannot use Oled SH1106 ! Cannot use at all ! IoT.Devices namespace doesnot have SH1106 or anything about "Oled 128x64". There are only Lcd1602 and Lcd2004, Ssd13xx

raffaeler, 2021-11-27:
If I understand well, you are able to use the dotnet/iot libraries but you didn't find any class that support the Oled SH1106.

This is quite normal. There are thousands of different devices available on the market and of course there is no library in any language that can support everything.

One of the goals of this repository is to providing a solid building block to the community so that anybody can contribute adding support for missing devices.
Most of the bindings (aka drivers) for the currently supported devices comes from community contributions, as well as bug fixing, examples, tests and docs. This is how open source works.

If you are interested to contribute to this repository, I would suggest to:
- download the datasheet of that display
- look into the source code of the current display and see if any current device is similar to the SH1106. If there are similarities, it could work as-is or it may require some changes. If you don't find anything, a similar device should at least give an idea how the device source code should be written.
- If you want to submit changes or entirely new code, you can "fork" this repository, work on the code and submit a "PR" (Pull Request) to this repository so that your code can be merged to the main libraries.

HTH

HyVong007, 2021-11-28:
> If I understand well, you are able to use the dotnet/iot libraries but you didn't find any class that support the Oled SH1106.
>
> This is quite normal. There are thousands of different devices [...]

This is a FEATURE REQUEST, label api-suggestion.

Because i don't see device "Oled SH1106" so i create new feature request. Now I doesnot have enough knowledges and skills, so i am waiting for community.

You can close this issues now or just remain it here and let everyone see. That's the way OPEN SOURCE PROJECT goes. Thank you so much !

raffaeler, 2021-11-28:
Sure, in fact I wrote:
>> If you are interested to contribute to this repository,

The problem with "waiting for someone else" is that he/she must also buy the device. This could take a while.

> You can close this issues now or just remain it here and let everyone see. That's the way OPEN SOURCE PROJECT goes. Thank you so much !

I am not here to close the issues but just a community member willing to help, that's it
You are welcome.

HyVong007, 2021-11-28:
> Sure, in fact I wrote:
>
>
> The problem with "waiting for someone else" is that he/she must also buy the device. This could take a while.
>
>
> I am not here to close the issues but just a community member [...]

According to you, i shouldnot notifiy everyone when i found some devices not supported/not available yet by Dotnet IOT ?  I think some one if buy this device will see this issue and comment here. Or I should keep quiet and hope forever for someone will add this device to Donet IOT ?

I think the better way is: Microsoft (Donet IoT) should create a topic about DEVICE REQUEST for IOT. > I will suggest Dotnet IOT to create that topic. Everyone will request devicw and everyone will add device support to Donet IOT. Thank you again, enthusiast member !

raffaeler, 2021-11-28:
> According to you, i shouldnot notifiy everyone when i found some devices not supported/not available yet by Dotnet IOT ? I think some one if buy this device will see this issue and comment here. Or I should keep quiet and hope forever for someone will add this device to Donet IOT ?

As I wrote before there are thousands devices and more to come every year. 
I believe it is unlikely that anyone buy and implement the code to contribute. It is more likely that someone who cares about that device implements the binding and publicly contributes to the project so that others can benefit from it.

Even in the Arduino community there are many unsupported devices (with no library at all). It is normal.

> I think the better way is: Microsoft (Donet IoT) should create a topic about DEVICE REQUEST for IOT. > I will suggest Dotnet IOT to create that topic. Everyone will request devicw and everyone will add device support to Donet IOT. Thank you again, enthusiast member !

There are two libraries in this repository:
- the GPIO library is the low-level one to provide support for the boards such as RPi / OrangePi and many others. 
- the binding library uses the GPIO library to provide the easiest possible way to add support to new devices that can be connected to the boards.

The primary focus of Microsoft is to create a robust support for the GPIO library and to make it easy for anyone else to create a binding. In fact there are bindings that are just a single class.

This repository is not a "wish list" for bindings. You can certainly ask for a binding, but it's just a request to the community, not to Microsoft. (I am not from Microsoft and maybe someone from the team can be more specific).

HTH

HyVong007, 2021-11-28:
> As I wrote before there are thousands devices and more to come every year. I believe it is unlikely that anyone buy and implement the code to contribute. It is more likely that someone who cares [...]

I think the Oled display with resolution 128x64 and driver SH1106 or driver SSD13xx  are VERY POPULAR because i visit many online shopping and i always see them, both SH1106 and SSD13xx.

I don't command anyone but only send a request to the community. Sorry if i make someone misunderstood because of my bad english.

raffaeler, 2021-11-28:
No problem at all @HyVong007.
If this display is becoming more and more popular, it will probably raise the possibility that someone will contribute with an implementation.
Let's see if someone with the device in his/her hands is willing to do it.

HyVong007, 2021-11-28:
> As I wrote before there are thousands devices and more to come every year. I believe it is unlikely that anyone buy and implement the code to contribute. It is more likely that someone who cares [...]

Ok not to waste your time or anyone's time here, I very love C# and i want to use C# .Net core to develope IOT, but currently Dotnet IOT has no support for "Oled SH1106 display".

Now in case no one can know, I think i must do the job by my self. I will research goooo...ooogle to create bindings for Oled SH1106 from low-level code with enough luckiness.

Sorry so much for wasting anyone's time here. I will think more carefully when creating Feature Request in github in the future because here is Open Source Community Project. I will close this issue after some hours to let more people can read the issue.

raffaeler, 2021-11-28:
If you want to try writing a binding, you can start from an existent one. For example:
https://github.com/dotnet/iot/tree/main/src/devices/Ssd13xx
But maybe a different device is a better starting point. You should investigate with the SH1106 datasheet and compare the features.

Good luck :)

HyVong007, 2021-11-28:
Thank you again, enthusiast member !
Nothing easy and free with no price or risk !

A-J-Bauer, 2022-01-17:
A while back I wrote a simple class for this [here](https://github.com/A-J-Bauer/OledI2c128x64SH1106).
(I2C, System.Drawing.Common and .Net Core)
Once I get my hands on a spare SSH1106 I will might rewrite this for .NET 6.0 and SkiaSharp.

HyVong007, 2022-02-01:
Thank you for your nice library for Oled display. I am very interested !

krwq, 2022-08-25:
[Triage] It would be nice to have this device but we're trying to avoid to keep issues open with device wish-list. If anyone is interested in contributing we can re-open this.

closed 2022-08-25 by krwq

ChilloDE, 2022-09-21:
So I guess I'm the developer OP is waiting for.

It seems like the protokoll of the [Ssd1306.cs](https://github.com/dotnet/iot/blob/fa1960cd38a1548abb6ba99b2a0842dde9ec1079/src/devices/Ssd13xx/Ssd1306.cs) is compatible to the SH1106.
I've got a display from az-delivery with an SH1106 oled driver and they are providing what seems like a copy of this [Adafruit_Python_SSD1306](https://github.com/adafruit/Adafruit_Python_SSD1306) library.

As I'm more confident using C# I was already using the .NET IoT library to connect my Raspberry and my Trinket via I2C.
Now I've added the display, so if the Ssd1306.cs dosn't work out for me I'll be extending on it to make it work.

I did not try this yet, I propably get to it later today. But I've compared a few commands, so far the byte-values all do seem to match.