dotnet/iot #1741: Contribution document proposal

raffaeler, 2021-12-03:
This is not a real draft of the document, but just a few phrases that may be helpful to write down the document discussed during the last triage about the contribution guidelines in this repository.
/cc @joperezr 

This repository [dotnet/iot](https://github.com/dotnet/iot) includes two category of projects that are quite different:
- `System.Device.Gpio` is the main library which includes the "building blocks" to support the boards and the base peripherals that are needed to interact with the physical world. 
- `devices` projects is a folder including all the bindings (AKA drivers) to support the physical sensors.

The `System.Device.Gpio` project is subject to API reviews and design guidelines that are similar to many other projects of the dotnet runtime and library family.
All the [devices](https://github.com/dotnet/iot/tree/main/src/devices) projects are different for at least these important aspects:
- They are not subject to API reviews
- They do require having a physical device in order to be implemented
- They are still reviewed by the repository triage team, but the evaluation is done only through the code review process.

While the community is always welcome to contribute to any of the [dotnet](https://github.com/dotnet) repositories, all the [devices](https://github.com/dotnet/iot/tree/main/src/devices) projects are expected to mostly be contributed outside the main teams, from any developer willing to provide her or his expertise with a specific device.
These projects are usually small in size, may provide a partial or full functionality for the given device, but they are still required to produce a minimal documentation and the tests ensuring the minimum quality bar.

krwq, 2022-04-28:
[Triage] @joperezr ping 😄

joperezr, 2022-05-02:
This is still on my radar but unfortunately haven't had time to work on this. I also want to work on https://github.com/dotnet/iot/issues/1836 which is documentation related, but I will probably get more time to do both later in . NET 7 release cycle.