dotnet/iot #1746: Pull-down not working for GPIOs for Pins over 26 on RPi4

raffaeler, 2021-12-10:
**Describe the bug**
Setting the pull-down on the RPi4 on GPIOs higher than 26 does not work on the RPi4.
I noticed the issue while working with the Button device which randomly triggers a low condition even if it is configured with a pull-down (the other end of the button is tied to `+Vcc`).

**Steps to reproduce**
- Configure the pin as input with pull-down
- Physically connect a long wire on the pin, but do not connect the wire to anything
- Wave the hand over the wire or move the wire close to other devices
**Expected behavior**
The pull-down should guarantee stable readings

**Actual behavior**
If the previous steps are done while looping  and printing the values, you should be able to see zeros and ones in the output

**Versions used** Pre-release 2.0.0,21754

This is probably not an issue with the managed code. This bug has been observed in the native libraries on the RPi4:
- https://forums.raspberrypi.com/viewtopic.php?t=244375
- https://sourceforge.net/p/raspberry-gpio-python/tickets/168/#e2fa
- https://github.com/RPi-Distro/raspi-gpio/commit/80fa7d04eafb3ea34fc6f2d32de5f1873b5fb369#diff-02e8509be66e4ea16bb88f157ad4e4c4764871247883c0fa22d3751499b088c7R474

The last link points to the raspi-gpio tool which has been changed to support the RPi4.
My question is whether those changes has been ported here or not.

pgrawehr, 2021-12-11:
Yes, those changes have been ported over long ago, and I'm sure that pull-ups and pull-downs work on the Pi4. There's even an integration test at `RaspberryPiDriverTests.InputPullResistorsWork` that tests this (our lab has only Pi3's though). However, I have never tested it with large pin numbers. But why should there be a difference?

raffaeler, 2021-12-11:
@pgrawehr I am seeing this issue on a project and seeing sudden changes of the pin button (Button device) right after booting the application.
The Button setup is straightforward: one pin to the RPi and the other one straight to Vcc.

I then made a search to see if there were known problems on the RPi with the pull-down and found those links. That's all I know for the moment. After xmas I will have the time to make some more tests.

You are more familiar than me with the infrastructural code. So, if you know that those changes were already applied, I will try to find other reasons.

krwq, 2022-10-13:
[Triage] can you re-test and make sure you're getting the right GPIO driver?

pgrawehr, 2022-10-27:
This looks like a problem in libgpiod to me. We should find out what error 16 means.

raffaeler, 2023-03-23:
I confirm that the error 16 is the issue that @HakanL observed.
Beyond that, the pull-ups work on with the `libgpiod-dev` library installed, but don't when using the `SysFs` one.
Not sure if this is a 'by design' behavior or not.

pgrawehr, 2023-03-23:
[Triage] To test this properly, don't use the `Button` binding, because that uses the underlying event driver (when the main driver is RaspberryPi) and this might result in confusing results. Better do a test like this (pseudo code):

```
foreach (var driver in {RaspberryPi3Driver, SysFsDriver, LibGpiodDriver})
{
    var pin = driver.OpenPin(PinNumber, PinMode.InputPullDown);
    while (some time)
    {
        var newValue = pin.Read();
        if (newValue == PinValue.High)
        {
            Console.WriteLine("The pin is high!");
        }
    }
}

Ellerbach, 2023-07-20:
[Triage] Closing per @raffaeler request as it's not directly related to the library and there are ways to test it.

closed 2023-07-20 by Ellerbach