dotnet/iot #1751: Syntax error in package targets file

pgrawehr, 2021-12-20:
The file `System.Device.Gpio.targets` in the nupkg's `buildtransitive` folder has a syntax error (missing apostrophe). This fails all builds depending on the package. 

```
<Project>
  <Target Name="EnsureProjectIsTargetingWindowsPlatformIfRequired"
          BeforeTargets="Build"
          Condition="'($(TargetFramework)' == 'net6.0' Or $(TargetFramework)' == 'net5.0') And
                                                       ^^^^^ here
                     '$(SuppressWindowsPlatformTargetingRequiredError)' == '' And
                     ('$(OutputType)' == 'Exe' Or '$(OutputType)'=='WinExe') And
                     $(RuntimeIdentifier.StartsWith('win'))">
    <Error Text="In order to use System.Device.Gpio package when running on Windows and targeting net5.0 or higher, it is required to add 'net6.0-windows10.0.xxxxx.0' (where xxxxx needs to be a number 17763 or above) to your project's target frameworks. To suppress this error, set the property %3CSuppressWindowsPlatformTargetingRequiredError%3Etrue%3C/SuppressWindowsPlatformTargetingRequiredError%3E in your project." />
  </Target>
</Project>
```

joperezr, 2021-12-20:
Great catch! want to submit a PR? if not I'm happy to take care of it.

pgrawehr, 2021-12-21:
I have the fix already prepared. I'll push it later today.

PR #1754 (2021-12-22): Fix buildTransitive/net5.0/System.Device.Gpio.targets
Fixes #1751 

There were a missplaced "(" and a missing apostrophe. 

###### Microsoft Reviewers: [Open in CodeFlow](https://portal.fabricbot.ms/api/codeflow?pullrequest=https://github.com/dotnet/iot/pull/1754)
closed 2021-12-22 by pgrawehr