dotnet/iot #1756: Reading Values from DHT22 with RaspPI Zero 2 W ->  0

FloriEnder, 2022-01-04:
Hi,
i tried to read Values from my DHT22 Sensor. But i only get the value "0" for Temperature and Humidity.
On my PI 4B works everything. I have four DHT22 Sensors and everyone Works when i read the values with Python3 on the same GPIO Pin. Other GPIO Pins dosn't work neither.

//C# Code
```
using Iot.Device.DHTxx; //Version -> 1.5.0
using System;

namespace TempHumidinCSharp
{
	class Program
	{
		static void Main()
		{
			using Dht22 Sensor1 = new(4);
                        //using Dht22 Sensor1 = new(17);
			while (true) {
				Console.WriteLine(Sensor1.Humidity.Value);
				Console.WriteLine(Sensor1.Temperature.Value);
			}
		}
	}
}
```
//C# Code End

//Pyhton Code
```
import time
import board
import adafruit_dht

# Initial the dht device, with data pin connected to:
# dhtDevice = adafruit_dht.DHT22(board.D4)

# you can pass DHT22 use_pulseio=False if you wouldn't like to use pulseio.
# This may be necessary on a Linux single board computer like the Raspberry Pi,
# but it will not work in CircuitPython.
dhtDevice = adafruit_dht.DHT22(board.D17, use_pulseio=False)

while True:
    try:
        # Print the values to the serial port
        temperature_c = dhtDevice.temperature
        temperature_f = temperature_c * (9 / 5) + 32
        humidity = dhtDevice.humidity
        print(
            "Temp: {:.1f} F / {:.1f} C    Humidity: {}% ".format(
                temperature_f, temperature_c, humidity
            )
        )

    except RuntimeError as error:
        # Errors happen fairly often, DHT's are hard to read, just keep going
        print(error.args[0])
        time.sleep(2.0)
        continue
    except Exception as error:
        dhtDevice.exit()
        raise error

    time.sleep(2.0)
```
//Pyhton Code End

Output on my Pi 4B:
pi@raspberrypi:~/Downloads/CSharp$ ./TempHumidinCSharp
99.9
26.6
99.9
26.6
99.9
26.6
99.9
26.6
99.9
26.6
99.9
26.6
....
99,9  Humidity because the Sensor on my Pi4 ist broken...

Output on my Pi zero 2 W:
pi@raspberrypi:~/CSharp $ ./TempHumidinCSharp
0
0
0
0
0
0
....

Output on my Pi zero 2 W with Python 3
pi@raspberrypi:~ $ python3 temp.py
Temp: 72.7 F / 22.6 C    Humidity: 46.9%
Temp: 72.7 F / 22.6 C    Humidity: 46.9%
Checksum did not validate. Try again.
Temp: 72.7 F / 22.6 C    Humidity: 46.9%
Temp: 72.7 F / 22.6 C    Humidity: 46.9%
....
Python Script from https://tutorials-raspberrypi.de/raspberry-pi-luftfeuchtigkeit-temperatur-messen-dht11-dht22/

I compiled the program on my Windows maschine with the following configuration

Configuration -> Release | Any CPU
Framework -> net5.0 and 3.5
Mode -> self-contained
Runtime -> liunx-arm
Create single file

//Dotnet --Info output

C:\Program Files (x86)\Microsoft Visual Studio\2019\Community>dotnet  --info
.NET SDK (gemäß "global.json"):
 Version:   5.0.403
 Commit:    bf9d168ac2

Laufzeitumgebung:
 OS Name:     Windows
 OS Version:  10.0.19044
 OS Platform: Windows
 RID:         win10-x64
 Base Path:   C:\Program Files\dotnet\sdk\5.0.403\

Host (useful for support):
  Version: 5.0.12
  Commit:  7211aa01b3

.NET SDKs installed:
  3.1.415 [C:\Program Files\dotnet\sdk]
  5.0.209 [C:\Program Files\dotnet\sdk]
  5.0.403 [C:\Program Files\dotnet\sdk]

.NET runtimes installed:
  Microsoft.AspNetCore.All 2.1.30 [C:\Program Files\dotnet\shared\Microsoft.AspNetCore.All]
  Microsoft.AspNetCore.All 2.2.8 [C:\Program Files\dotnet\shared\Microsoft.AspNetCore.All]
  Microsoft.AspNetCore.App 2.1.30 [C:\Program Files\dotnet\shared\Microsoft.AspNetCore.App]
  Microsoft.AspNetCore.App 2.2.8 [C:\Program Files\dotnet\shared\Microsoft.AspNetCore.App]
  Microsoft.AspNetCore.App 3.0.3 [C:\Program Files\dotnet\shared\Microsoft.AspNetCore.App]
  Microsoft.AspNetCore.App 3.1.17 [C:\Program Files\dotnet\shared\Microsoft.AspNetCore.App]
  Microsoft.AspNetCore.App 3.1.21 [C:\Program Files\dotnet\shared\Microsoft.AspNetCore.App]
  Microsoft.AspNetCore.App 5.0.8 [C:\Program Files\dotnet\shared\Microsoft.AspNetCore.App]
  Microsoft.AspNetCore.App 5.0.12 [C:\Program Files\dotnet\shared\Microsoft.AspNetCore.App]
  Microsoft.NETCore.App 2.1.28 [C:\Program Files\dotnet\shared\Microsoft.NETCore.App]
  Microsoft.NETCore.App 2.1.30 [C:\Program Files\dotnet\shared\Microsoft.NETCore.App]
  Microsoft.NETCore.App 2.2.6 [C:\Program Files\dotnet\shared\Microsoft.NETCore.App]
  Microsoft.NETCore.App 2.2.8 [C:\Program Files\dotnet\shared\Microsoft.NETCore.App]
  Microsoft.NETCore.App 3.0.3 [C:\Program Files\dotnet\shared\Microsoft.NETCore.App]
  Microsoft.NETCore.App 3.1.17 [C:\Program Files\dotnet\shared\Microsoft.NETCore.App]
  Microsoft.NETCore.App 3.1.21 [C:\Program Files\dotnet\shared\Microsoft.NETCore.App]
  Microsoft.NETCore.App 5.0.0 [C:\Program Files\dotnet\shared\Microsoft.NETCore.App]
  Microsoft.NETCore.App 5.0.8 [C:\Program Files\dotnet\shared\Microsoft.NETCore.App]
  Microsoft.NETCore.App 5.0.12 [C:\Program Files\dotnet\shared\Microsoft.NETCore.App]
  Microsoft.WindowsDesktop.App 3.0.3 [C:\Program Files\dotnet\shared\Microsoft.WindowsDesktop.App]
  Microsoft.WindowsDesktop.App 3.1.17 [C:\Program Files\dotnet\shared\Microsoft.WindowsDesktop.App]
  Microsoft.WindowsDesktop.App 3.1.21 [C:\Program Files\dotnet\shared\Microsoft.WindowsDesktop.App]
  Microsoft.WindowsDesktop.App 5.0.0 [C:\Program Files\dotnet\shared\Microsoft.WindowsDesktop.App]
  Microsoft.WindowsDesktop.App 5.0.8 [C:\Program Files\dotnet\shared\Microsoft.WindowsDesktop.App]
  Microsoft.WindowsDesktop.App 5.0.12 [C:\Program Files\dotnet\shared\Microsoft.WindowsDesktop.App]

Ellerbach, 2022-01-05:
Please use the pattern here to read: <https://github.com/dotnet/iot/tree/main/src/devices/Dhtxx#1-wire-protocol>
Just replace from the sample Dth11 by Dht22. And it will work properly.
Important: you **must** wait enough time (recommendation is close to a second) to read the sensor. Reading more often won't give you better values also won't make proper reads.

FloriEnder, 2022-01-05:
I tried the solution but it can't read the sensor.

pi@raspberrypi:~/CSharp $ ./TempHumidinCSharp
Error reading DHT sensor
05/01/2022 15:14:37
Error reading DHT sensor
05/01/2022 15:14:42
Error reading DHT sensor
05/01/2022 15:14:47
Error reading DHT sensor
05/01/2022 15:14:52
Error reading DHT sensor
05/01/2022 15:14:56
Error reading DHT sensor
05/01/2022 15:15:01
Error reading DHT sensor
05/01/2022 15:15:06
Error reading DHT sensor
05/01/2022 15:15:11
Error reading DHT sensor
05/01/2022 15:15:15
Error reading DHT sensor
05/01/2022 15:15:20
Error reading DHT sensor
05/01/2022 15:15:25
Error reading DHT sensor
05/01/2022 15:15:30
Error reading DHT sensor
05/01/2022 15:15:35
Error reading DHT sensor
05/01/2022 15:15:39