dotnet/iot #1758: Random error while waiting for pin interrupts

pelikhan, 2022-01-07:
**Describe the bug**

**Steps to reproduce**

* Open Jacdac.sln in https://github.com/microsoft/jacdac-dotnet
* Deploy Jacdac.NET.Playground to a raspberry pi equipped with https://microsoft.github.io/jacdac-docs/devices/microsoft/research/jmspibridgev37v37
* launch `dotnet System.NET.Playground.dll spi devtools`

**Expected behavior**

Don't crash or provide a way to handle issue.

**Actual behavior**

```
Unhandled exception. System.IO.IOException: Error while waiting for pin interrupts.
   at System.Device.Gpio.Drivers.SysFsDriver.WasEventDetected(Int32 pollFileDescriptor, Int32 valueFileDescriptor, Int32& pinNumber, CancellationToken cancellationToken)
   at System.Device.Gpio.Drivers.SysFsDriver.DetectEvents()
   at System.Threading.Thread.StartCallback()
```

**Versions used**

```
pi@raspberrypi:~/dotnet $ dotnet --info
.NET SDK (reflecting any global.json):
 Version:   6.0.101
 Commit:    ef49f6213a

Runtime Environment:
 OS Name:     raspbian
 OS Version:  11
 OS Platform: Linux
 RID:         linux-arm
 Base Path:   /home/pi/.dotnet/sdk/6.0.101/

Host (useful for support):
  Version: 6.0.1
  Commit:  3a25a7f1cc

.NET SDKs installed:
  6.0.101 [/home/pi/.dotnet/sdk]

.NET runtimes installed:
  Microsoft.AspNetCore.App 6.0.1 [/home/pi/.dotnet/shared/Microsoft.AspNetCore.App]
  Microsoft.NETCore.App 6.0.1 [/home/pi/.dotnet/shared/Microsoft.NETCore.App]
```

pgrawehr, 2022-01-09:
Try installing `libgpiod-dev`. That should work around the problem.

pgrawehr, 2022-01-27:
@pelikhan We've been having a few problems using SysFs in the past. I don't remember exactly what we did there about that. I think catching the exception and retrying was normally working. However, since SysFs is deprecated and slow, we switched to using libgpiod as default low-level library when it's available. 

So just run `sudo apt install -y libgpiod-dev` and the library should pick it up automatically. I've never seen the above error with libgpiod and it is orders of magnitude faster (compare #1470).

pelikhan, 2022-01-27:
Ha this is very helpful!

pelikhan, 2022-01-27:
This comment was very helpful. I think that it should be added a new section in https://docs.microsoft.com/en-us/dotnet/iot/deployment as it may be useful for anyone running .net iot core on the Pi. This is where I would logically would have found it.
This ticket can be closed from my pov, thanks for the help.

pgrawehr, 2022-01-27:
Yes, these documents are slightly out of date. We should update them.

pelikhan, 2022-01-27:
I have another thread on install woes https://github.com/dotnet/iot/issues/1783