dotnet/iot #1765: MfRc522 - method ListenToCardIso14443TypeA always returns false

artzie92, 2022-01-18:
Hi I have problem on my RaspberryPI with MfRC522 in Iot.Device.Bindings library. The method ListenToCardIso14443TypeA always returns false. I have a similar application in Node.js and it works perfectly fine.

Application code

```csharp
using System;
using System.Device.Gpio;
using System.Device.Spi;
using System.Threading;
using Iot.Device.Mfrc522;
using Iot.Device.Rfid;

try
{

    GpioController gpioController = new GpioController();
    int pinReset = 21;

    SpiConnectionSettings connection = new(0, 1);
    connection.ClockFrequency = 10_000_000;
    SpiDevice spi = SpiDevice.Create(connection);
    MfRc522 mfrc522 = new(spi, pinReset, gpioController, false);

    bool res;
    Data106kbpsTypeA card;
    do
    {
        res = mfrc522.ListenToCardIso14443TypeA(out card, TimeSpan.FromSeconds(2));

        if (res)
        {
            Console.WriteLine("card");
        }
        else
        {
            Console.WriteLine("No card.");
        }
        Thread.Sleep(res ? 0 : 1000);
    }
    while (!res);
}
catch (Exception ex)
{
    throw;
}
```

dotnet version: 5 and 6 (both doesn't work)
packages:
"Iot.Device.Bindings" Version="1.5.0"
"System.Device.Gpio" Version="1.5.0"

***Current results:***
[image]

***Device:***
[image]

raffaeler, 2022-01-19:
Hi @artzie92 the way the device is initialized may cause some problems.
I also had problems and solved using this code.
Please try this code (ignore the logger)

```
    public class RfidManager
    {
        private readonly GpioController _controller;
        private readonly ILogger? _logger;
        private readonly SpiConnectionSettings _spiSettings;
        private SpiDevice _spi;
        private MfRc522 _reader;

        public RfidManagerDotnet(GpioController controller, int pinReset, int pinSS, ILogger? logger = null)
        {
            _logger = logger;

            _controller = controller;
            _spiSettings = new(0, 0);
            _spiSettings.ClockFrequency = 5_000_000;
            _spi = SpiDevice.Create(_spiSettings);
            _reader = new(_spi, pinReset);

            Console.WriteLine($"Version {_reader.Version}");
        }

        public bool TryRead(out RfidCardData? cardData)
        {
            try
            {
                Data106kbpsTypeA card;
                cardData = null;

                if (!_reader.ListenToCardIso14443TypeA(out card, TimeSpan.FromSeconds(2)))
                    return false;

                var cardId = GetCardId(card);
                cardData = new RfidCardData(cardId, "Mifare");
                return true;
            }
            catch (Exception err)
            {
                throw new RfidException("Error reading the Rfid card", err); // just a fancy custom exception, nothing more
            }
        }

        private string GetCardId(Data106kbpsTypeA card) => Convert.ToHexString(card.NfcId);
    }
```

artzie92, 2022-01-19:
Hi @raffaeler. The only change which I see is spiSettings = new(0,0) instead of   SpiConnectionSettings connection = new(0, 1). After changing from 1 to 0 it has started working. 
Thank you for your answer!

raffaeler, 2022-01-19:
Sure, it was easier for me to paste working code :-)

You are welcome. If you believe this is ok, please close the issue.

closed 2022-01-19 by artzie92