dotnet/iot #1783: Raspberry Pi instructions about registering dotnet tool

pelikhan, 2022-01-27:
In https://docs.microsoft.com/en-us/dotnet/iot/deployment#deploying-a-framework-dependent-app, the instructions provide some command line register the dotnet tool 

[image]

These instructions work in the bash terminal (over SSH). However, the "dotnet" tool is forgotten once we start a new terminal. Is there a way to register the dotnet tool once and for all?

pgrawehr, 2022-01-27:
Yes, of course. I generally even install dotnet globaly to `/usr/share/dotnet` instead of $HOME, but since one generally uses only one user on the Pi, that doesn't matter much. 

You can register a global link to the `dotnet` command using
```bash
> sudo ln -s $HOME/.dotnet/dotnet /usr/bin/dotnet
```

pelikhan, 2022-01-27:
no dice

[image]

pgrawehr, 2022-01-28:
Then you should check where this points to. What does `ls -l /usr/bin/dotnet` say? And if you do have that file, you normally don't need to do anything, as it is already there. Unless it points to a non-existing or incomplete installation.

krwq, 2022-01-28:
@pelikhan, `echo $PATH` on new terminal and search for path which resembles `/usr/bin`, then make sure you got symbolic link of dotnet in there

joperezr, 2022-01-28:
Another option rather than linking that I like to use in Debian-based systems (which raspbian is) is to simply write a `dotnet.sh` script file with the contents
```sh
PATH=$PATH:/home/pi/.dotnet #<== Or whichever directory you downloaded dotnet executable in
```

And saving that bash script inside `/etc/profile.d/` folder. All of the bash scripts in there will always get executed when any user logs in and opens a shell, so effectively dotnet will be part of your path everytime you connect or open a terminal from the Raspberry pi.

Ellerbach, 2022-01-29:
@pelikhan is there any reason why you really need to have the framework installed on the RPI? 
I never ever do it. Instead, I build on my PC and deploy the needed resources on the RPI.
`dotnet publish -r linux-arm`
and then copy the publish folder (using scp or anything else)
`chmod +x ./nameofyourapp`
Nothing else needed.

raffaeler, 2022-01-29:
When I install the runtime or the sdk, I follow these steps that always worked for me:
1. download the appropriate `.tar.gz` and extract the content
```
sudo mkdir -p /opt/dotnet6.0.1
sudo tar zxf aspnetcore-runtime-6.0.1-linux-arm.tar.gz -C /opt/dotnet6.0.1
```
2. Create a symbolic link
```
sudo ln -s /opt/dotnet6.0.1/dotnet /usr/local/bin
```
> you can remove the symbolic link with `sudo rm /usr/local/bin/dotnet`

3. Setup the env vars
Interactive users needs the following exports to be added to the .bashrc. The same vars are needed in the `.service` file when you want to run an ASP.NET or console application as a service. These vars are required for ASP.NET apps to properly run, while other apps may run even without them.

```
cd
nano .bashrc
(add the following to the file)
export DOTNET_ROOT=/opt/dotnet6.0.1
export DOTNET=/opt/dotnet6.0.1/dotnet
```

4. Optional: set the path (alternative to the symbolic link)
```
export PATH=$PATH:/opt/dotnet6.0.1/dotnet
```

raffaeler, 2022-01-29:
@Ellerbach the reasons to install the runtime are:
- avoid re-deploying the entire framework files in every folder of the app
- update the minor releases of .NET when any security updated is published in one shot. This ensure the compatibility with every app published on the RPi and fixes the vulnerabilities for all the apps running there.

pgrawehr, 2022-01-29:
I agree to @raffaeler here, it's much easier to have the runtime installed once only.

However, there's no need to install every runtime version in a separate folder. The following works just fine as well:
```
sudo mkdir /usr/share/dotnet
sudo tar zxf aspnetcore-runtime-6.0.1-linux-arm.tar.gz -C /usr/share/dotnet
```
All versions are unpacked to the same folder, but sinde the tar contains only subfolders (except for the `dotnet` executable itself) this works fine. The tool will automatically chooses the correct runtime from all the ones that are available.

raffaeler, 2022-01-29:
@pgrawehr I never do that because I deploy services with `systemd`. For security reasons you really want to:
- create a new user
- configure the minimum set of permissions (like accessing the GPIO, etc.)
- run the service as the new user

Also, Linux can run a service with password-less and profile-less system users (as also Windows can do).
In this scenario you need to deploy .NET system-wide, hence the choice of `/opt/` folder.

Besides the `dotnet` path issue, you really need also the other env variables for ASP.NET to work. At the very end, exposing few html pages to show the app status is very handy, therefore I don't see any reason to create a simple console in most of the cases.

pgrawehr, 2022-01-29:
Ok, that's a quite different use case you have there. I'm running a desktop app as local user, most of the time on a Raspi that is not even connected to the internet or when it is, only trough a firewall. So security is not an issue.

Either case, we should improve that documentation and write suggestions for the most common scenarios. The current documentation is somewhere between to complicated, outdated or wrong.

raffaeler, 2022-01-29:
@pgrawehr definitely agree the scenarios are different.
But I believe that most of the users will connect the RPi to the internet and that needs to automatically run the app at startup. In this case a service is the best choice as it can be set-up to automatically restart if crashes (systemd provides this and other features).

The documentation should definitely be improved. I would add that things changed over the versions because in the past the `--self-contained` setting was off by default while now is on. This is just an example.

pelikhan, 2022-01-29:
I am just following the instructions at https://docs.microsoft.com/en-us/dotnet/iot/deployment which documents both options, so I expect both to work.

raffaeler, 2022-01-29:
Going back to your original issue @pelikhan, you should just print the env variables and the .bashrc file (when it does not work) to understand if there is something wrong in your setup.

There is nothing special about the dotnet tool, it's just an executable that must be in path to be found. It's just an issue with the OS, not the `dotnet` tool.

pelikhan, 2022-01-29:
Will do next week. Thanks!

Ellerbach, 2022-01-30:
> @Ellerbach Laurent Ellerbach FTE the reasons to install the runtime are:
> 
> * avoid re-deploying the entire framework files in every folder of the app
> * update the minor releases of .NET when any security updated is published in one shot. This ensure the compatibility with every app published on the RPi and fixes the vulnerabilities for all the apps running there.

You can use few more elements on the build:
`dotnet publish -r linux-arm64 -p:PublishSingleFile=true --self-contained true -c Release`

And you'll get a single file for the deployment. And you're always sure all works fine. You can signa and distribute it as a release.  Put it in a nuget or anything else. No dependencies, always clean!

raffaeler, 2022-01-30:
@Ellerbach Sure, single file deployment is very handy but the compilation times are longer.
Also, if you deploy many services on a single RPi and a new security vulnerability gets fixed in .NET, you don't want to recompile every single app.

Ellerbach, 2022-01-31:
> @Ellerbach Laurent Ellerbach FTE Sure, single file deployment is very handy but the compilation times are longer. Also, if you deploy many services on a single RPi and a new security vulnerability gets fixed in .NET, you don't want to recompile every single app.

The contrary. I want to be able to update the application. So I can properly run the tests before and make sure it's working properly. Updating something that all images dependent on without testing it is not good. And it is much easier to deploy and manage 1 application than a full framework. 
Also, in most cases, the build will be done in a pipeline. So not much time. Never build on a pi, it's a loss of time and energy. Better build in a pipeline (Linux or Windows), and then deploy the app properly.
In most of those case I btw, ot use direct deployment on the device but rather container with Azure IoT Edge.
Again, deploying the full framework on the device is not a good practice.

raffaeler, 2022-01-31:
> Again, deploying the full framework on the device is not a good practice.

@Ellerbach I never read this elsewhere, this is curious.
The test story is no different on a RPi than any other deployment scenario. That should be done elsewhere, not on a production machine, even when deploying the app as self contained.

In any case I agree not to deploy the SDK unless very special cases. 

Anyway, the OP now has all the possibilities at his fingertips.

Ellerbach, 2022-01-31:
> I never read this elsewhere, this is curious.

I've been managing quite a lot of project deploying elements on the edge on small devices. Best practices is thru containers. 
The containers have the minimum number of elements installed, so avoiding a full framework when possible (some languages do not unfortunately allow it). Reason: reduce the code deployed, reduce possible security issues with elements which are not needed.

Most deployments are done thru container management patterns like Azure IoT Edge or Kubernetes like (K8S, K3S).
In short, the less deployed, the best.

Full SDK only needed to build the containers, then you usually have a second container where you copy what has been build. This is not .NET related, it's a pattern across languages/platforms.

pelikhan, 2022-01-31:
Yes

raffaeler, 2022-01-31:
@Ellerbach I totally agree on containers. I first demo-ed docker on the Pi more than 5 years ago

raffaeler, 2022-01-31:
Anyway, let's put this way. There is a differen between what is possible and what is the suggested best practice.
We may agree or not on running with the shared runtime, it does not matter.

Since the docs on this repo provide guides to setup the runtime, let's find what is wrong and fix the docs accordingly. The reason why I created my own repo few years ago is because I didn't find updated guides on how to setup the RPi properly with .NET.
This is the right place, let's fix what does not work.

Do you agreee @Ellerbach ?

pelikhan, 2022-01-31:
I was mostly following the docs line by line on a new raspbian distro. This is still discovering .net on pi; this is a stage where I don’t really want to bring an additional layer of complexity with containers.

Why isn’t it “sudo apt install dotnet”?

Basically I want a script to (1) setup dotnet on the pi once and for all (2) press F5 in VS and get everything copied and running on the pi. Yes I am lazy and acting like a spoiled brat ;)

Ellerbach, 2022-01-31:
> Do you agreee @Ellerbach

Let's say I do agree and will help then both of you @raffaeler and @pelikhan :-)

> Why isn’t it “sudo apt install dotnet”?

Well, that does exist! You need to add first the Microsoft Debian repo. See https://docs.microsoft.com/en-us/dotnet/core/install/linux-debian

Here is the link explaining how to install properly on almost any Linux version: https://docs.microsoft.com/en-us/dotnet/core/install/linux you have sub docs for each. In most cases it's just few lines of script.

You can as well use Pete's https://www.petecodes.co.uk/install-and-use-microsoft-dot-net-5-with-the-raspberry-pi/ which is the 1 line script you want.

Are you both happy? :-)

raffaeler, 2022-01-31:
> Are you both happy? :-)

My goal here is to answer the provide an answer to the OP's issue.

We can certainly debate the runtime installation separately, but it is probably better to involve Rich as well, as most of the guidelines on .NET docs is to use the shared installation :)

Ellerbach, 2022-01-31:
> My goal here is to answer the provide an answer to the OP's issue.

I have the same goal ;-) I use the smiley as humour. Sorry if that was not clear!

And again, there are details docs, everything works fine if you still want to deploy the full framework, see the docs, I even tried (works). Now, we have to be clear that this is not a good pattern.

pgrawehr, 2022-01-31:
@Ellerbach Unfortuntately, there still seems to be no apt packages for linux-arm32 available. So you do have to go the long way.

Ellerbach, 2022-01-31:
> @Ellerbach Laurent Ellerbach FTE Unfortuntately, there still seems to be no apt packages for linux-arm32 available. So you do have to go the long way.

Arg, amr32... Right...

Ellerbach, 2022-01-31:
So then, Pets's approach with 1 line for .NET 5.0 and .NET 6.0 is a great way:
wget -O - https://raw.githubusercontent.com/pjgpetecodes/dotnet5pi/master/install.sh | sudo bash
wget -O - https://raw.githubusercontent.com/pjgpetecodes/dotnet6pi/master/install.sh | sudo bash

raffaeler, 2022-01-31:
> I have the same goal ;-) I use the smiley as humour. Sorry if that was not clear!
> 
> And again, there are details docs, everything works fine if you still want to deploy the full framework, see the docs, I even tried (works). Now, we have to be clear that this is not a good pattern.

Sure, I understand :)
I don't want to be a troublemaker, it is good clarifying the pros/cons of each deployment as this was discusses with the runtime team a long time ago. But this is not the right place.

Ellerbach, 2022-01-31:
> I don't want to be a troublemaker

You are not my dear friend :-) All good, let's then try to find the best solution for this problem (whatever I can think about the pattern ;-))

pelikhan, 2022-01-31:
It would be nice to add an option to https://docs.microsoft.com/en-us/dotnet/core/tools/dotnet-install-script to support setting the global path, so that step 2a/b can be collapsed.

[image]

The original intent of this thread is not to solve all possible deployment scenario btw. There's valid arguments for all of those based on the problem requirements.

pelikhan, 2022-01-31:
And thanks to everyone on the thread who's been sharing tips about tip. Pi noob appreciating.

joperezr, 2022-01-31:
Seems like this thread was derailed there for a bit 😆, but @pelikhan I assume you were able to figure out how to register the dotnet tool globally? Just want to make sure that's the case in order for us to close this issue. For the general documentation that you pointed out, I agree it could use with a bit more information, so it would be great if you could log an issue in [dotnet/docs](https://github.com/dotnet/docs) where the actual documentation can be fixed.

And, just to clarify a bit of what was discussed earlier here for future reference (even though this isn't really related to your question), deployment of shared framework applications is definitely a good practice just as much as self contained deployment is, and they both have pros and cons that should be evaluated in per-case basis to determine which deployment applies best for you. You can get more info on each type of deployment here: https://docs.microsoft.com/en-us/dotnet/core/deploying/deploy-with-cli

pelikhan, 2022-01-31:
I have been derailed by other things, so I won't be able to come back to it soon. Maybe you can transfer this issue to dotnet/docs.

joperezr, 2022-01-31:
For sure, I'm happy to open that issue for you.

joperezr, 2022-01-31:
I opened issue https://github.com/dotnet/docs/issues/28023 to clarify in the dotnet install scripts docs page how to persist the installation of dotnet when using those bash scripts. I'll go ahead and close this issue as it seems like the original question was answered, but @pelikhan feel free to reopen if you have any follow up questions or log new issues whenever you have any other issues, and we'll be glad to help. Have fun playing with your Raspberry Pi! 😄

closed 2022-01-31 by joperezr