dotnet/iot #1787: Prior contact for device creation

matsujirushi, 2022-02-02:
Hi,
I am going to create a new device in Iot.Device.Bindings and make a pull request.
However, I am worried about conflict with other developers.
Is there a mechanism for communicating in advance?

joperezr, 2022-02-02:
Hello @matsujirushi thanks for being interested in contributing with a binding. In general, new individual device bindings are usually developed by creating a brand new folder under src/devices folder, and adding all of your files there, so it is very rare to have conflicts with other PRs that are adding different bindings. The only potential conflict, is if you are either updating a binding that already exists, or if you are adding a new binding which is part of an existing device binding family. 

I would suggest taking a quick look a these instructions: https://github.com/dotnet/iot/tree/main/src/devices#contributing-a-binding which explain in detail what the process is for contributing a new binding. In terms of it getting included in Iot.Device.Bindings, you don't have to do anything special for that since it should just pick it up automatically as long as you follow those instructions I pointed at.

Is there a specific binding you were thinking of contributing? I ask just to see if it is something that I'm aware that somebody else is already working on, or to see if it is something that can be added to an existing device family.

matsujirushi, 2022-02-02:
Hi @joperezr ,
Thanks for reply and information.

I have a lot of [Grove modules](https://wiki.seeedstudio.com/Grove_System/) on hand.
I am going to add [SCD30](https://wiki.seeedstudio.com/Grove-CO2_Temperature_Humidity_Sensor-SCD30/) first to understand the steps of contribution.

joperezr, 2022-02-02:
I see, we do already have the [GrovePi sensor family](https://github.com/dotnet/iot/blob/main/src/devices/GrovePi/README.md) which already has some module implementations on it, so you might want to have a look and see if the ones you want to contribute are already there. I do see a [temperature module](https://github.com/dotnet/iot/blob/main/src/devices/GrovePi/Sensors/GrooveTemperatureSensor.cs) in there already, but I believe it is not exactly the SCD30 so perhaps we can add this new module there.

Ellerbach, 2022-02-02:
> I am going to add [SCD30](https://wiki.seeedstudio.com/Grove-CO2_Temperature_Humidity_Sensor-SCD30/) first to understand the steps of contribution.

That's great and seems we don't have this sensor. It is "just" packaged as a Grove sensor but behind it's a normal I2C sensor. So you can directly add this one at /src/devices in a Scd30 folder. And in the docs, you can of course point to the Grove documentation.

And on your question about conflicts, it's very rare as most people are working on either bug fix, either a new sensor. Adding those don't really impact each others. Let us know if you need any assistance!

matsujirushi, 2022-02-02:
> you can directly add this one at /src/devices in a Scd30 folder.

I got.

FYI
Grove Pi+ block diagram is [here](https://qiita.com/matsujirushi/items/c4eb769409703d9be956#%E3%83%96%E3%83%AD%E3%83%83%E3%82%AF%E5%9B%B3).
This time, I will use the Raspberry Pi 4, [Grove Base HAT](https://wiki.seeedstudio.com/Grove_Base_Hat_for_Raspberry_Pi/), and [Grove - SCD30](https://wiki.seeedstudio.com/Grove-CO2_Temperature_Humidity_Sensor-SCD30/) to check the operation.

closed 2022-02-02 by matsujirushi

Ellerbach, 2022-02-02:
> I see, we do already have the [GrovePi sensor family](https://github.com/dotnet/iot/blob/main/src/devices/GrovePi/README.md)

Since this implementation, we added board abstraction and many other things there! So I would have to implement all this for the non I2C sensors. So they could be used with or without GrovePi. Those in the repo are mainly simple analogique or digital sensors. The I2C ones should not be there.

Ellerbach, 2022-02-02:
And @pgrawehr, I'm quite sure it would be fairly easy to flash GrovePi with the Firmata implementation and basically benefit of the exact same. If ever I have spare time, I may play with this ;-)