dotnet/iot #1791: Build Scd4x

matsujirushi, 2022-02-06:
Hi,

I try to build the `Scd4x.csproj` only.

1. Open `Scd4x.sln` on Visual Studio 2022.
2. Rebuild `Scd4x.csproj`.

An error occurs.

```
1>CSC : error CS0006: Metadata file 'C:\Users\takashi\Documents\Repo\iot_dotnet\artifacts\bin\System.Device.Gpio\Debug\netstandard2.0\System.Device.Gpio.dll' could not be found
1>CSC : error CS0006: Metadata file 'C:\Users\takashi\Documents\Repo\iot_dotnet\src\devices\System.Device.Model\bin\Debug\netstandard2.0\System.Device.Model.dll' could not be found
```

Open `System.Device.Gpio.sln` and rebuild `System.Device.Gpio.csproj`, the `System.Device.Gpio.dll` error is resolved, but the `System.Device.Model.dll` is not.

Could you tell me how to build `Scd4x`?

pgrawehr, 2022-02-07:
Before you open any of the solutions in the project, run `build.cmd` in the root directory. This will once build the whole project and bring in all dependencies. It should solve your problem.

joperezr, 2022-02-07:
As @pgrawehr mentioned, it is required today before building any other project that you run `build.cmd` in windows or `build.sh` elsewhere first, which should restore everything required for you to open and build individual projects and build them. Will close this issue as the question has been answered, but @matsujirushi feel free to reopen if you have any follow up questions.

closed 2022-02-07 by joperezr