dotnet/iot #1793: Builds fails (due to tests)

richlander, 2022-02-14:
I'm running the build on 64-bit Raspberry Pi OS. It is failing due to not being able to find .NET Core 3.1.

I run `build.sh`. Here is a relevant bit of the console log from that.

```bash
rich@raspberrypi:~/iot $ ./build.sh
  Determining projects to restore...
  All projects are up-to-date for restore.
Attempting to install dotnet from public_location.
dotnet-install: Note that the intended use of this script is for Continuous Integration (CI) scenarios, where:
dotnet-install: - The SDK needs to be installed without user interaction and without admin rights.
dotnet-install: - The SDK installation doesn't need to persist across multiple CI runs.
dotnet-install: To set up a development environment or to run apps, use installers rather than this script. Visit https://dotnet.microsoft.com/download to get the installer.

dotnet-install: .NET Core Runtime with version '3.1.21' is already installed.
```

And then later:

```bash
Starting test execution, please wait...
A total of 1 test files matched the specified pattern.
Testhost process exited with error: It was not possible to find any compatible framework version
The framework 'Microsoft.NETCore.App', version '3.1.0' (arm64) was not found.
  - The following frameworks were found:
      6.0.0 at [/home/rich/iot/.dotnet/shared/Microsoft.NETCore.App]
      6.0.2 at [/home/rich/iot/.dotnet/shared/Microsoft.NETCore.App]
You can resolve the problem by installing the specified framework and/or SDK.
The specified framework can be found at:
  - https://aka.ms/dotnet-core-applaunch?framework=Microsoft.NETCore.App&framework_version=3.1.0&arch=arm64&rid=debian.11-arm64
. Please check the diagnostic logs for more information.

Test Run Aborted.
  ServoMotor.Tests -> /home/rich/iot/src/devices/ServoMotor/tests/bin/Debug/netcoreapp3.1/ServoMotor.Tests.dll
  Ssd13xx.Tests -> /home/rich/iot/src/devices/Ssd13xx/tests/bin/Debug/netcoreapp3.1/Ssd13xx.Tests.dll
/home/rich/iot/build.proj(65,5): error : Unit tests failed. Please check the detailed log to find out which ones failed.
```

And `dotnet --info`:

```bash
rich@raspberrypi:~/iot $ dotnet --info
.NET SDK (reflecting any global.json):
 Version:   6.0.100
 Commit:    9e8b04bbff

Runtime Environment:
 OS Name:     debian
 OS Version:  11
 OS Platform: Linux
 RID:         debian.11-arm64
 Base Path:   /home/rich/iot/.dotnet/sdk/6.0.100/

Host (useful for support):
  Version: 6.0.2
  Commit:  839cdfb0ec

.NET SDKs installed:
  6.0.100 [/home/rich/iot/.dotnet/sdk]

.NET runtimes installed:
  Microsoft.AspNetCore.App 6.0.0 [/home/rich/iot/.dotnet/shared/Microsoft.AspNetCore.App]
  Microsoft.NETCore.App 6.0.0 [/home/rich/iot/.dotnet/shared/Microsoft.NETCore.App]
  Microsoft.NETCore.App 6.0.2 [/home/rich/iot/.dotnet/shared/Microsoft.NETCore.App]

To install additional .NET runtimes or SDKs:
  https://aka.ms/dotnet-download
```

joperezr, 2022-02-14:
Thanks for logging @richlander I do see why this is happening. If you take a look at our current global.json file:

https://github.com/dotnet/iot/blob/d84c3e8b25b5c3e3363f096647487ab62ee0b26f/global.json#L4-L8

We are restoring netcoreapp3.1 in order to run those tests but only for x64 platforms as that is what most people are running. In order to fix this issue, we just need to add another entry there to ensure that arm64 also restores the 3.1 runtime.

richlander, 2022-02-14:
Note that macOS doesn't have an Arm64 build for .NET Core 3.1.

richlander, 2022-02-14:
> dotnet-install: .NET Core Runtime with version '3.1.21' is already installed.

Isn't that also a bug?

joperezr, 2022-02-14:
>Isn't that also a bug?

It might, but that is not in our code. That is part of the output of calling the [dotnet install scripts](https://docs.microsoft.com/en-us/dotnet/core/tools/dotnet-install-script#:~:text=%20dotnet-install%20scripts%20reference%20%201%20Name.%20,%3Cauto%3E%2C%20amd64%2C%20x64%2C...%205%20Examples.%20%20More%20) when having a global.json file.

joperezr, 2022-02-14:
> Note that macOS doesn't have an Arm64 build for .NET Core 3.1. macOS Arm64 support starts with .NET 6.

Interesting. We can for sure omit the netcoreapp3.1 tests from OSX when running on Arm64.

richlander, 2022-02-14:
Right. I assume it is not our bug. However, it doesn't seem to directly relate to `global.json` since I don't the 3.1 install after the fact. I'll file another issue.

krwq, 2022-08-25:
[Triage] Seems from the thread there is some external issue, is there any issue tracking that?

joperezr, 2022-09-01:
[Triage]: This should be fixed now. We haven't seen any issues in CI due to this and we have changed the targets for our tests. Closing for now.

closed 2022-09-01 by joperezr