dotnet/iot #1799: device.Capture("filename.jpg") creates an invalid file!

alvipeo, 2022-02-19:
**Describe the bug**

According to the docs here - [https://github.com/dotnet/iot/blob/main/src/devices/Media/README.md#videodevice](https://github.com/dotnet/iot/blob/main/src/devices/Media/README.md#videodevice) - this should save a jpg to a file:
```
VideoConnectionSettings settings = new VideoConnectionSettings(busId: 0, captureSize: (2560, 1920), pixelFormat: PixelFormat.YUYV);
using VideoDevice device = VideoDevice.Create(settings);
// Capture static image
device.Capture("/home/pi/jpg_direct_output.jpg");
```

but the file is created corrupted and/or in wrong format.

**Steps to reproduce**

just run your code or provide a complete working example.

**Versions used**

```
	<PropertyGroup>
		<OutputType>Exe</OutputType>
		<TargetFramework>net6.0</TargetFramework>
		<Platforms>AnyCPU;ARM64</Platforms>
		<LangVersion>Latest</LangVersion>
		<Nullable>enable</Nullable>
		<TreatWarningsAsErrors>true</TreatWarningsAsErrors>
	</PropertyGroup>
...
	<ItemGroup>
		...
		<PackageReference Include="Iot.Device.Bindings" Version="2.0.0" />
		<PackageReference Include="System.Device.Gpio" Version="2.0.0" />
	</ItemGroup>
```

then I build and publish it to Raspberry Pi 3:
```
dotnet publish -r linux-arm --self-contained True -c Debug -p:PublishSingleFile=true -p:GenerateRuntimeConfigurationFiles=true
```

# another variant I tried

```
    private void CaptureThePic(CancellationToken cancellationToken)
    {
        /*
         *
         * https://github.com/dotnet/iot/blob/main/src/devices/Media/README.md
         *
         */

        _logger.LogInformation("Capturing the pic...");

        VideoConnectionSettings settings = new(0, (1920, 1080), PixelFormat.YUV420);
        using VideoDevice device = VideoDevice.Create(settings);

        string picName = DateTimeOffset.UtcNow.ToString("s").Replace(":", "");
        string picFileName = $"{PICS_FOLDER_S}/p{picName}.jpg";

        // Get image stream, convert pixel format and save to file
        byte[] imgBytes = device.Capture();
        using MemoryStream ms = new(imgBytes);
        Color[] colors = VideoDevice.Yv12ToRgb(ms, settings.CaptureSize);
        Bitmap bitmap = VideoDevice.RgbToBitmap(settings.CaptureSize, colors);

        if (cancellationToken.IsCancellationRequested) return;

        bitmap.Save(picFileName, ImageFormat.Jpeg);

        // Capture static image

        _logger.LogInformation($"Saving pic as '{picFileName}'...");

        if (cancellationToken.IsCancellationRequested) return;

        // Capture static image
        device.Capture(picFileName);

        _logger.LogInformation("Saved.\n");
    }
```

same problem.

Ellerbach, 2022-06-10:
@alvipeo just to make sure to compare the same code: on one side you create the video device with PixelFormat.YUYV (non working code) and on the other side with PixelFormat.YUV420 (working code).
If you are creating with the same working one, is the code working or not at all?

CodedBeard, 2022-06-13:
>     * This other issue has been closed but probably it was not fixed by the linked pull request: [VideoDevice issues with color accuracy. #1688](https://github.com/dotnet/iot/issues/1688)

@raffaeler Yes, I can confirm that issue was not resolved by the PR, I've just been super busy and haven't had a chance to follow up since it was closed.

raffaeler, 2022-06-13:
Thanks for your confirmation @CodedBeard. I suspected that when I was testing and reviewing all the issues related to the VideoDevice.
I also see the green rectangle full screen as for that issue.

@joperezr @krwq Could you please reopen issue #1688?

jesseryoung, 2022-06-13:
When I was investigating https://github.com/dotnet/iot/issues/1727 I noticed that VideoDevice will not raise an exception if you attempt to set the format to something the driver doesn't support. I think this is because it isn't looking at the return value from calling VIDIOC_S_FMT:
https://github.com/dotnet/iot/blob/1cae134bbd2958bb1fa7ac608169c33afdba3533/src/devices/Media/VideoDevice/Devices/UnixVideoDevice.cs#L404-L417

According to the V4L2 documentation, the response to this call should be an error code when the driver doesn't support that pixel format.

You should be able to see what pixel formats the driver you're using supports using v42l-ctl.
```sh
v4l2-ctl --list-formats-ext -d /dev/videoX
```
Where X is your video device.

However, @raffaeler I think your conclusion is correct here, upgrading off the legacy V4L2 to the libcam library is probably the best route.

raffaeler, 2022-06-14:
Hi @jesseryoung while I agree the return code must be checked, probably the author of this library assumed that the user would use one of the values returned from the supported formats.
In my tests I specifically tested the supported formats and the list is very long for all the 4 cameras, which includes the YUV format specifiied by the sample code.
After posting I also tested other common formats like RGB24 but they lead to the same results.

> However, @raffaeler I think your conclusion is correct here, upgrading off the legacy V4L2 to the libcam library is probably the best route.

Since V4L2 behavior is emulated by libcamera, I would first like to understand what is failing exactly here, and providing a solution on this stack first. In a second step, I will expose the situation to the `triage` team and discuss what to in relationship with the newer libcamera specs.

Ellerbach, 2022-06-14:
> if you have some idea/hint, you are welcome

I don't have any straight forward idea. And agree with @jesseryoung on:

> upgrading off the legacy V4L2 to the libcam library is probably the best route

Do you have an idea of what will be the implication on the API to import? any changes there?

> I noticed that VideoDevice will not raise an exception if you attempt to set the format to something the driver doesn't support. I think this is because it isn't looking at the return value from calling VIDIOC_S_FMT

That is correct, in this implementation, developer is responsible to call the supported format. There is the ability in the API to get the supported formats. We could call the function and check that the format is supported if that helps. And raise an exception in the case it's not.

CodedBeard, 2022-06-14:
@raffaeler I might be misunderstanding the docs, but I took [this section](https://www.raspberrypi.com/documentation/accessories/camera.html#v4l2)  to mean that the V4L2 compatability shim isn't available to 3rd parties and only works with the new libcamera apps.

>The V4L2 interface for Unicam is available only when using libcamera. 

If my understanding is correct, I think that means it has to be re-written to use the new driver directly.

raffaeler, 2022-06-14:
I see @CodedBeard but I understand from these posts (authors looks credible) that there is "just" the need to configure them appropriately with the media api:
https://forums.raspberrypi.com/viewtopic.php?t=322076
https://lwn.net/Articles/881164/

I tried some of the configurations but I still can't see a result.
More specifically from the second link, I am still not able to use the media-ctl as suggested.
If you understand differently, I am interested to know :-)

raffaeler, 2022-06-14:
As a side note, I added to my fork the code that checks all the `result code` and the `errno` from the interop (ioctl, etc.).
I can now see the all the `ioctl` calls **failing** which means that:
- either we can configure the `unicam` driver to emulate correctly the old `v4l2` driver
- or we just stay with this behavior specifying in the readme that the current binding works with the *legacy* support in `raspi-config` (the one that allows raspivid and raspistill to work as well).

Of course, if anyone beyond @CodedBeard can provide additional hints/suggestions/test outcome, is more than welcome.

In any case, it appears the best solution is to create a new binding embracing the new `libcamera` architecture.

raffaeler, 2022-06-14:
@CodedBeard I have very good news. I am going to write down a recap but need a while.
In the meantime, don't waste time on the issue because I could just run (two minutes ago and **not** extensively tested) a modified binding that works without enabling the legacy layer but just use the compatibility layer provided by `libcamera`.

CodedBeard, 2022-06-14:
@raffaeler cool, as I couldn't get any of those settings to work either :)

CodedBeard, 2022-06-15:
@raffaeler great, ill give option 2 a go this evening. Do you want me to put any feedback under #1875 instead?

raffaeler, 2022-06-15:
Sure @CodedBeard, that would be great.
I did not have much time to make other tests today. As soon as we see this compatibility layer working decently, I will make a pull request with this and other changes I already made on my side.

These are the modified PInvokes that I currently declared. All these calls in the bindings should not point to `Libc` anymore.
In the pull request I'll make sure that this is transparent to the user of course.

```
    [DllImport("v4l2-compat.so", SetLastError = true)]
    internal static extern int open([MarshalAs(UnmanagedType.LPStr)] string pathname, Interop.FileOpenFlags flags);

    [DllImport("v4l2-compat.so")]
    public static extern int close(int fd);

    [DllImport("v4l2-compat.so", SetLastError = true)]
    public static extern int ioctl(int fd, int request, IntPtr argp);

    [DllImport("v4l2-compat.so", SetLastError = true)]
    public static extern IntPtr mmap(IntPtr addr, int length, MemoryMappedProtections prot, MemoryMappedFlags flags, int fd, int offset);

    [DllImport("v4l2-compat.so")]
    public static extern int munmap(IntPtr addr, int length);

    public enum MemoryMappedProtections
    {
        PROT_NONE = 0x0,
        PROT_READ = 0x1,
        PROT_WRITE = 0x2,
        PROT_EXEC = 0x4
    }

    [Flags]
    public enum MemoryMappedFlags
    {
        MAP_SHARED = 0x01,
        MAP_PRIVATE = 0x02,
        MAP_FIXED = 0x10
    }
```

raffaeler, 2022-07-01:
I just updated my comment above with all the details about the short/medium/long term strategies to access the video camera:
https://github.com/dotnet/iot/issues/1799#issuecomment-1154057026

raffaeler, 2023-03-10:
@serber Unfortunately not. 
If you use the legacy stack, you should be fine. Libcamera currently does not provide any wrapper that is binary-stable (C interface). Until that, integrating it here is not an easy task.

closed 2023-11-17 by msftbot[bot]