dotnet/iot #1830: Raspberry Pi 400 throws PlatformNotSupportedException in Raspberry Pi 3 Driver

Callum6677, 2022-03-09:
**Describe the bug**

On a raspberry pi 400, initialising a GPIO controller results in usage of the SysFs driver rather than the expected RaspberryPi3Driver. Upon altering my code to force the raspberry pi 3 driver, I was presented with a PlatformNotSupportedException

**Steps to reproduce**

```csharp
var controller = new GpioController(PinNumberingScheme.Logical, new RaspberryPi3Driver());
```

**Expected behavior**

The driver to be successfully created

**Actual behavior**

PlatformNotSupportedException thrown with message: "Not a supported Raspberry Pi type: RaspberryPi400"

**Versions used**

System.Device.GPIO 2.1.0.
dotnet 6.0.3

Callum6677, 2022-03-09:
Issue appears to be in RaspberryPi3Driver.CreateInternalRaspberryPi3LinuxDriver. RaspberryPi400 is missing from the switch cases

pgrawehr, 2022-03-10:
We do have a known issue there, see #1798 

Nobody had a good idea on how to make the detection more generic so far, though.

john-mills-nz, 2022-07-10:
I am also experiencing this issue on a Raspberry Pi 400 running 32-bit Raspberry Pi OS with .NET 6.0.6. It _should_ work since the 400 is pretty close to a 4 Model B.

Having had a look at the driver code, I agree it's the detection mechanism/switch statement that needs fixing to support the RPi 400.

```
.NET SDK (reflecting any global.json):
 Version:   6.0.301
 Commit:    43f9b18481

Runtime Environment:
 OS Name:     raspbian
 OS Version:  11
 OS Platform: Linux
 RID:         linux-arm
 Base Path:   /home/john-mills/.dotnet/sdk/6.0.301/

Host (useful for support):
  Version: 6.0.6
  Commit:  7cca709db2

```

PR #1885 (2022-07-14): Support Raspberry Pi 400
Fixes #1830 by modifying `RaspberryPi3Driver.CreateInternalRaspberryPi3LinuxDriver()` to support the Raspberry Pi 400.

###### Microsoft Reviewers: [Open in CodeFlow](https://portal.fabricbot.ms/api/codeflow?pullrequest=https://github.com/dotnet/iot/pull/1885)
closed 2022-07-14 by joperezr