dotnet/iot #1836: Improve generated documentation in docs.microsoft.com

Ellerbach, 2022-03-22:
Part of a continuous effort to improve the documentation, there are elements to be done to improve the all up experience:

- Add a reference to the readme from each binding folder to the source code of each main class
  - should look like this in the intellisense comments: [!INCLUDE[detailed usage](~/includes/Ads115/README.md)]
  - the folder name will have to match the name of the directory
  - the file name is always READM.md as so far we only have this one
  - in some cases like RTC, there are subfolder, follow the same subfolder strcture
- When creating a release the tool which will transform the READMEs need to be run. Tool located here: https://github.com/dotnet/iot/tree/main/tools/GenerateDocFxStructure
  - the tool will create all the directories necessary, the sub directories if needed, move the README
  - the tool will move as well all the images in another specific directory with a proper structure
  - the tool will adjust all the necessary links from the pictures
  - the tool will adjust all relative links to the repo to transform them into absolute links
  - **all** the generated files need to be moved with the xml docs generated content for the PR into the iot-api-docs repository
- The [docfx.json](https://github.com/dotnet/iot-api-docs/blob/main/iot-api-docs/docfx.json) file from the iot-api-docs must be adjusted to include the new files as content (include and media), maybe other adjustments needed

pgrawehr, 2022-03-24:
@Ellerbach I'm a bit confused as to which steps need to be done now. Is this ticket required to get #1805 resolved?

Ellerbach, 2022-03-24:
> @Ellerbach Laurent Ellerbach FTE I'm a bit confused as to which steps need to be done now. Is this ticket required to get #1805 resolved?

Related, yes. Those steps are necessary to improve the doc experience you reference. And by doing this, the docs will be 1. much better and 2. up to date.

The source of the docs.microsoft.com you reference is here: https://github.com/dotnet/iot-api-docs so that's as well the reason of the steps.

@joperezr is working on all this.