dotnet/iot #1837: Error with BuildHAT

dhusej, 2022-03-24:
**Describe the bug**

Hi,
I wanted to test BuildHAT with Raspberry Pi4 and in the test example I got next error:

Unhandled exception. System.Resources.MissingManifestResourceException: Could not find the resource "Iot.Device.BuildHat.Resource.resources" among the resources "Iot.Device.Bindings.BigFontMap.txt" embedded in the assembly "Iot.Device.Bindings", nor among the resources in any satellite assemblies for the specified culture. Perhaps the resources were embedded with an incorrect name.
   at System.Resources.ManifestBasedResourceGroveler.HandleResourceStreamMissing(String fileName)
   at System.Resources.ManifestBasedResourceGroveler.GrovelForResourceSet(CultureInfo culture, Dictionary`2 localResourceSets, Boolean tryParents, Boolean createIfNotExists)
   at System.Resources.ResourceManager.InternalGetResourceSet(CultureInfo culture, Boolean createIfNotExists, Boolean tryParents)
   at System.Resources.ResourceManager.GetObject(String name, CultureInfo culture, Boolean wrapUnmanagedMemStream)
   at System.Resources.ResourceManager.GetObject(String name, CultureInfo culture)
   at Iot.Device.BuildHat.Resource.get_firmware()
   at Iot.Device.BuildHat.Brick.CheckForFirmwareAndUpload()
   at Iot.Device.BuildHat.Brick.Initialize()
   at Iot.Device.BuildHat.Brick..ctor(String port, GpioController controller, Int32 reset, Boolean shouldDispose)

**Steps to reproduce**

```csharp
using System;
using System.ComponentModel;
using System.Threading;
using Iot.Device.BuildHat;
using Iot.Device.BuildHat.Models;
using Iot.Device.BuildHat.Motors;
using Iot.Device.BuildHat.Sensors;

namespace ConsoleApp1
{
    internal class Program
    {
        static void Main(string[] args)
        {
            Console.WriteLine("Hello, BuildHat!");
            bool continueToRun = true;

            // On Windows, connected through a serial dongle:
            //using Brick brick = new("COM3");
            // On a Raspberry PI, you'll use:
            using Brick brick = new("/dev/serial0");
            var info = brick.BuildHatInformation;
            Console.WriteLine($"version: {info.Version}, firmware date: {info.FirmwareDate}, signature:");
            Console.WriteLine($"{BitConverter.ToString(info.Signature)}");
            Console.WriteLine($"Vin = {brick.InputVoltage.Volts} V");
            Console.WriteLine("Select what you want to test:");
            Console.WriteLine(" 1. Display elements details");
            Console.WriteLine(" 2. Display connection/disconnecion");

            while (!Console.KeyAvailable)
            {
                Thread.Sleep(100);
            }

            var choice = Console.ReadKey();

            switch (choice.KeyChar)
            {
                case '1':
                    DisplayElementDetails();
                    break;
                case '2':
                    DisplayConnectionDisconnection();
                    break;

                default:
                    break;
            }

            void DisplayElementDetails()
            {
                Console.WriteLine("Displaying details of all the connected elements");
                // Display all the details of all the sensors
                for (int i = 0; i < 4; i++)
                {
                    SensorType sensor = brick.GetSensorType((SensorPort)i);
                    Console.Write($"Port: {i} {(Brick.IsMotor(sensor) ? "Sensor" : "Motor")} type: {sensor} Connected: ");

                    if (Brick.IsActiveSensor(sensor))
                    {
                        ActiveSensor activeSensor = brick.GetActiveSensor((SensorPort)i);
                        Console.WriteLine($"{activeSensor.IsConnected}");
                        foreach (var mode in activeSensor.ModeDetails)
                        {
                            Console.WriteLine($"  M{mode.Number} {mode.Name} {mode.Unit}");
                            Console.WriteLine($"    format count={mode.NumberOfData} type={mode.DataType} chars={mode.NumberOfCharsToDisplay} dp={mode.DecimalPrecision}");
                            foreach (var minmax in mode.MinimumMaximumValues)
                            {
                                Console.WriteLine($"    {minmax.TypeValues} min={minmax.MinimumValue} max={minmax.MaximumValue}");
                            }
                        }

                        foreach (var combi in activeSensor.CombiModes)
                        {
                            Console.Write($"  C{combi.Number} ");
                            foreach (var m in combi.Modes)
                            {
                                Console.Write($"{m} ");
                            }

                            Console.WriteLine();
                        }

                        Console.WriteLine($"Speed: {activeSensor.SpeedPid.Pid1} {activeSensor.SpeedPid.Pid2} {activeSensor.SpeedPid.Pid3} {activeSensor.SpeedPid.Pid4}");
                        Console.WriteLine($"Position: {activeSensor.PositionPid.Pid1} {activeSensor.PositionPid.Pid2} {activeSensor.PositionPid.Pid3} {activeSensor.PositionPid.Pid4}");
                    }
                    else
                    {
                        var passive = (Sensor)brick.GetSensor((SensorPort)i);
                        Console.WriteLine(passive.IsConnected);
                    }
                }

                Console.WriteLine("Press a key to continue");
            }

            void DisplayConnectionDisconnection()
            {
                while (!Console.KeyAvailable)
                {
                    Console.Clear();
                    Console.CursorTop = 0;
                    for (int i = 0; i < 4; i++)
                    {
                        Console.CursorLeft = 0;
                        SensorType sensor = brick.GetSensorType((SensorPort)i);
                        Console.Write($"Port: {i} {(Brick.IsMotor(sensor) ? "Sensor" : "Motor")} type: {sensor} Connected: ");
                        if (sensor != SensorType.None)
                        {
                            if (Brick.IsMotor(sensor))
                            {
                                if (Brick.IsActiveSensor(sensor))
                                {
                                    var motor = (ActiveMotor)brick.GetMotor((SensorPort)i);
                                    Console.WriteLine($"{motor.IsConnected}");
                                }
                                else
                                {
                                    var motor = (PassiveMotor)brick.GetMotor((SensorPort)i);
                                    Console.WriteLine(motor.IsConnected);
                                }
                            }
                            else
                            {
                                if (Brick.IsActiveSensor(sensor))
                                {
                                    var motor = (ActiveSensor)brick.GetSensor((SensorPort)i);
                                    Console.WriteLine(motor.IsConnected);
                                }
                                else
                                {
                                    var motor = (Sensor)brick.GetSensor((SensorPort)i);
                                    Console.WriteLine(motor.IsConnected);
                                }
                            }
                        }
                    }

                    Thread.Sleep(100);
                }

                Console.ReadKey();
            }
        }
    }
}
```

**Expected behavior**

Read version, firmware, voltage

**Actual behavior**

Error in line "using Brick brick = new("/dev/serial0");"

**Versions used**

- `dotnet --info` on the machine being used to build: .NET 5
- `dotnet --info` on the machine where app is being run: .NET 5
- Version of `System.Device.Gpio` package 2.1.0
- Version of `Iot.Device.Bindings` package 2.1.0

pgrawehr, 2022-03-25:
Pinging @Ellerbach 

It appears that the required resource is not correctly included in the full build, or has a different name than in the individual binding packages.

Ellerbach, 2022-03-25:
> It appears that the required resource is not correctly included in the full build, or has a different name than in the individual binding packages.

Let me look at all this. I identified the resource issue. Will investigate and quickly provide a fix.

@dhusej if you want, you can clone the Iot repository and directly run the sample there with the source reference. It will work.

Ellerbach, 2022-03-25:
@dhusej fix in PR, you'll have to wait for the new nuget to get a full fix (should be a mater of week or so) and you can use the daily one once the PR will be merges. And you still have the option with the code I described to make it working.
@pgrawehr please review. Tested and seems to work fine now.

PR #1838 (2022-03-28): Adjusting to add missing resources for BuildHAT
- Fixes #1837
- Tested on the generated dll from the build command with the same code as the sample
closed 2022-03-28 by Ellerbach