dotnet/iot #1839: Do not throw ObjectDisposedException in ExtendedCommandHandler.Dispose()

SupinePandora43, 2022-03-29:
# The problem

https://github.com/dotnet/iot/blob/80d67e0176fb49820adab393711cc5db1ecdd0c2/src/devices/Arduino/ArduinoBoard.cs#L701

# Solution

Remove `if(_isDisposed) throw new ObjectDisposedException();` somewhere.

# Alternatives

* `public bool IsDisposed { get; private set; }`

###### maybe it doesn't throw anything at all and try/catch is reduant here
###### ~~remove `api-suggestion` label~~

pgrawehr, 2022-03-31:
Can you elaborate on what the problem is you are facing?

SupinePandora43, 2022-03-31:
> Can you elaborate on what the problem is you are facing? 

Uhm... I don't?

SupinePandora43, 2022-03-31:
> @SupinePandora43 Not really. Yes, catching ObjectDisposedException is not really a nice thing and should probably be avoided, but are you having any actual problems with the current implementation? Such as the exception falling through?

No, I'm just confused by this style of disposing

Ellerbach, 2022-03-31:
> No, I'm just confused by this style of disposing

To be more precise, what does confuse you?