dotnet/iot #1843: Roll-forward from 3.1 -> 6.0 fails

richlander, 2022-04-03:
I was playing with some 3.1 samples and trying to roll them forward to 6.0. That didn't work. I'm unaware of a compatibility issue that would have caused this. If I run the 3.1 sample on the 3.1 runtime, it works. If I update the sample to target 6.0, then it similarly works. I tried this with multiple samples.

I know we made some changes during .NET 5 or 6 where we changed the way the GPIO library was packaged, particularly as it relates to Windows (although I'm on Linux).

I have three basic questions:

- Is this compatibility issue known?
- If so, is it documented?
- Why does it show up as a failure to create CoreCLR?

Here is what I see:

```bash
rich@raspberrypi:~/iot/src/devices/ShiftRegister/samples $ dotnet bin/Debug/netcoreapp3.1/ShiftRegister-driver.dll
It was not possible to find any compatible framework version
The framework 'Microsoft.NETCore.App', version '3.1.0' (arm64) was not found.
  - The following frameworks were found:
      6.0.0 at [/home/rich/iot/.dotnet/shared/Microsoft.NETCore.App]
      6.0.2 at [/home/rich/iot/.dotnet/shared/Microsoft.NETCore.App]

You can resolve the problem by installing the specified framework and/or SDK.

The specified framework can be found at:
  - https://aka.ms/dotnet-core-applaunch?framework=Microsoft.NETCore.App&framework_version=3.1.0&arch=arm64&rid=debian.11-arm64
rich@raspberrypi:~/iot/src/devices/ShiftRegister/samples $ dotnet --roll-forward Major bin/Debug/netcoreapp3.1/ShiftRegi
ster-driver.dll
Failed to create CoreCLR, HRESULT: 0x80070057
rich@raspberrypi:~/iot/src/devices/ShiftRegister/samples $ ~/.dotnet/dotnet bin/Debug/netcoreapp3.1/ShiftRegister-driver
.dll
Driver for ShiftRegister
Register bit length: 8
Using GPIO
Light up three of first four LEDs
Light up all LEDs, with ShiftBit
Dim up all LEDs, with ShiftBit
^CWrite a set of values with ShiftByte
Value: 1
```

Note: The `dotnet` in the path has just .NET 6. The .NET at `/.dotnet` has .NET Core 3.1 and .NET 6.

joperezr, 2022-04-03:
Major version roll-forward is not a supported scenario in .NET, and instead more of a "do at your own risk" scenario. I would be happy to debug to figure out exactly what is going on here, but I bet the issue you are hitting actually doesn't have to do a lot with our IoT libraries, and instead is a problem with some other types from the framework itself that are causing this. Especially since the error is being hit very early in the execution (trying to initialize CoreCLR).

Do you have a repro project? I would love to take a look as to why CoreCLR is failing to initialize.

joperezr, 2022-04-03:
I guess one good way to make sure this is not IoT related, you could try commenting out the piece of code that actually uses our IoT libraries and try to repro again. My hunch is that you would still get a repro in that case.

richlander, 2022-04-04:
I can try that.

However, major version roll forward is a supported scenario and we rely on it in multiple ways.

pgrawehr, 2022-04-04:
Related: #1103 #1099 #1807

I believe this is caused by the fact that the winmd support was removed from. NET 5. When the newer runtime sees a winmd reference in an assembly,  the CLR startup fails exactly with the error above.

So to run your app under NET 5 or later, you need an iot build for that runtime.

krwq, 2022-04-28:
[Triage] @richlander this is happening because we removed winmd in .NET, is this by design?

closed 2022-04-28 by krwq