dotnet/iot #1851: input pin pin changed callback not firing

fbd-ss, 2022-04-15:
As the title says when setting up an input pin with RegisterCallbackForPinValueChangedEvent, the callback never gets fired.

The board is a raspberry Pi 4b, with the service running under systemd as a backgound service.
System.Device.Gpio version 2.1.0

Having no dramas with outputs.
The input is working as the value can be read as changed directly after it has been set.

Any ideas anyone having dramas with this?

pgrawehr, 2022-04-16:
Have you installed libgpiod? This usually fixes this. Run `sudo apt install libgpiod-dev`

fbd-ss, 2022-04-16:
Managed to get it going, i had the non dev version installed.  Not sure if it was that or i was calling open_pin for a second time before registering the callback and an exception was getting thrown but not showing up in the unhandled exception handler.

Q, do i have to explicitly set up the constructor like "new GpioController( PinNumberingScheme.Logical, new LibGpiodDriver( 0 ) )" to use the libgpiod library?  If just setting up like "new GpioController()" will it revert to the old sys driver.

pgrawehr, 2022-04-16:
No, normally that should work correctly automatically, unless you are facing another instance of #1798 . Can you check your board revision?

fbd-ss, 2022-04-16:
Revision           : b03112
SoC                : BCM2711

fbd-ss, 2022-04-16:
Ill put this down to i had libgpiod not libgpiod-dev installed then in InitializeInterruptDriver it defaulted to SysFs and the fact then the pin was being opened a second time freaked out the driver and threw an exeption before the handler was getting registered.

Ive just tried it with the default constructor and all seems well.

Thanks D.

pgrawehr, 2022-04-21:
Closing this issue as it appears to be resolved.

closed 2022-04-21 by pgrawehr