dotnet/iot #1859: Add support for Raspberry Pi 4 rev. C03115

JeffreyStorm, 2022-05-10:
Dear Team,

I would really need the support for this revision, because with this I get the "InputPulldown not supported" error, and with previous revisions it is working fine. I am using the 2.1 version of this package.

Here are some extra info:
Model information : Raspberry Pi 4 model B Rev 1.5
Revision : C03115
SoC: BCM2711

If this cannot be done in the near future, could anyone provide me some way, to do it myself ?
Thanks in advance

krwq, 2022-05-10:
Would you be interested in contribution? Most likely the change you need to make is here: https://github.com/dotnet/iot/blob/main/src/System.Device.Gpio/System/Device/Gpio/RaspberryBoardInfo.cs#L150 - you will need to write some simple test to make sure the fix works though

JeffreyStorm, 2022-05-11:
Dear krwq,
First of all, many thanks for helping me out wiht this issue.
I'm using the latest official  2.1 release. I also checked the link you pasted here and I can see the revision number as well in there, so it should work. I will try couple things today ( like removing/readding the nuget package and/or downloading and building the source manually ) and I will post back the results.