dotnet/iot #1878: Exception "initialization error no gpio controllers exist on this system."

nmorel76700, 2022-06-28:
**Describe the bug**

Hello
I would like to use a GPIO port present on the Kbox A 250 hardware from Kontron but when initializing the class, I got the error "initialization error no gpio controllers exist on this system."
However, the hardware is recognized in the device manager.

[image]

**Steps to reproduce**
// put sample code here

   `var gpio = new GpioController();`

// put sample code here
`using System;
using System.Collections.Generic;
using System.ComponentModel;
using System.Data;
using System.Drawing;
using System.Linq;
using System.Text;
using System.Threading.Tasks;
using System.Windows.Forms;
using System.Device.Gpio;
using System.Device.Gpio.Drivers;

namespace TestGPIO2
{
    public partial class Form1 : Form
    {
        private const int DEFAUT_TENSION_PIN = 2;
        private GpioController gpioCtrl;
        private System.Device.Gpio.Drivers.Windows10Driver gpioCtrl2;
        public Form1()
        {
            InitializeComponent();

            try
            {
                gpioCtrl = new GpioController();
                gpioCtrl.OpenPin(DEFAUT_TENSION_PIN, PinMode.Input);
                ajoutEvent("Initialisation ok");
                try
                {
                    gpioCtrl.RegisterCallbackForPinValueChangedEvent(DEFAUT_TENSION_PIN, PinEventTypes.Falling | PinEventTypes.Rising, new PinChangeEventHandler(this.OnInputPinValueChanged)); //Exception thrown
                    ajoutEvent("abonnement à l'évènement réussis");
                }
                catch (Exception e)
                {
                    ajoutEvent("Erreur d'abonnement à l'évènement");
                    ajoutEvent(e.Message);
                }
            }
            catch (Exception e)
            {
                ajoutEvent("Initialisation erreur");
                ajoutEvent(e.Message);
            }

        }

        private void OnInputPinValueChanged(object sender, PinValueChangedEventArgs arguments)
        {
            ajoutEvent("Déclenechement de lévènement de changement de pin");
            ajoutEvent("Senders : " + sender.ToString());
            ajoutEvent("Number pin : " + arguments.PinNumber);
            ajoutEvent("Change type :" + arguments.ChangeType.ToString());
        }

        private void ajoutEvent(string s_text)
        {
            tb_Event.Text = tb_Event.Text + "\n\n" + s_text;
        }
    }
 }
`

Expected behavior

I expect such an object to be created.

Actual behavior

I get a System.NotSupportedException with the message "No GPIO controllers exist on this system"

**Versions used**

Ellerbach, 2022-06-28:
Same as #1411. See the thread, not yet supported.

krwq, 2022-06-28:
I'm wondering if we could adapt our linux RPi3 driver which accesses memory directly and do it similarly on Windows. It would require admin privileges but still better than not working at all

pgrawehr, 2022-06-28:
@krwq How could adopting the existing Rpi  driver work? This is an Intel based system.

krwq, 2022-06-30:
I meant the idea of using memory directly

krwq, 2022-10-13:
[Triage] Decreasing priority because this is outside of our control and we're focusing on Linux at the moment