dotnet/iot #1886: `Iot.Device.Common.ValueArray<T>` is not safe and contains GC holes

DaZombieKiller, 2022-07-14:
**Describe the bug**
I want to preface this by stating that I'm not a consumer of these APIs, I just happened to come across this while perusing [source.dot.net](https://source.dot.net/).

The [`ValueArray<T>`](https://github.com/dotnet/iot/blob/1e0f186df8ee1c9f8fca44446bb6426f972d4ee3/src/devices/Common/Iot/Device/Common/ValueArray.cs) type introduced in https://github.com/dotnet/iot/pull/1701 stores a sequence of 8 elements, and allows consumers to retrieve a `Span<T>` wrapping them. However, the method for which it does this is not safe and causes the GC to lose track of the references, meaning that the object they belong to may be eligible for garbage collection while the `Span<T>` is in use.

For reference, this is the current implementation of `ValueArray<T>.AsSpan`:
```cs
public unsafe Span<T> AsSpan()
{
    return new Span<T>(Unsafe.AsPointer(ref _e0), _count);
}
```
And this would be an implementation that retains GC tracking (compatible with .NET Core/Standard 2.1 and newer):
```cs
public Span<T> AsSpan()
{
    return MemoryMarshal.CreateSpan(ref _e0, count);
}
```
Unfortunately, `MemoryMarshal.CreateSpan` is not available on .NET Standard 2.0. A close equivalent to this API used to exist that would have made it possible to create `Span<T>` instances for `ValueArray<T>`s stored within classes safely (see https://github.com/dotnet/runtime/issues/27690).

As the `ValueArray<T>` type is stored within `class`es, changing it to a `ref struct` is also not an option, meaning that there is no safe way to implement `AsSpan` on `ValueArray<T>` under .NET Standard 2.0.

raffaeler, 2022-07-14:
> However, the method for which it does this is not safe and causes the GC to lose track of the references

Please consider that `T` must be a value type and `ValueArray` is a struct (whose members are all value types), so the GC is not involved here.

Given that `Span<T>` never takes ownership of the memory but just "view" it, and since `T` is a value type, as long as `ValueArray<T>` exists, the memory is always reachable. But maybe I did not understand what you mean.

Could you please post a snippet showing how can this possibly happen?

pgrawehr, 2022-07-14:
I have seen occasional failures of the associated test cases (e.g. Count being 0 after clearly adding an element). Could this be related?

raffaeler, 2022-07-14:
> I have seen occasional failures of the associated test cases (e.g. Count being 0 after clearly adding an element). Could this be related?

Could you please link here the code that is randomly failing?

DaZombieKiller, 2022-07-14:
@raffaeler
> Please consider that `T` must be a value type and `ValueArray` is a struct (whose members are all value types), so the GC is not involved here.

This is not true. A `struct` can contain objects and can be stored _within_ an object (as `ValueArray` is in several locations). `ValueArray` is not constrained to `unmanaged`, so there is no restriction on whether `T` can contain GC pointers or not.

> Given that `Span<T>` never takes ownership of the memory but just "view" it, and since `T` is a value type, as long as `ValueArray<T>` exists, the memory is always reachable. But maybe I did not understand what you mean.

A `Span<T>` might not _own_ the memory it references, but it is supposed to keep it alive so that the GC doesn't collect it. When you convert a GC pointer (`ref`) to an unmanaged pointer (`T*`), the GC is no longer able to "see" the pointer, and thus it is no longer kept alive.

DaZombieKiller, 2022-07-14:
> Could you please post a snippet showing how can this possibly happen?

Run this sample under .NET Framework in release configuration:
```cs
using System;
using System.Runtime.CompilerServices;
using System.Runtime.InteropServices;

unsafe
{
    var obj1 = new Class();
    var span = obj1.Array.AsSpan();
    span[0]  = 500;
    GC.Collect(GC.MaxGeneration, GCCollectionMode.Forced, true, true);
    GC.WaitForPendingFinalizers();
    Console.WriteLine(span[0]);
}

sealed class Class
{
    public ValueArray<int> Array;
    ~Class() { Array = default; Console.WriteLine("~Class"); }
}

[StructLayout(LayoutKind.Sequential)]
struct ValueArray<T>
    where T : struct
{
    T _item0, _item1, _item2, _item3, _item4, _item5, _item6, _item7;
    public unsafe Span<T> AsSpan() => new Span<T>(Unsafe.AsPointer(ref _item0), 8);
}
```
You might expect the output of this sample to be:
```
500
~Class
```
But it will actually be:
```
~Class
<some garbage value>
```
This is because the `Span<T>` is not tracking the object, so it was reclaimed by the garbage collector. As a result, the code is reading from memory that it should not be accessing.

PJB3005, 2022-07-14:
To be clear here: there are two separate issues at play.

On .NET Framework, Span uses a "slower" implementation that needs to keep track of an `object` to keep the object live. This is not the case on modern runtimes where the GC allows it to track a `ref` directly. This is why creating the span like this can cause the object to be prematurely GC'd as the above code shows, and also why there's no safe way to do this in NS2.0.

The usage of `Unsafe.AsPointer` however is still a problem on modern runtimes. If the GC runs the split moment after the call to `AsPointer` but before the span is constructed, the GC could move the backing object without updating the unmanaged pointer (depending on how aggressive the JIT is with tracking locals this might also result in "object freed prematurely" though).

Also if you use this value array in a local, you can return the span up the stack and the language won't tell you, allowing you to access garbage memory.

raffaeler, 2022-07-14:
@DaZombieKiller 
> This is not true. A `struct` can contain objects and can be stored _within_ an object (as `ValueArray` is in several locations). `ValueArray` is not constrained to `unmanaged`, so there is no restriction on whether `T` can contain GC pointers or not.

Wait, let's just talk about value types because all the other comments are related to the GC where I agree there could be safety issues.

While generally speaking a `struct` can contain a reference type field, I was talking about [`ValueArray<T>`](https://github.com/dotnet/iot/blob/1e0f186df8ee1c9f8fca44446bb6426f972d4ee3/src/devices/Common/Iot/Device/Common/ValueArray.cs) which only contains value type fields and `T` is constrained to value types (`where T : struct`). BTW C# does **not** allow unmanaged types to be used when the struct constraint is applied.
```csharp
public struct ValueArray<T> : IList<T>,
        ICollection<T>, IEnumerable<T>, IReadOnlyCollection<T>,
        IReadOnlyList<T>
        where T : struct
```

PJB3005, 2022-07-14:
The GC troubles are not because the array itself contains reference type fields (which btw it still can since you have a `struct` constraint, not an `unmanaged` one). The problem is that you need to take a reference *to* the value array, which itself is on the heap in some cases.

DaZombieKiller, 2022-07-14:
> BTW C# does not allow unmanaged types to be used when the struct constraint is applied.

An `unmanaged` type is merely a `struct` that doesn't contain any GC pointers. I assume you meant that it does not allow _managed_ types? This is also not true though, as while it disallows using a `class` directly for `T`, you can still use a `struct` that _contains_ a `class` as one of its fields.

So, as a result, `ValueArray<T>` can indirectly contain a GC pointer if `T` does, because it is constrained to `struct` and not `unmanaged`. Either way though, the GC issues still apply.

pgrawehr, 2022-07-14:
@raffaeler I don't recall exactly, but it was one of the tests in `src/devices/Common/tests/ValueArrayTests.cs` that occasionally fails, even though there's clearly no timing or otherwise thread-related stuff involved in those tests that could make them flaky.

raffaeler, 2022-07-14:
@pgrawehr Since I seriously doubt this class is used outside the binding you created, in this issue In my first comments to this thread I was trying to understand the impact of this. At the beginning it looked like (to me) a very corner case, but I understand that it is instead a (still) dangerous bug.

It must be definitely fixed and what you are reporting about the test may definitely fall in that as the GC is non-deterministic.
I believe you can proceed in fixing it as you proposed during the triage earlier today.

pgrawehr, 2022-07-14:
@DaZombieKiller Since the comments above mention multiple problems, I want to make sure I understand it correctly: If we did not have the netstandard2.0 constraint, the fix you proposed in the initial issue 

```csharp
public Span<T> AsSpan()
{
    return MemoryMarshal.CreateSpan(ref _e0, _count);
}
```
would be all that's necessary? Or would there still be more changes required? We discussed this issue in triage today and one possible option is to remove the method from the netstandard2.0 API. (replacing it with something more compatible, but maybe slower for the few users requiring a netstandard2.0 compliant library)

DaZombieKiller, 2022-07-14:
@pgrawehr That would be sufficient to make the API safe and plug the GC holes for the existing uses in the dotnet/iot libraries (as far as I can tell anyway, I have not reviewed it thoroughly.)

However, the API still cannot be made entirely safe without the upcoming [`[RefEscapes]`](https://github.com/dotnet/runtime/issues/72074) attribute in .NET 7, or the `unscoped` keyword slated for C# 12. The lack of a `[RefEscapes]`/`unscoped` feature in the past is likely one reason why types similar to `ValueArray` in the BCL are all private (such as `ValueStringBuilder`), as the compiler cannot properly track lifetimes associated with them to ensure safety.

Here's an example of a usage that is still unsafe even when using `MemoryMarshal.CreateSpan`:
```cs
Span<int> SomeMethod()
{
    var array = new ValueArray<int>();
    // do some processing on the array
    
    // The span being returned here will be referencing memory that belongs to
    // this method's stack space, and will be invalid as soon as we have returned.
    // [RefEscapes] would make this scenario an error, preventing the issue.
    return array.AsSpan();
}
```

On .NET Standard 2.0 the implementation could use a normal managed array internally, which would be safe.

raffaeler, 2022-07-15:
@pgrawehr IMO it would be better to avoid trying to make this class work for any generic case. Let's focus instead to the typical usage inside our bindings. How often a new array is allocated for the typical binding?
If the array can be maintained allocated for a long time, using `Memory<T>` (which owns the memory) should not be an issue.
If instead the memory is continuously re-allocated, `MemoryPool`/`ArrayPool` renting facilities should be enough to minimize the amount of GC work.

PR #1890 (2022-07-29): Fix a possible GC hole
Fixes #1886 (partially) 

Will need another update when the
new [RefEscapes] attribute is available, to make this
finally really safe.

Note: This removes the public AsSpan() method for
netstandard2.0, because it cannot be made safe at all there.

CC: @DaZombieKiller, @raffaeler, @joperezr 

###### Microsoft Reviewers: [Open in CodeFlow](https://portal.fabricbot.ms/api/codeflow?pullrequest=https://github.com/dotnet/iot/pull/1890)
closed 2022-07-29 by joperezr