dotnet/iot #1896: Mcp3008 Running on Raspberry PI 3b+ Returns the same value when queried in a spin loop

Growney, 2022-07-24:
I am attempting to read the value of a Mcp3008 on my raspberry pi 3b + looking to achieve a sample rate of 1khz on each of the channels. However when I attempt to read the value from the ADC it returns the initial value then never changes even when the input changes. Only if I add a delay between reads does it change but adding a delay will not allow me to achieve my target sample rate.

**Steps to reproduce**

Perform a read on a mcp3008 in a loop without a delay.

```csharp
var spiDevice = SpiDevice.Create(new SpiConnectionSettings(0, 0));
var adc = new Mcp3008(spiDevice);

while (true)
{
    int value = adc.Read(0);
    Console.WriteLine(value);
}
```

**Expected behavior**

Value should update with each read or at least after a couple of reads

**Actual behavior**

Value stays the same as the first value used.

**Versions used**

Build Machine
.NET SDK (reflecting any global.json):
 Version:   6.0.400-preview.22301.10
 Commit:    25580ffe7a

Runtime Environment:
 OS Name:     Windows
 OS Version:  10.0.22000
 OS Platform: Windows
 RID:         win10-x64
 Base Path:   C:\Program Files\dotnet\sdk\6.0.400-preview.22301.10\

global.json file:
  Not found

Host:
  Version:      6.0.7
  Architecture: x64
  Commit:       0ec02c8c96

Run Machine

.NET SDK (reflecting any global.json):
 Version:   6.0.302
 Commit:    c857713418

Runtime Environment:
 OS Name:     debian
 OS Version:  11
 OS Platform: Linux
 RID:         debian.11-arm64
 Base Path:   /opt/dotnet/sdk/6.0.302/

global.json file:
  Not found

Host:
  Version:      6.0.7
  Architecture: arm64
  Commit:       0ec02c8c96

System.Device.Gpio V - 2.1.0
Iot.Device.Bindings V - 2.1.0

Growney, 2022-07-26:
I managed to solve my issue by changing the clock frequency and data length to better match my device. I can take from this that either the device was returning the same value because the connection settings were wrong or there is still an issue in the library.

joperezr, 2022-07-28:
[Triage]: This seems to be by design from the chip itself. If you take a [look at the specs](https://cdn-shop.adafruit.com/datasheets/MCP3008.pdf) You'll see that the maximum sampling rates are:
- 200 ksps max at 5V
- 75 ksps max at 2.7V

Let us know if that didn't answer your question, but since this seems to be by design, I will go ahead and close the issue for now. Feel free to reply or reopen if you think otherwise.

closed 2022-07-28 by joperezr