dotnet/iot #1913: Add support apt-get install dotnet-iot

krwq, 2022-08-25:
We should create apt package for dotnet iot installation which brings all needed dependencies (i.e. libgpiod).

Currently we can install dotnet with apt (https://ubuntu.com/blog/install-dotnet-on-ubuntu). I think it's worth extending this to .NET IoT as well.

Ref: https://github.com/dotnet/iot/issues/1758

cc: @pgrawehr @joperezr @Ellerbach @raffaeler @richlander

pgrawehr, 2022-08-26:
Good idea. However, IIRC, apt installation is only supported on x86/x64, and not on arm/arm64.

raffaeler, 2022-08-26:
According to @richlander (https://devblogs.microsoft.com/dotnet/dotnet-6-is-now-in-ubuntu-2204/) arm64 builds are coming.

For the Raspbian OS, I understand that for the moment is just a [nice to have](https://devblogs.microsoft.com/dotnet/dotnet-6-is-now-in-ubuntu-2204/#comment-16339).

I doubt the dotnet team is willing to publish the arm32 builds even if in the IoT space they are precious because of the memory constraints (such as RPi Zero 2).

raffaeler, 2022-08-26:
> Good idea. However, IIRC, apt installation is only supported on x86/x64, and not on arm/arm64.

Yes, as I wrote, not now but they will come, the real question is when :-)

richlander, 2022-11-26:
That's a question for the Canonical folks. The .NET packages are theirs. They are still busy with getting .NET 7 and Arm64 packages in place so any expansion will take more time. We haven't discussed Arm32.

krwq, 2025-07-31:
[Triage] all it would do is install libgpiod which doesn't seem worth it