dotnet/iot #1920: [API Proposal]: Adding GpioPin to GpioController in .NET IoT

Ellerbach, 2022-09-06:


pgrawehr, 2022-09-11:
One observation that just jumped on me: If `GpioPin` is _not_ sealed, it's methods should be abstract or virtual, otherwise it is not really extensible.

pgrawehr, 2022-10-10:
After the above comments, I think the proposal should be cleaned up as follows:

```diff
namespace System.Device.Gpio;

public class GpioController : IDisposable
{
// Changing current OpenPin overloads to return a GpioPin instead of void. Binary breaking, non source breaking.
-   public void OpenPin(int pinNumber) { }
+   public GpioPin OpenPin(int pinNumber) { }

-   public void OpenPin(int pinNumber, PinMode mode) { }
+   public GpioPin OpenPin(int pinNumber, PinMode mode) { }

-   public void OpenPin(int pinNumber, PinMode mode, PinValue initialValue) { }
+   public GpioPin OpenPin(int pinNumber, PinMode mode, PinValue initialValue) { }
}

+public class GpioPin : IDisposable
+{
+    public GpioPin(int pinNumber, GpioController controller) {};
+    public virtual int PinNumber { get; }
+    public virtual PinMode GetPinMode() { }
+    public virtual bool IsPinModeSupported(PinMode pinMode) { }
+    public virtual void SetPinMode(PinMode value) { }
+    public virtual PinValue Read() { }
+    public virtual void Write(PinValue value) { }
+    public event PinChangeEventHandler ValueChanged;
+    protected virtual void Dispose(bool disposing) {};
+    public void Dispose() {};
+}
```

Open Questions: 
### Should `GpioPin` be Disposable? 
Pro: 
- Can just use an using statement around an OpenPin call
- Owners of the GpioPin class (e.g. bindings) can clean up after themselves
Con:
- The pin is still owned by the GpioController, making `GpioPin` disposable appears to move ownership

### Should the ctor take `GpioController` or `GpioDriver`?
The later may allow for some performance optimization, while it allows bypassing the checks of the controller. Bypassing the GpioController could allow for dangling `GpioPin` instances (instances pointing to closed pins or pins whose designation has changed).  

### Should the class be sealed instead?
This would slightly increase performance, but obviously not allow hardware-dependent implementations.

### Does the ctor need to be public?
I believe so, since `GpioController` (who will normally create instances of this type) is public and inheritable.

### Do we need the `Toggle()` method?
Depending on the hardware, implementing it in a thread-safe fashion might be difficult or even impossible. So rather not present a method that may or may not work.

### What's the risk in the binary breaking change from `void OpenPin()` to `GpioPin OpenPin()`?
What can possibly go wrong?

raffaeler, 2022-10-10:
I agree with @pgrawehr. Let me add a couple of notes:
1. I prefer the binary breaking change on `OpenPin`. Returning `GpioPin` gives more sense to the proposal, increase the usability and IMO resolve the doubts around the ownership of the GpioPin.
2. `Toggle`. Reading the state and toggling manually could make the transition longer than expected. This method could use the `GC.TryStartNoGCRegion` to minimize the delays.
3. I would love to see the `Pulse` method as well: `void Pulse(int microseconds)` which also could use `GC.TryStartNoGCRegion`.

Not sure if I will be able to join the call, sorry.

krwq, 2022-10-10:
For 1, 100% agree - for 2/3 I'd only consider that if there is a hardware support - for Toggle I believe we can support it at least on some drivers I'm not sure about pulses

raffaeler, 2022-10-10:
@krwq I understand your point but given the library is an abstraction over many vendors, it is not likely to see the same support for all of them.
I believe we can offer an "optimistic" implementation where the hardware is used where available or a software implementation where it is not.
It is roughly simiilar to the current support for pull-ups / pull-downs. Not all the Gpio support them, but the library provides the option in any case (even if there is no software equivalent).

joperezr, 2022-10-10:
Here are my few notes on the open questions from above:

> Should GpioPin be Disposable?

I could be convinced otherwise, but I wouldn't mark it as disposable. Disposable types either hold native resources, or have members that do which need to be disposed of. GpioPin's case doesn't apply to any of those, as the native resources are actually hold by the driver, and GpioPin doesn't have any other members that need to be disposed of. As for the comment regarding cleaning up after use, device bindings should just call Close() instead.

> Should the ctor take GpioController or GpioDriver?
> Does the ctor need to be public?

My answer here would be: no, the constructor for Gpio Pin should only be internal, and the only way to get a GpioPin back should be via calling OpenPin in a controller. If you want a GpioPin dervied type back, then the way to do it is via creating a controller which has a specific custom driver passed in, which returns this custom gpiopin back. For that, this proposal is also missing changing the driver's open pin to also return GpioPin so that derived drivers can return derived GpioPins.

> Should the class be sealed instead?

Unless we have a very specific use in mind,  I'd start with making it sealed, and then in the future we can unseal it if we need to. We can't go the other way around.

> Do we need the Toggle() method?

I would prefer not to. We have a Write method, and Toggle is a method that can be very easily added on the consuming code as a helper. In any case, I think this decision doesn't need to be made now, so I'd rather not to include it as part of this proposal, and then add it later as a subsequent proposal if we see it provides enough value.

> What's the risk in the binary breaking change from void OpenPin() to GpioPin OpenPin()?

Risk is pretty high. The main issue will be that this is a binary breaking change and it will very likely break anyone in the library case. By this I mean: Imagine a own a robotic company which has a robot, and I wanted to ship a nuget package with a binding for my robot that internally uses System.Device.Gpio. I will depend on one version of that, for example, in version 2.2. If a consuming app then wants to use my robot nuget package, they won't be able to use our 3.0 System.Device.Gpio package as well, since if they do, it would break the robot package. In other words, any code that depends on System.Device.Gpio today would be binary roken after this change, and would be forced to recompile.

Ellerbach, 2022-10-11:
> Unless we have a very specific use in mind, I'd start with making it sealed, and then in the future we can unseal it if we need to. We can't go the other way around.

I'm always more in favor of not sealing. But we can for sure.

> Toggle. Reading the state and toggling manually could make the transition longer than expected. This method could use the GC.TryStartNoGCRegion to minimize the delays

Toggle is actually available natively on many hardware. Most extenders like the FT ones have this capability. Also in MCU it's always the case making the change of a pin really fast. Now, that's as well a reason we should not seal the implementation to allow faster usage.

> I would love to see the Pulse method as well: void Pulse(int microseconds) which also could use GC.TryStartNoGCRegion

Like @krwq, I'm not super in favor of the pulse one. It will always be very dependent of the driver to get a maximum performance. Something that can be added later. Also the pulse one is most of the time a PWM, so I would recommend then to go for a PWM pin instead at least it makes things clear.

For the rest aligned with the comments from @joperezr.

josesimoes, 2022-10-11:
My take on this:

1. _Toggle_: I keep thinking this is very useful and it should be added. For platforms that don't support it natively the best course of action would be to perform a read and toggle in the low level layer.
Those few lines of code wont' have any noticeable impact on overall performance. And for sure that will much faster than performing the usual approach of read in C#, toggle it and write to the GPIO. 
I don't understand why this should be a problem regarding thread safety execution.
One last motivation: with this, we'll be closer to have API compatibility between IoT and [.NET nanoFramework](https://docs.nanoframework.net/api/System.Device.Gpio.GpioPin.html#System_Device_Gpio_GpioPin_Toggle). 😉 

2. _OpenPin()_: sometimes breaking changes are required and have to be assumed. In this case I think having `OpenPin()` returning a `GpioPin` would be preferable. It not only makes the API more "object oriented" like (as it should be, being C#) but it also makes it clear who owns the `GpioPin` and it makes it soooo much easier in code to be able to take that `GpioPin` object after it's being created and do stuff with it. As opposed to have to, at a later stage, have to go and having to grab it from some other place. Plus (you'll forgive me on keeping insisting on this 😅), it would be in line with .NET nanoFramework: [OpenPin()](https://docs.nanoframework.net/api/System.Device.Gpio.GpioController.html#System_Device_Gpio_GpioController_OpenPin_System_Int32_) and following.

3. _Pulse()_: I don't think this belongs here... This is more in the realm of PWM or timers. If the GPIO API is bringing _time_ into play, that would be much useful implementing a property to handle debounce at driver level. Like there is in .NET nanoFramework: [DebounceTimeout](https://docs.nanoframework.net/api/System.Device.Gpio.GpioPin.html#System_Device_Gpio_GpioPin_DebounceTimeout).

raffaeler, 2022-10-11:
> Risk is pretty high.

@joperezr got it. If the breaking change is unavoidable, I would rather prefer to introduce a new, let's say, `GpioController2` satisfying the new contract. BTW I don't like the `2` appended to the end, it is just an example.

@Ellerbach : About `Pulse`, I didn't even think to pwm. There are many times where ICs require a pulse to enable features, reset, etc.. The timings for the pulse are critical and this is why we can provide an appropriate implementation.

With regards to `Toggle` and `Pulse`, I am trying to see it from the opposite perspective. We have the chance to write the best possible implementation (eventually with hw support or suspended GC timings). On the other end the user risks to write down a lot of useless and error-prone code. Let's think about it.

bartonjs, 2022-11-01:
* The binary breaking change seems to be mitigated by a fairly tight coupling to higher level types that ship out of the same repository, and low usage of these primitive types directly.
* Based on discussion it seems like the controller needs a Toggle method as well  (or TogglePin, whatever you think is best)
* During discussion it was stated that GpioPin was going to reach through straight to the driver, potentially bypassing any overridden methods in the controller class.  This may be too much of a surprise.
  * It was mentioned that the straight-to-the-driver behavior could be conditional on whether this.GetType() == typeof(GpioController).

```diff
namespace System.Device.Gpio;

public class GpioController : IDisposable
{
// Changing current OpenPin overloads to return a GpioPin instead of void. Binary breaking, non source breaking.
-   public void OpenPin(int pinNumber) { }
+   public GpioPin OpenPin(int pinNumber) { }

-   public void OpenPin(int pinNumber, PinMode mode) { }
+   public GpioPin OpenPin(int pinNumber, PinMode mode) { }

-   public void OpenPin(int pinNumber, PinMode mode, PinValue initialValue) { }
+   public GpioPin OpenPin(int pinNumber, PinMode mode, PinValue initialValue) { }

+   public void Toggle(int pinNumber) { }
}

+public sealed class GpioPin
+{
+    public int PinNumber { get; }
+    public PinMode GetPinMode() { }
+    public bool IsPinModeSupported(PinMode pinMode) { }
+    public void SetPinMode(PinMode value) { }
+    public PinValue Read() { }
+    public void Write(PinValue value) { }
+    public event PinChangeEventHandler ValueChanged;
+    public void Toggle() { }
+}
```

Ellerbach, 2022-12-14:
Closing this issue as the implementation has been done and merged.

closed 2022-12-14 by Ellerbach