dotnet/iot #1928: LibGpiodDriverEventHandler.SubscribeForEvent calls GetIOException incorrectly

jdbruner, 2022-09-20:
**Describe the bug**

On failure, LibGpiodDriverEventHandler.SubscribeForEvent throws an exception as follows:

```csharp
throw ExceptionHelper.GetIOException(ExceptionResource.RequestEventError, _pinNumber, Marshal.GetLastWin32Error());
```

However, GetIOException is declared as follows:

```csharp
public static IOException GetIOException(ExceptionResource resource, int errorCode = -1, int pin = -1) 
```

The order of the second and third arguments is reversed, resulting in a confusing exception message.

**Steps to reproduce**

This can be verified by inspection of the source code, but if you want a repro procedure, then subscribe to events on a pin that is already in use.

**Expected behavior**

Throw an exception indicating the pin number and error 16 (EBUSY).

**Actual behavior**

Throw an exception indicating pin number 16 and error number equal to the pin number.

**Versions used**

System.Device.Gpio v2.1.0

jdbruner, 2022-09-24:
I would be (the fix is trivial), but I'm not set up to rebuild it on my machine before submitting the PR.

--John

@jdbruner<https://nam12.safelinks.protection.outlook.com/?url=https%3A%2F%2Fgithub.com%2Fjdbruner&data=05%7C01%7C%7C43c4d229419d43fd54ad08da9cb1a35a%7C84df9e7fe9f640afb435aaaaaaaaaaaa%7C1%7C0%7C637994584020571376%7CUnknown%7CTWFpbGZsb3d8eyJWIjoiMC4wLjAwMDAiLCJQIjoiV2luMzIiLCJBTiI6Ik1haWwiLCJXVCI6Mn0%3D%7C3000%7C%7C%7C&sdata=BFgH9AmDyNbUsNEjLFAkB3DGc%2FZHPlSSEV1ENA%2BrVFQ%3D&reserved=0> are you interested in offering a fix?

pgrawehr, 2022-09-24:
@jdbruner Just run the `build.cmd` file in the repository root. I think you need no prerequisites for this.

PR #1932 (2022-09-29): fix call to GetIOException (issue 1928)
Fixes #1928
As described in the issue, in ```LibGpiodDriverEventHandler.SubscribeForEvent``` the last two arguments to ```GetIOException``` were reversed.

###### Microsoft Reviewers: [Open in CodeFlow](https://portal.fabricbot.ms/api/codeflow?pullrequest=https://github.com/dotnet/iot/pull/1932)
closed 2022-09-29 by joperezr