dotnet/iot #1935: Windows IoT support

raffaeler, 2022-09-29:
Apparently, most of our users in this library are using Linux distributions and not Windows IoT.

Should we continue to support Windows IoT for the dotnet IoT libraries or would it be better to simplify the library and just focus on Linux support?

Any user from the community that is using Windows IoT is kindly invited to comment explaining the scenarios where they use the libraries from this repository on Windows.

Thank you

pgrawehr, 2022-09-30:
To make it clear: We're not intending to remove support for Windows desktop versions in general. This is just about removing the support for Windows Iot Core Arm32, running on Raspberry Pi 3b. 

We do look for someone to implement GPIO/PWM/I2C drivers for Windows 11 on Arm64, see #1640 and others.
Windows is perfectly fine for running Iot applications using external interfaces, e.g. FT4222 or Arduino.

krwq, 2025-07-31:
[Triage] We already dropped supported for Windows IoT

closed 2025-07-31 by krwq