dotnet/iot #1940: Master build is broken after Version updates

pgrawehr, 2022-10-02:
**Describe the bug**

Running `build -pack` on current master results in these errors:

```
C:\projects\iot\.dotnet\sdk\7.0.100-rc.1.22431.12\Sdks\NuGet.Build.Tasks.Pack\build\NuGet.Build.Tasks.Pack.targets(203,
6): error : Für dieses Projekt kann kein Paket erstellt werden, weil die Paketerstellung deaktiviert wurde. Fügen Sie d
er Projektdatei <IsPackable>true</IsPackable> hinzu, um die Erstellung eines Pakets aus diesem Projekt zu ermöglichen.
[C:\projects\iot\src\devices\Media\VideoDevice\samples\MjpegStream\MjpegStream.csproj]
C:\projects\iot\.dotnet\sdk\7.0.100-rc.1.22431.12\Sdks\NuGet.Build.Tasks.Pack\build\NuGet.Build.Tasks.Pack.targets(221,
5): error NU5104: Eine stabile Version eines Pakets darf keine Abhängigkeit mit einem Vorabrelease aufweisen. Ändern Si
e entweder die Versionsspezifikation der abhängigen Komponente "System.Device.Gpio [2.3.0-dev, )", oder aktualisieren S
ie das Versionsfeld in der NUSPEC-Datei. [C:\projects\iot\tools\ArduinoCsCompiler\samples\WeatherStation\WeatherStation
.csproj]
C:\projects\iot\.dotnet\sdk\7.0.100-rc.1.22431.12\Sdks\NuGet.Build.Tasks.Pack\build\NuGet.Build.Tasks.Pack.targets(221,
5): error NU5104: Eine stabile Version eines Pakets darf keine Abhängigkeit mit einem Vorabrelease aufweisen. Ändern Si
e entweder die Versionsspezifikation der abhängigen Komponente "System.Device.Gpio [2.3.0-dev, )", oder aktualisieren S
ie das Versionsfeld in der NUSPEC-Datei. [C:\projects\iot\tools\ArduinoCsCompiler\ArduinoCsCompiler.csproj]
C:\projects\iot\.dotnet\sdk\7.0.100-rc.1.22431.12\Sdks\NuGet.Build.Tasks.Pack\build\NuGet.Build.Tasks.Pack.targets(221,
5): error NU5104: Eine stabile Version eines Pakets darf keine Abhängigkeit mit einem Vorabrelease aufweisen. Ändern Si
e entweder die Versionsspezifikation der abhängigen Komponente "System.Device.Gpio [2.3.0-dev, )", oder aktualisieren S
ie das Versionsfeld in der NUSPEC-Datei. [C:\projects\iot\tools\ArduinoCsCompiler\samples\BlinkingLed\BlinkingLed.cspro
j]
```

As a follow-up, `build -pack /p:GenerateCompatibilitySuppressionFile=true` doesn't work and I can't update the suppression file (there are some expected breakages in a pending PR)

**Steps to reproduce**

Run
```
git clean -dfx
build -pack
```

**Expected behavior**
No errors

**Actual behavior**
Build fails
**Versions used**
Main branch from 2022-09-30, revision d7f6c14826ab4aa43066f66d788c864db28507a0

CC: @joperezr

joperezr, 2022-10-03:
You are right, thanks for logging this @pgrawehr. Seems like a problem with my recent changes to make the build faster. I'm working on a fix and will push a PR in a sec that should fix it.

joperezr, 2022-10-04:
FWIW, the reason why this wasn't broken in CI and builds still worked after that change, was because our CI and official builds don't use the `-pack` argument, instead call `build.cmd /p:BuildPackages=true`. In any case, after the changes here, things should start to work again when using the `-pack` argument.

closed 2022-10-04 by joperezr