dotnet/iot #1958: How can I streaming raw data from Raspberry pi to a server and view it.

wangyiwu, 2022-10-16:
I want to use GRPC to streaming data from Raspberry to another server and using that data to view it. How can I get raw data from Raspberry pi camera.

raffaeler, 2022-10-16:
Hi @wangyiwu this is a topic where I spent a lot of time.
The camera output depends on the raspbian version you are using:
- the "legacy" is what we used for years  and is based on a proprietary code which was used from the Raspberry PI foundation to create apps like raspivid and raspistill. These utilities provide the support to the embedded hardware compression in H264 compression.
- the [libcamera](https://libcamera.org/) support was added recently as the [default for Raspbpian](https://www.raspberrypi.com/documentation/accessories/camera.html). It provides access to the camera stream but **not** to the hardware compression which must be controlled separately with user's code. Libcamera organization has no plan to provide hardware comrpession support in their code. More details on libcamera [here](https://github.com/dotnet/iot/issues/1875)

If `libcamera` is active, you cannot use the old tools like 'raspivid'.  This means you have to use `raspi-config` to turn on the legacy support if you want to use `raspivid` to get the stream with a very low amount of code.

These are the steps I did to get the stream from C#:
- I run raspivid with the `Process` class and capture the output binary stream. Provided you pass the appropriate arguments to the process, you will receive raw H.264 binary stream
- I read the stdout from C# and analyze the buffer of the stream in order to isolate the frames. This is needed because the javascript H264 decoders are not able to do it and can't know where each frame starts. This step requires you read the H264 specs and decide which profile you want to adopt. The algorithm and signature depends on the used profile.
- You send the stream in some way (see below)
- On the javascript side, use a standard library like: https://www.npmjs.com/package/h264-live-player

Frankly I discourage you from using GRPC. You don't need "RPC-like" calls.
Instead, I used Websocket which is extremely efficient and has one big advantage: each websocket has two separate channel: one for text, the other for binary.
This means that you can send video data on the binary stream while sending metadata on the text channel at the same time.
The result is very efficient.

You can watch me presenting this stuff at dotnext conference in St Petersburg in 2019: https://www.youtube.com/watch?v=Bn475hRbq7c
The session goes far beyond that because I also use OpenCV from C# for facial recognition to center the face in the video by moving the camera with a couple of servo motors.

I already enhanced the code and I have more plans, but I will need some time before being able to publish the code.

wangyiwu, 2022-10-16:
@raffaeler Thanks for your answer.

raffaeler, 2022-10-16:
You are welcome @wangyiwu.
Feel free to reopen this issue if you have any additional questions.

closed 2022-10-16 by raffaeler

wangyiwu, 2022-10-30:
Hi @raffaeler, Now I'm using MQTTNET to build a broker then My .net App which I run on Raspberry PI push data to. After that I using an other .net app to subscribe the brober to get byte array data. Is that the best way to get picture data in Raspberry PI?. I dont see the source code in your conference, can you share it?

raffaeler, 2022-10-30:
Hi @wangyiwu the MQTT protocol was created to support disconnected scenarios, not for streaming purposes which implicitly requires an active connection.

You may want to use MQTT to push data coming from sensors (temperature, humidity or even single shots from the camera). This way, the data does not get lost and will reach the other end (one or multiple endpoints) even after several days.

Streaming video is a different story because you want to see what's happening in real-time. This requires an active connection and this was the reason why I used the Websocket protocol.
Websocket is a TCP connection negotiated starting from an initial HTTP request. This protocol is very cheap in terms of bandwith usage and allows transferring large amounts of data with the lowest possible impact on the transmitter and the receiver.

With regards to my source code, I need to work on the codebase before publishing it. I will do as soon as I have some time, it won't be very soon as I am currently very busy in some projects.

If you have other questions, feel free to write it down here.

wangyiwu, 2022-10-31:
I dont want dirrect access to my Raspberry PI, I want send data to an server which have a good performance to do some stuff like Machine learning and some handle before save it to database. I'm planning to build a server using Signalr and stream it to the Frontend.

raffaeler, 2022-10-31:
@wangyiwu In any case, MQTT was born to enqueue relatively small messages to a server, not to send videos or streaming.
You also mention saving the video on a database. Be careful that databases are not meant for large blob storages. If you want to save videos, save them on the filesystem and only persist metadata on the database.

If you want to offload the video recognition to a different machine, I personally suggest to stream the video content from the RPi to the server. This is done as follows:
- Create on the RPi an ASP.NET application with either Websocket server or SignalR
- The other server will act as a client and attach the live streaming in order to process the images
Depending on the code that you will use to recognize the images, you may need to recognize the H.264 frames start/end in the streaming. I had to do it, but this depends on the libraries you use.

With regards to Websockets/SignalR. If the Raspberry PI streaming will not be accessible from a browser, I strongly suggest to use Websockets instead of SignalR. The overhead may be significant in terms of performance. The ASP.NET Websocket support works very well and is not hard to set up.

HTH

wangyiwu, 2022-10-31:
Thanks @raffaeler for your suguestion!

raffaeler, 2022-11-22:
Hi @wangyiwu no, it is not.
The reason is that libcamera is C++ only and unfortunately C++ does not have a stable ABI over compilers and versions.

In order to make it work, we should compile libcamera in this repository and this is not good for this repository because it would require compiling multiple System.Devices.xxx assemblies for each platform and cpu architecture.

Alternatively, libcamera should (and they probably will) create a C wrapper. C has a stable ABI and we could go on with simple PInvokes to the C wrapper. I was said by a libcamera developer that they will probably create the C wrapper, but there is no defined deadline.

Also, take in account that libcamera is **not** going to give you the hardware compressed H.264 stream. The previous stack in the raspberry PI is instead providing the hardware compressed H.264 stream. Without compression you will obtain a very large stream in terms of network bandwidth.

My suggestion is to stick with the "legacy" streaming support until libcamera will hopefully provide a C wrapper. (To be clear, this is not just a dotnet issue, any language and framework has the same problem).

wangyiwu, 2022-11-22:
Thanks for the answer.  Currently Im still using lagacy to take picture, then send it to MQTT, then I do some middleware pipeline to process the image, It still working fine.

raffaeler, 2022-11-22:
Great to hear that @wangyiwu, I am happy to hear that you have a working solution.

Sometime ago, during the triage meeting, we discussed the adoption of libcamera and it is certainly a "nice to have" but their current support is poor in comparison to the legacy stack, therefore we prefer to wait for the new versions, hoping that they will provide a C wrapper.

FYI, The Raspberry Foundation is providing support for the Python interop with libcamera and they do have the same problem. The huge difference is that they ship the operating system, therefore it is not a problem for them recompile libcamera and ensuring that their interoperability code with Python works correctly.
This is very different from our situation because we prefer to stay platform-neutral (pure IL code) as much as possible and avoid compiling native code.