dotnet/iot #1961: Improve Nuget Binding package description

raffaeler, 2022-10-20:
**Describe the bug**

The `Iot.Device.Bindings` description is currently the following:
```
This package provides a set of Device Bindings that use System.Device.Gpio package to communicate with a microcontroller.
``` 

This is not precise as most of the bindings provides a simpler way to communicate with many other devices. Among those, there may be microcontrollers as well.

Proposed description:
```
This package provides a set of Device Bindings that use System.Device.Gpio package to communicate with a many popular devices and ICs  such as sensors, motors, signal converters, expanders and microcontrollers.
The complete list of supported devices can be found here: https://github.com/dotnet/iot/tree/main/src/devices
```

Affected file: [iot](https://github.com/dotnet/iot/tree/14d0596c936a141bda4b10e9eee31f8955edba36)/[src](https://github.com/dotnet/iot/tree/14d0596c936a141bda4b10e9eee31f8955edba36/src)/[Iot.Device.Bindings](https://github.com/dotnet/iot/tree/14d0596c936a141bda4b10e9eee31f8955edba36/src/Iot.Device.Bindings)

krwq, 2025-01-23:
[Triage] We should consider adding README to the package so that Nuget can pick this up

Ellerbach, 2025-01-25:
this has been done:

* https://www.nuget.org/packages/System.Device.Gpio#readme-body-tab
* https://www.nuget.org/packages/Iot.Device.Bindings

closed 2025-01-25 by Ellerbach