dotnet/iot #1979: Pin mode is not set to PWM when creating a PwmChannel

MaxRev-Dev, 2022-11-09:
**Describe the bug**
After enabling channels e.g., changing `dt-overlay` in /boot/config.txt it's good to mention about explicit mode change of the pin.
Referring to a closed #1470. This should have been fixed in #1128, but it seems like it's not. PWM mode is not set when creating a `PwmChannel`. 
It's not documented (nor fixed) and not obvious what does not work exactly without debugging and time wasting.

**Steps to reproduce**

1. Run any script/command that will change PIN mode. For ex:
 `gpio -g mode 18 out`

2. Run C# app
```csharp
    // Testing with SG90 connected to BCM18 pin (Phy.12)
    using var board = Board.Create();
    using var channel = board.CreatePwmChannel(0, 0, 50, 0); // 50 Hz, 0% duty cycle 
    using var servoMotor = new ServoMotor(channel, 180, 500, 2500);
    // angles for SG90
    var values = new[] { 0, 45, 90 };
    foreach (var value in values)
    { 
        servoMotor.WriteAngle(value);
        Thread.Sleep(500);
    }
```

**Expected behavior**
After changing pin to pwm it works again. `gpio -g mode 18 pwm`
PWM mode should be enabled when channel is activated.
Currently the fix for me looks like this:
```csharp
    using (var p = Process.Start("gpio", "-g mode 18 ALT5"))
    {
        p.WaitForExit();
    }
```

**Actual behavior**
Nothing happens. I had to dig into the reference source but found nothing.

**Versions used**
2.2.0

<details>
  <summary>Dotnet Info</summary>

```
.NET SDK (reflecting any global.json):
 Version:   6.0.301
 Commit:    43f9b18481

Runtime Environment:
 OS Name:     raspbian
 OS Version:  10
 OS Platform: Linux
 RID:         linux-arm
 Base Path:   /home/user/.dotnet/sdk/6.0.301/

Host (useful for support):
  Version: 6.0.6
  Commit:  7cca709db2

.NET SDKs installed:
  6.0.301 [/home/user/.dotnet/sdk]

.NET runtimes installed:
  Microsoft.AspNetCore.App 6.0.6 [/home/user/.dotnet/shared/Microsoft.AspNetCore.App]
  Microsoft.NETCore.App 6.0.6 [/home/user/.dotnet/shared/Microsoft.NETCore.App]

To install additional .NET runtimes or SDKs:
  https://aka.ms/dotnet-download
```
</details>

- Version of `System.Device.Gpio` 2.2.0
- Version of `Iot.Device.Bindings` 2.2.0

Ellerbach, 2022-11-09:
See how to activate properly the PWM mode in a sustainable way without having to run any privilege code: https://github.com/dotnet/iot/blob/main/Documentation/raspi-pwm.md

Once setup properly, you won't have to run any command line.

pgrawehr, 2022-11-09:
I'll have to take a look, but IIRC 18 is not the default for PWM channel 0. 12 is. Can you check (with `gpio readall`) whether that pin changes the mode instead?

pgrawehr, 2022-11-09:
@Ellerbach the problem is not about permissions here, if I understand correctly,  it's about wrong pin modes.Since the ticket describes exactly what the feature is for, I'm quite sure it must be some usage or usability problem (like we miss a method overload or so)

MaxRev-Dev, 2022-11-09:
@Ellerbach this is not related to PWM activation in overlay, neither to root privileges. Docs were read several times, all is set.
@pgrawehr the behavior of GPIO 12 the same as for GPIO 18. And yes, I'm monitoring the modes via `gpio readall`. I don't think so but will take a look later when will have access to the test board. Also, this does not matter if initially pin mode was different from PWM.

pgrawehr, 2022-11-10:
@MaxRev-Dev I just had a look at the code, and I can confirm: The default pin for PWM channel 0 is 12, not 18. To use pin 18 for PWM, you need to change your line 
```csharp
 using var channel = board.CreatePwmChannel(0, 0, 50, 0); // 50 Hz, 0% duty cycle 
```
to
```csharp
using var channel = board.CreatePwmChannel(0, 0, 50, 0, 18, PinNumberingScheme.Logical); // 50 Hz, 0% duty cycle 
```
That should hopefully fix the issue.

MaxRev-Dev, 2022-11-10:
Mea culpa. Can confirm that I was missing that overload and the mode of pin changes.
One another thing. I don't know whether it was a cache or something, but PWM output sometimes is weird. It's in microseconds and frequency is floating between 150 kHz and 400 kHz. It can occur after some manipulations on pwm. Reboot helps to fix this.

closed 2022-11-10 by MaxRev-Dev