dotnet/iot #1984: Shall we deprecate netcoreapp3.1 build?

pgrawehr, 2022-11-27:
I was attempting to upgrade a few dependencies (like `Microsoft.Extensions.Logging` to V7.0.0, but this causes a lot of warnings of the form 

```
C:\Users\Patrick\.nuget\packages\microsoft.extensions.logging.abstractions\7.0.0\buildTransitive\netcoreapp2.0\Microsof
t.Extensions.Logging.Abstractions.targets(4,5): error : Microsoft.Extensions.Logging.Abstractions 7.0.0 doesn't support
 netcoreapp3.1 and has not been tested with it. Consider upgrading your TargetFramework to net6.0 or later. You may als
o set <SuppressTfmSupportBuildWarnings>true</SuppressTfmSupportBuildWarnings> in the project file to ignore this warnin
g and attempt to run in this unsupported configuration at your own risk. [C:\projects\iot3\src\devices\Nmea0183\tests\N
mea0183.Tests.csproj::TargetFramework=netcoreapp3.1]
C:\Users\Patrick\.nuget\packages\microsoft.extensions.dependencyinjection\7.0.0\buildTransitive\netcoreapp2.0\Microsoft
.Extensions.DependencyInjection.targets(4,5): error : Microsoft.Extensions.DependencyInjection 7.0.0 doesn't support ne
tcoreapp3.1 and has not been tested with it. Consider upgrading your TargetFramework to net6.0 or later. You may also s
et <SuppressTfmSupportBuildWarnings>true</SuppressTfmSupportBuildWarnings> in the project file to ignore this warning a
nd attempt to run in this unsupported configuration at your own risk. [C:\projects\iot3\src\devices\Nmea0183\tests\Nmea
0183.Tests.csproj::TargetFramework=netcoreapp3.1]
```

We can either:
- Suppress the warning
- make the dependency TFM-dependent
- drop netcoreapp 3.1 support completely

Since we would likely keep netstandard2.0 support, I would suggest the third option. I can't think of a reason why one would create a new project with net3.1, and projects still under development should be easy to upgrade.

joperezr, 2022-11-28:
netcoreapp 3.1 goes out of support very soon (December 13th) so I think it is totally fine to drop that asset from our packages and instead just have the new LTS version (net6.0) which we already target. That said, I think dropping a TFM support from the packages should be considered a breaking change that is only done when bumping the major version. I don't see why we can't do it now, but just wanted to mention that this should be planned for 3.0 whenever we are ready to ship that.

pgrawehr, 2022-11-29:
I agree to that. But since we intend to do that 3.0 step anyway after the `GpioPin` change, this would be good to do simultaneously. I have the PR almost ready.

raffaeler, 2022-11-29:
+1, next time we should discuss what exactly we want to plan for 3.0, which should happen quite soon.

Ellerbach, 2022-11-29:
> I have the PR almost ready.

Go for it, fully aligned!

PR #1990 (2023-01-06): Update Microsoft.Extensions.Logging to v7.0.0, drop netcoreap3.1 support
Fixes #1984 

Updates Microsoft.Extensions.Logging and Microsoft.Extensions.Logging.Abstractions to v7.0.0. Since they officially no longer support netcoreapp3.1, that is dropped now, as discussed in #1984. 

All tests are now by default built against .net6.0, some additionally against .net48 (to test netstandard2.0 support). 

###### Microsoft Reviewers: [Open in CodeFlow](https://portal.fabricbot.ms/api/codeflow?pullrequest=https://github.com/dotnet/iot/pull/1990)
closed 2023-01-06 by pgrawehr