dotnet/iot #1986: Consider dropping `netstandard2.0` support

richlander, 2022-11-30:
I was just working on a binding and needing to workaround the lack of Index/Range support in .NET Standard 2.0. It would be great if we could switch the repo to .NET 6+ targeting.

It seems like we have a few options:

- Status quo.
- Break out the few bindings that we know have utility on .NET Framework into a separate package (that targets .NET Standard 2.0).
- Drop support for .NET Standard 2.0 entirely, and the concrete platforms (like .NET Framework) that are affected by that.

With .NET Core 3.1 going EOL (#1984), now is a great time to have this discussion.

raffaeler, 2022-11-30:
I already asked this to @joperezr for my serial port branch. 
Also, we are already considering dropping targeting Windows. If this is confirmed, I believe that netstandard2.0 can go away as well.
@Ellerbach You don't need this for the nanoframework, right?

joperezr, 2022-11-30:
The one reason why we still want to keep NETStandard 2.0 support is in favor of Mono. There are some popular devices where .NET (Core) is not supported yet, for example, the Raspberry Pi Zero. However, some of these do support Mono, so that is where it adds a lot of value to have .NET Standard 2.0.

raffaeler, 2022-11-30:
Sorry, I forgot that, you are perfectly right (of course).

richlander, 2022-12-01:
I thought that we had a build of Mono from dotnet/runtime that worked for Armv6 that supported the modern TFMs. Is that right @marek-safar?

marek-safar, 2022-12-01:
That's correct armv6 is community supported in dotnet/runtime and there are builds (including source build) available for example for Alpine distro.

pgrawehr, 2022-12-01:
> I thought that we had a build of Mono from dotnet/runtime that worked for Armv6 that supported the modern TFMs.

That confuses me. A build from dotnet/runtime would not be a mono build, would it? That would be a "modern" .NET Core build, right?

I do think we should still support .netstandard2.0 in the upcoming 3.0 release and maybe drop that support in 4.0. See related #1984 for deprecating .netcoreapp3.1.

joperezr, 2022-12-01:
[Triage]: We discussed this during triage and agreed that probably the best course of action would be to drop netcoreapp3.1 TFM for our next release (3.0), and then look into this issue a bit more to make sure that dropping .NET Standard 2.0 won't break any important scenarios or devices that we care about. If it is decided that this TFM is in fact no longer needed, we will drop it in our next major release (4.0)

richlander, 2022-12-12:
Sounds like a plan to me.

krwq, 2023-03-09:
[Triage] We don't have any compelling reason to remove this support. Because of netstandard2.0 we can run on mono. Because of that feels like it's better to do nothing.

closed 2023-03-09 by krwq