dotnet/iot #1991: `.dotnet/x64` used on Arm64 machine

richlander, 2022-12-03:
I was just building this repo. I noticed references to `x64` in this script output. Why is that used on an Arm64 machine? That's a waste of resources since those assets cannot be used.

```bash
rich@raspberrypi:~/iot $ ./build.sh
  Determining projects to restore...
  All projects are up-to-date for restore.
Attempting to install dotnet from public_location.
dotnet-install: Note that the intended use of this script is for Continuous Integration (CI) scenarios, where:
dotnet-install: - The SDK needs to be installed without user interaction and without admin rights.
dotnet-install: - The SDK installation doesn't need to persist across multiple CI runs.
dotnet-install: To set up a development environment or to run apps, use installers rather than this script. Visit https://dotnet.microsoft.com/download to get the installer.
dotnet-install: .NET Core Runtime with version '3.1.21' is already installed.
Attempting to install dotnet from public_location.
dotnet-install: Note that the intended use of this script is for Continuous Integration (CI) scenarios, where:
dotnet-install: - The SDK needs to be installed without user interaction and without admin rights.
dotnet-install: - The SDK installation doesn't need to persist across multiple CI runs.
dotnet-install: To set up a development environment or to run apps, use installers rather than this script. Visit https://dotnet.microsoft.com/download to get the installer.
dotnet-install: Attempting to download using primary link https://dotnetcli.azureedge.net/dotnet/Runtime/6.0.11/dotnet-runtime-6.0.11-linux-x64.tar.gz
dotnet-install: Extracting zip from https://dotnetcli.azureedge.net/dotnet/Runtime/6.0.11/dotnet-runtime-6.0.11-linux-x64.tar.gz
dotnet-install: Adding to current process PATH: `/home/rich/iot/.dotnet/x64`. Note: This change will be visible only when sourcing script.
```

Sharing what should be obvious:

```bash
~/iot/.dotnet/x64 $ ./dotnet
-bash: ./dotnet: cannot execute binary file: Exec format error
rich@raspberrypi:~/iot/.dotnet/x64 $ ls
dotnet  dotnet-install.sh  host  LICENSE.txt  shared  ThirdPartyNotices.txt
```

pgrawehr, 2022-12-04:
No, we're currently not running any integration tests on Arm64 (and actually no code at all). The helix is still using Arm32, and our linux build servers are x64. 

We're currently dropping netcoreapp3.1 support, so this problem will magically go away ;-)

I think I haven't tested building on Arm64 (since previously, also the Arm32 build didn't work). I always use crosscompilation to build for the Raspberry Pi, using something like `dotnet publish <myproject.csproj> --runtime linux-arm64`. That works fine.

joperezr, 2022-12-05:
> We're currently dropping netcoreapp3.1 support, so this problem will magically go away ;-)

I don't think so. What @richlander is saying is that building the repo on Arm64 is not working, and I think removing netcoreapp3.1 would affect that. We should just take a closer look at the global json and how we are restoring the CLI to make sure that we restore the right one for the current platform.

richlander, 2022-12-09:
It would be good to get coverage on Arm64. It's our premier offering, independent of the % of folks using it.

joperezr, 2023-01-12:
I believe this is the culprit:
https://github.com/dotnet/iot/blob/839335b65c91e51e37f44cd026beaebe86de7f97/global.json#L5-L8

I'll take a look at what other repos do.

joperezr, 2023-01-12:
One way to easily fix this is to change all of our tests so that they run on net8.0. That should fix the issue you were hitting.