dotnet/iot #1996: FT232H I2C Multi byte reads does not work

gabbla, 2022-12-17:
> TL;DR Reading more than 1 byte using the FT232H I2C bus result in a bus lock. The single byte reads work as expected.

I am writing a program to control a test bench which is equipped with several IO expanders (PCA9554), a temperature sensor SHT31 and and ADC (ADS7828). The bench is connected to the PC via a FT232H based cable. I am able to interact with the IOEXPs just fine, but when I need to read a temperature or an ADC channel the bus stalls. I investigated further on the ADC side and found that reading more than 1 byte results in a unusable bus.

The reading process is simple: write the channel you want to read, then read 2 bytes.

[image]
[image]
[image]

The hardware has been validated with a nucleo, which works as expected:

[image]

> The above Logic2 captures are available:  [ft232h_dotnet.zip](https://github.com/dotnet/iot/files/10252015/ft232h_dotnet.zip).

**Steps to reproduce**

Since the ADS7828 is not available in the library, I wrote my driver. I guess you can try to read more than 1 byte from any device and get the same issue.

```csharp
        public short Read(InputMultiplexer ch)
        {
            Span<byte> writeBuff = stackalloc byte[1]
            {
                ChannelToByte(ch)
            };
            Span<byte> readBuff = stackalloc byte[2];

            i2cDevice.WriteRead(writeBuff, readBuff);
            return BinaryPrimitives.ReadInt16BigEndian(readBuff);
        }
```

**Expected behavior**

It should be possible to read more than 1 byte while using the FT232H binding.

**Versions used**

```
dotnet --info
.NET SDK:
 Version:   7.0.101
 Commit:    bb24aafa11

Runtime Environment:
 OS Name:     Windows
 OS Version:  10.0.19045
 OS Platform: Windows
 RID:         win10-x64
 Base Path:   C:\Program Files\dotnet\sdk\7.0.101\

Host:
  Version:      7.0.1
  Architecture: x64
  Commit:       97203d38ba

.NET SDKs installed:
  7.0.101 [C:\Program Files\dotnet\sdk]

.NET runtimes installed:
  Microsoft.AspNetCore.App 3.1.32 [C:\Program Files\dotnet\shared\Microsoft.AspNetCore.App]
  Microsoft.AspNetCore.App 6.0.12 [C:\Program Files\dotnet\shared\Microsoft.AspNetCore.App]
  Microsoft.AspNetCore.App 7.0.1 [C:\Program Files\dotnet\shared\Microsoft.AspNetCore.App]
  Microsoft.NETCore.App 3.1.32 [C:\Program Files\dotnet\shared\Microsoft.NETCore.App]
  Microsoft.NETCore.App 6.0.12 [C:\Program Files\dotnet\shared\Microsoft.NETCore.App]
  Microsoft.NETCore.App 7.0.1 [C:\Program Files\dotnet\shared\Microsoft.NETCore.App]
  Microsoft.WindowsDesktop.App 3.1.32 [C:\Program Files\dotnet\shared\Microsoft.WindowsDesktop.App]
  Microsoft.WindowsDesktop.App 6.0.12 [C:\Program Files\dotnet\shared\Microsoft.WindowsDesktop.App]
  Microsoft.WindowsDesktop.App 7.0.1 [C:\Program Files\dotnet\shared\Microsoft.WindowsDesktop.App]

Other architectures found:
  x86   [C:\Program Files (x86)\dotnet]
    registered at [HKLM\SOFTWARE\dotnet\Setup\InstalledVersions\x86\InstallLocation]

Environment variables:
  Not set

global.json file:
  Not found

Learn more:
  https://aka.ms/dotnet/info

Download .NET:
  https://aka.ms/dotnet/download
```

- `System.Device.Gpio` 2.3.0-prerelease.22520.10
- `Iot.Device.Bindings` 2.3.0-prerelease.22520.10

Ellerbach, 2022-12-20:
Yes, you are write. This is a limitation and should be addressed. It seems there are few more issues with the FT232H.
This should require investigations.

Ellerbach, 2022-12-20:
Note FT4222 has been proved to work nicely: https://github.com/dotnet/iot/tree/main/src/devices/Ft4222

So that can be a good solution for you as well.

gabbla, 2022-12-20:
Thank you for the confirmation. I'll take a look during the holidays to see if I can spot anything strange. 
Unfortunately I need to stick to the FT232 since my company has stocked several units. Thanks for the suggestion tho!

Gab

Ellerbach, 2022-12-30:
Closing this one as seems to be fixed by the last PR!

PR #2001 (2022-12-30): FT232H: SetDataBitsLowByte requires value before direction (#1996)

`Fixes #1996`

Reading more than 1 byte using FT232H I2C bus resulted in an unusable bus because the command used to set the GPIOs direction and values had the parameters appended in the wrong order. This PR fix the order. Tested in the same condition as described in #1996.

###### Microsoft Reviewers: [Open in CodeFlow](https://portal.fabricbot.ms/api/codeflow?pullrequest=https://github.com/dotnet/iot/pull/2001)
closed 2022-12-30 by Ellerbach