dotnet/iot #2011: I2C-Device0 on Raspberry PI

crlDB, 2023-01-14:
I want to use I2C device 0 on a raspberry PI. Following problem only occurs for the I2C-device 0, it works fine for I2C-device 1  

---
- Scanning the bus, device is visible (0x26)  
> sudo i2cdetect -y 0  
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f  
00:                         -- -- -- -- -- -- -- --  
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --  
20: -- -- -- -- -- -- 26 -- -- -- -- -- -- -- -- --  
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --  
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --  
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --  
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --  
70: -- -- -- -- -- -- -- --  
---
- Python code works fine, receives the data on the inputs of the MCP23017

```python
import smbus2

# MCP23017 address (A0-A2 pins tied to GND)
MCP23017_ADDRESS = 0x26

# I2C bus for Raspberry Pi
BUS = 0

# Initialize I2C bus
bus = smbus2.SMBus(BUS)

# Read data from all GP ports
data = bus.read_i2c_block_data(MCP23017_ADDRESS, 0x12, 2)

# Print data
print(data)
```
---
- Running c# code fails, receiving Error 110 performing I2C data transfer.
```csharp
 var i2cSettings2 = new I2cConnectionSettings(0, 0x26);
 var i2cDevice2 = I2cDevice.Create(i2cSettings2);
 var mcp230172 = new Mcp23017(i2cDevice2);

```
---
- after running .net app :
  - i2cdetect -y 0 fails
            0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f  
        00:                         -- -- -- -- -- -- -- --  
        10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --  
        20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --  
        30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --  
        40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --  
        50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --  
        60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --  
        70: -- -- -- -- -- -- -- --  
  - Python code fails, get TimeoutError: [Errno 110] Connection timed out
---
- Configuration
  - Raspberry PI 4 Model B Rev 1.2
  - PRETTY_NAME="Debian GNU/Linux 11 (bullseye)"
  - NAME="Debian GNU/Linux"
  - VERSION_ID="11"
  - VERSION="11 (bullseye)"  
  - uname -m > aarch64
---
- Packages
  - "Iot.Device.Bindings" Version="2.2.0"  
  - "System.Device.Gpio" Version="2.2.0"
 --- 
- Dotnet --info on the machine used to build

.NET SDK:  
 Version:   7.0.101  
 Commit:    bb24aafa11  

Runtime Environment:  
 OS Name:     Windows  
 OS Version:  10.0.19044  
 OS Platform: Windows  
 RID:         win10-x64  
 Base Path:   C:\Program Files\dotnet\sdk\7.0.101\  

Host:  
  Version:      7.0.1  
  Architecture: x64  
  Commit:       97203d38ba   

---  
- dotnet --info on the machine where app runs    

.NET SDK:
 Version:   7.0.101
 Commit:    bb24aafa11

Runtime Environment:
 OS Name:     debian  
 OS Version:  11  
 OS Platform: Linux  
 RID:         debian.11-arm64  
 Base Path:   /home/pi/.dotnet/sdk/7.0.101/  

Host:  
  Version:      7.0.1  
  Architecture: arm64  
  Commit:       97203d38ba  

---

raffaeler, 2023-01-14:
Hi @crlDB how did you connect the MCP23017 pins?
As stated from the datasheet, this pin must be biased for the device to work correctly.
If you leave it floating, the pin state will randomly change from 0 and 1, making the device randomly working. This would explain the behavior you are seeing.

I strongly suggest to connect the RESET (which is negated) to another GPIO and setting this GPIO to HIGH at the very beginning (even before setting up the I2C communication). This allows you to eventualy reset (setting 0 and then back 1) the MCP23017 in case the communication does not work anymore.

Finally do not forget to put two pull-up resistors (values may range from 1.2K to 10K) on the two I2C lines.

crlDB, 2023-01-14:
Hi @raffaeler

I think the hardware is OK because:
- i2cdetect -y 0, gives a good result, node 0x26 is found
- can access node 0x26 (MCP23017) through a python program

Besides the same hardware connected to I2C-device-**1** is working perfect with .net code.

Only the combination I2C-device-**0** and .net code breaks the port completely.  The I2CDetect / python program works again after rebooting the PI.

raffaeler, 2023-01-14:
What you see does not necessarily proves that the wiring is ok:
- Bus 0 and Bus 1 may have different internal connections. If you didn't provide the pullups, this may be the reason for failing. Pullups are mandatory in the I2C specs.
- The random failing are pretty normal when a reset oscillates. The datasheet clearly states that you should keep it high in order for the device to reliably respond.

What you see is strange because we have plenty of I2C devices working on both buses. So I don't believe this may be caused from the dotnet IoT I2C library.

crlDB, 2023-01-14:
Hi @raffaeler

The same hardware in combination with the same port (I2C-device-**0**) can be accessed completeley by python.
Surely this proves that the hardware is OK, or am I missing something.

pgrawehr, 2023-01-14:
Please show the contents of ` /boot/config.txt`. There could be a pin assignment conflict.

crlDB, 2023-01-14:
Hi @pgrawehr 

I just add 'dtparam=i2c_vc=on' to activate port0
 
```
# For more options and information see
# http://rpf.io/configtxt
# Some settings may impact device functionality. See link above for details

# uncomment if you get no picture on HDMI for a default "safe" mode
#hdmi_safe=1

# uncomment the following to adjust overscan. Use positive numbers if console
# goes off screen, and negative if there is too much border
#overscan_left=16
#overscan_right=16
#overscan_top=16
#overscan_bottom=16

# uncomment to force a console size. By default it will be display's size minus
# overscan.
#framebuffer_width=1280
#framebuffer_height=720

# uncomment if hdmi display is not detected and composite is being output
#hdmi_force_hotplug=1

# uncomment to force a specific HDMI mode (this will force VGA)
#hdmi_group=1
#hdmi_mode=1

# uncomment to force a HDMI mode rather than DVI. This can make audio work in
# DMT (computer monitor) modes
#hdmi_drive=2

# uncomment to increase signal to HDMI, if you have interference, blanking, or
# no display
#config_hdmi_boost=4

# uncomment for composite PAL
#sdtv_mode=2

#uncomment to overclock the arm. 700 MHz is the default.
#arm_freq=800

# Uncomment some or all of these to enable the optional hardware interfaces
dtparam=i2c1=on
dtparam=i2c_arm=on
dtparam=i2c_vc=on
#dtparam=i2s=on
#dtparam=spi=on

# Uncomment this to enable infrared communication.
#dtoverlay=gpio-ir,gpio_pin=17
#dtoverlay=gpio-ir-tx,gpio_pin=18

# Additional overlays and parameters are documented /boot/overlays/README

# Enable audio (loads snd_bcm2835)
dtparam=audio=on

# Automatically load overlays for detected cameras
#camera_auto_detect=1

# Automatically load overlays for detected DSI displays
display_auto_detect=1

# Enable DRM VC4 V3D driver
#dtoverlay=vc4-kms-v3d
max_framebuffers=2

# Run in 64-bit mode
arm_64bit=1

# Disable compensation for displays with overscan
disable_overscan=1

[cm4]
# Enable host mode on the 2711 built-in XHCI USB controller.
# This line should be removed if the legacy DWC2 controller is required
# (e.g. for USB device mode) or if USB support is not required.
otg_mode=1

[all]

[pi4]
dtoverlay=vc4-fkms-v3d
# Run as fast as firmware / board allows
arm_boost=1

[all]
enable_uart=1
gpu_mem=128
```

pgrawehr, 2023-01-15:
I have never tried this (and I can't test it right now), but I think `dtparam=i2c_arm=on` and `dtparam=i2c_vc=on` are actually redundant. Since I2C bus 0 is enabled by default, you should not need to configure anything for it. Try again without modifications, please.

crlDB, 2023-01-15:
Hi @pgrawehr 

The line 'dtparam=i2c_vc=on' is needed.

When the line is inactive:   
- sudo i2cdetect -y 0 > fails  
- python program on I2Cdevice-0 > fails  
- .net program on I2Cdevice-0 > fails  
  
When the line is active:    
- sudo i2cdetect -y 0 > OK  
- python program on I2Cdevice-0 > OK  
- .net program on I2Cdevice-0 > fails  
    
The 'dtparam=i2c_arm=on' does not affect the result

crlDB, 2023-01-15:
I found what the problem was, in a other part of the program GPIO0 and GPIO1 were used as input.
Something like this.

```csharp
GpioController _gpioCtrl = new GpioController(PinNumberingScheme.Logical);

// interrupt1
_gpioCtrl.OpenPin(0, PinMode.InputPullUp);
// interrupt2
_gpioCtrl.OpenPin(1, PinMode.InputPullUp);

```

The SDA and SCK of the I2CDevice-0 are situated on the same pins. (pin 27/28)

Running the code results in a crash of the I2C-DEVICE0. Even after stopping the program, an I2Cdetect does not work. The port can only be released by restarting the PI.
Obvious a mistake on our part, but a complete crash of the port is not the expected behavior?

raffaeler, 2023-01-15:
Both Python and this library just make calls on the underlying native drivers.
When a process closes, the OS is responsible for closing outstanding open handles.

If the driver does not restore a working state, the problem is probably there.
You can try creating a repro for the driver you are using (most probably libgpiod) and open a bug in their repository.

I now close this issue. Feel free to reopen it, should you find something wrong in the I2C code pertinent to this repository.
Thank you

closed 2023-01-15 by raffaeler

pgrawehr, 2023-01-15:
@crlDB Thanks for reporting back. It is known issue (to me) that may cause strange behavior such as this. I suspect that the problem isn't that the driver crashes, but that the pin multiplexer is configured incorrectly after you run your program. As you probably know, the Raspberry Pi has a multiplexer that switches the different devices (I2C, PWM, SPI, GPIO, etc) to the individual pins. If a pin is in GPIO mode, the I2C driver, even if it is active, just sees nothing attached. The configuration file `config.txt` both enables the driver _and_ sets the mulitplexer so that the given driver has access to the relevant pins. 

I recommend you use the `Board` class to create I2C and GPIO controllers. This will automatically set the multiplexer to the correct settings (if possible) and also prevents the error you where facing, as it disallows using one pin for multiple modes simulataneously.

crlDB, 2023-01-15:
thanks for this clear answer