dotnet/iot #2025: Error 107 performing I2C data transfer

ygoe, 2023-01-30:
**Describe the bug**

When I try to write bytes to a device, I get an exception with the message "Error 107 performing I2C data transfer". I have no idea what to do now. This error and the whole behaviour is not documented. What does it mean? Where can I find the problem?

**Steps to reproduce**

```csharp
var settings = new I2cConnectionSettings(2, 0x51);
var dev = I2cDevice.Create(settings);
dev.Write(new byte[] { 10, 5, 0, 255, 0, 1, 9 });
```

**Expected behavior**

It should write the bytes. Doesn't look like that did happen.

**Actual behavior**

```
- System.IO.IOException: Error 107 performing I2C data transfer.
  at System.Device.I2c.UnixI2cBus.WriteReadCore(UInt16 deviceAddress, Byte* writeBuffer, Byte* readBuffer, UInt16 writeBufferLength, UInt16 readBufferLength)
  at System.Device.I2c.UnixI2cBus.Write(Int32 deviceAddress, ReadOnlySpan`1 buffer)
  at System.Device.I2c.UnixI2cDevice.Write(ReadOnlySpan`1 buffer)
  at MyCode()...
```

**Versions used**

```
dotnet --info
.NET SDK:
 Version:   7.0.102
 Commit:    4bbdd14480

Laufzeitumgebung:
 OS Name:     Windows
 OS Version:  10.0.19045
 OS Platform: Windows
 RID:         win10-x64
 Base Path:   C:\Program Files\dotnet\sdk\7.0.102\

Host:
  Version:      7.0.2
  Architecture: x64
  Commit:       d037e070eb

.NET SDKs installed:
  6.0.400 [C:\Program Files\dotnet\sdk]
  7.0.101 [C:\Program Files\dotnet\sdk]
  7.0.102 [C:\Program Files\dotnet\sdk]

.NET runtimes installed:
  Microsoft.AspNetCore.App 3.1.32 [C:\Program Files\dotnet\shared\Microsoft.AspNetCore.App]
  Microsoft.AspNetCore.App 6.0.13 [C:\Program Files\dotnet\shared\Microsoft.AspNetCore.App]
  Microsoft.AspNetCore.App 7.0.2 [C:\Program Files\dotnet\shared\Microsoft.AspNetCore.App]
  Microsoft.NETCore.App 3.1.32 [C:\Program Files\dotnet\shared\Microsoft.NETCore.App]
  Microsoft.NETCore.App 6.0.13 [C:\Program Files\dotnet\shared\Microsoft.NETCore.App]
  Microsoft.NETCore.App 7.0.2 [C:\Program Files\dotnet\shared\Microsoft.NETCore.App]
  Microsoft.WindowsDesktop.App 3.1.32 [C:\Program Files\dotnet\shared\Microsoft.WindowsDesktop.App]
  Microsoft.WindowsDesktop.App 6.0.13 [C:\Program Files\dotnet\shared\Microsoft.WindowsDesktop.App]
  Microsoft.WindowsDesktop.App 7.0.2 [C:\Program Files\dotnet\shared\Microsoft.WindowsDesktop.App]

Other architectures found:
  x86   [C:\Program Files (x86)\dotnet]
    registered at [HKLM\SOFTWARE\dotnet\Setup\InstalledVersions\x86\InstallLocation]

Environment variables:
  Not set

global.json file:
  Not found

Learn more:
  https://aka.ms/dotnet/info

Download .NET:
  https://aka.ms/dotnet/download
```

* System.Device.Gpio version 2.2.0
* Iot.Device.Bindings not installed

Other details:

The app runs with .NET 7 on Linux-ARM64 and was built as a self-contained single file executable.

Ellerbach, 2023-02-01:
Thanks for reporting this, few questions (supposing you are running your code on a Linux/Unix based CPU based machine):
* Do you have I2C activated?
  * If you have a Raspberry, check this out: https://github.com/dotnet/iot/blob/main/Documentation/raspi-i2c.md
  * If you don't have a Raspberry, make sure your user is added to the group to access I2C, you can try with sudo as well
* Make sure you have a connected device on the I2C bus 2
  * Install the i2c tools:
    *  For Ubuntu/Debian:
       `sudo apt-get update && sudo apt-get install i2c-tools`
    * For RedHat/Fedora/CentOS:
      `sudo yum install i2c-tools`
    * After installation run
      `i2cdetect -y 2` as you seem to use the I2C 2 bus in your configuration
    * You should see a device present at the address 0x51

ygoe, 2023-02-01:
Thank you, my device address was wrong.

But what does "error 107" mean? If it's about "there is no such device", why can't the message say just that? Error numbers do not exist in the documentation or elsewhere. I'm wondering if they even exist in the source code or if that's a random static message string to confuse people?

pgrawehr, 2023-02-02:
These are just the error codes returned from the operating system. We generally don't know more about the error than this number. You can use the linux `errno` command to get more information (install the `moreutils` package).
Error 107 means
```
Transport endpoint is not connected
```

Ellerbach, 2023-02-02:
So closing this issue at it seems all is now working properly.

closed 2023-02-02 by Ellerbach