dotnet/iot #2036: Cannot Read DHT22 with Raspberry Pi 3b

blanthor, 2023-02-13:
**Describe the bug**
Cannot read DHT22 on Raspberry Pi 3b.

**Steps to reproduce**
### In WSL: 
  - installed dotnet 6
  - Followed instructions in Lab 1 of [RPi dotnet  core c# Labs](https://github.com/gloveboxes/Create-RaspberryPi-dotNET-Core-C-Sharp-IoT-Applications/blob/master/labs/Lab_1_Build_dot_NET_Core_app/README.md)
 - verified that deployment and debug works
 - replaced Program.cs with code from [Program.cs in Samples](https://github.com/dotnet/iot/tree/main/src/devices/Dhtxx/samples)
 - deployed code
 - had to run through ssh and debugging in VS Code in WSL wasn't supported
```csharp
// put sample code here -  this is the exact sample code as written in repo (alternative below also doesn't work)
void Dht(DhtBase dht)
{
    while (!Console.KeyAvailable)
    {
        Temperature temperature = default;
        RelativeHumidity humidity = default;
        bool success = dht.TryReadHumidity(out humidity) && dht.TryReadTemperature(out temperature);
        // You can only display temperature and humidity if the read is successful otherwise, this will raise an exception as
        // both temperature and humidity are NAN
        if (success)
        {
            Console.WriteLine($"Temperature: {temperature.DegreesCelsius:F1}\u00B0C, Relative humidity: {humidity.Percent:F1}%");

            // WeatherHelper supports more calculations, such as saturated vapor pressure, actual vapor pressure and absolute humidity.
            Console.WriteLine(
                $"Heat index: {WeatherHelper.CalculateHeatIndex(temperature, humidity).DegreesCelsius:F1}\u00B0C");
            Console.WriteLine(
                $"Dew point: {WeatherHelper.CalculateDewPoint(temperature, humidity).DegreesCelsius:F1}\u00B0C");
        }
        else
        {
            Console.WriteLine("Error reading DHT sensor");
        }

        // You must wait some time before trying to read the next value
        Thread.Sleep(2000);
    }
}
```
// The alternative is one I added to the Lab
````csharp
using System;
using Iot.Device.CpuTemperature;
using System.Threading;
//using Iot.Device.DHTxx;
using Iot.Device.DHTxx;
using UnitsNet;
using System.Device.Gpio;
using System.Device.Gpio.Drivers;

namespace dotnet.core.iot
{
    class Program
    {
        private static int pin = 37;
        private static GpioDriver driver = new RaspberryPi3Driver();
        private static GpioController controller = new GpioController(PinNumberingScheme.Board);

        static CpuTemperature temperature = new CpuTemperature();
        static void Main(string[] args)
        {
            int count = 0;

            SetUpGPIO();

            while (true)
            {
                if (temperature.IsAvailable)
                {
                    Console.WriteLine($"The CPU temperature is {temperature.Temperature.DegreesCelsius}");
                }
                else
                {
                    Console.Write(".");
                    if(count++ > 5)
                    {
                        Console.WriteLine("\ntemperature reading is not available.");
                        break;
                    }
                }
                Thread.Sleep(2000); // sleep for 2000 milliseconds, 2 seconds

                ReadDHT();

            }
        }

        static void SetUpGPIO()
        {
            pin=26;
            driver = new RaspberryPi3Driver();
            controller = new GpioController(PinNumberingScheme.Logical);
        }
        static void ReadDHT()
        {

            // GPIO Pin 
            //using (var dht = new Dht11(pin))
            using (var dht = new Dht22(pin, pinNumberingScheme: PinNumberingScheme.Logical))
            {
                Temperature temperature = default;
                RelativeHumidity humidity = default;
                //bool success = dht.TryReadHumidity(out humidity) && dht.TryReadTemperature(out temperature);
                bool success = dht.IsLastReadSuccessful;
                // You can only display temperature and humidity if the read is successful otherwise, this will raise an exception as
                // both temperature and humidity are NAN
                if (success)
                {
                    humidity = dht.Humidity;
                    temperature = dht.Temperature;
                    
                    Console.WriteLine($"Temperature: {temperature.DegreesCelsius:F1}\u00B0C, Relative humidity: {humidity.Percent:F1}%");
                }
                else
                {
                    Console.WriteLine("Error reading DHT sensor");
                }
            }
        }
    }
}
````

**Expected behavior**
A new DTH22 object should be initialized with the ability to read Temperature and Humidity.
IsLastReadSuccessfull should be true

**Actual behavior**
IsLastReadSuccessfull should be true
Temperature and Humidity return default values.
Excecptions are thrown within the controller and the driver objects

**Versions used**
Raspberry Pi 3b with Raspberry Pi OS: "Raspbian GNU/Linux 9 (stretch)"
dotnet 6.0.405
WSL: Ubuntu 20.04 LTS (Focal Fossa)

krwq, 2023-02-16:
[Triage] Can you provide how you connected your DHT22 and show exception message you're getting

pgrawehr, 2023-02-18:
Why are you compiling in WSL? That just complicates things.

closed 2023-02-26 by msftbot[bot]