dotnet/iot #2053: Issue with Arduino tests

krwq, 2023-03-17:
Recently Arduino tests started failing with:

```
Error during build: Platform 'esp32:esp32' not found: platform not installed
Try running `arduino-cli.exe core install esp32:esp32`
```

It could be specific to some machines.

These tests are disabled here: https://github.com/dotnet/iot/pull/2051

https://github.com/dotnet/iot/blob/e12acc2a629e23daa6c4b3a5c15349450f6ffc6b/azure-pipelines.yml#L93-L113

pgrawehr, 2023-03-18:
@krwq Sorry, I strongly oppose this change. The error is a problem with the runners, not with our code. For some reason, the installation of the esp32 module sometimes fails: 

```
2023-03-11T13:35:08.8339032Z Installing platform esp32:esp32@2.0.7...
2023-03-11T13:35:11.3731402Z runtime: out of memory: cannot allocate 536870912-byte block (1115357184 in use)
2023-03-11T13:35:11.3732358Z fatal error: out of memory
```
For some reason, the runners appear to not always be exactly equivalent. I have the impression it depends on the overall load of the build machines (it's typically better early in the morning than in the afternoon).

pgrawehr, 2023-03-18:
Again: this is a problem of the infrastructure, not of our tests. We should just retry in such cases. Disabling tests until everything is green is not a solution.

joperezr, 2023-03-20:
tests are one thing, but I think I agree with @krwq on disabling Arduino specific build. We have talked about this several times during triage, but essentially the purpose of CI is to ensure and protect against build breaks from the incoming code. Given that the Arduino binding (and other code that might impact it) are very rarely touched, having this leg as part of the build is not really giving us much benefit. In other words, I don't think this build has ever failed due to a legitimate break in the Arduino binding, except for PRs that were actually touching it which was mostly just work done by @pgrawehr. In the past we had agreed we would move this pipeline to a separate leg that can be invoked on-demand, so I see the change @krwq made on #2051 as part of that process, we just now need to create a new yml and move this code there. After that we can just create a new pipeline that runs on demand.

Ellerbach, 2023-08-17:
[Triage] Closing as disabled because they were running for too long and they are anyway run locally when touching the library.

closed 2023-08-17 by Ellerbach