dotnet/iot #2064: FtCommon.GetDevices() outputs wrong information

COssfeld, 2023-04-03:
**Describe the bug**

I want to use the FT232H from adafruit with dotnet 6 and https://github.com/dotnet/iot/tree/2.2/src/devices/Ft232H

**Steps to reproduce**
```csharp
<PackageReference Include="Iot.Device.Bindings" Version="2.2.0" />
```

```csharp
using System;
using Iot.Device.FtCommon;

var devices = FtCommon.GetDevices();
Console.WriteLine($"{devices.Count} available device(s)");
foreach (var device in devices)
{
    Console.WriteLine($"  {device.Description}");
    Console.WriteLine($"    Flags: {device.Flags}");
    Console.WriteLine($"    Id: {device.Id}");
    Console.WriteLine($"    LocId: {device.LocId}");
    Console.WriteLine($"    Serial number: {device.SerialNumber}");
    Console.WriteLine($"    Type: {device.Type}");
}

if (devices.Count == 0)
{
    Console.WriteLine("No device connected");
    return;
}

```

**Expected behavior**
This example results on using the same code on a windows machine with D2XX driver 2.12.36.4
[image]

**Actual behavior**
This is what I get on ubuntu 22.04.2 LTS with D2XX driver 1.4.27
I installed the driver like this:
https://www.ftdichip.com/Drivers/D2XX/Linux/ReadMe-linux.txt
Then I added linker path:
sudo ldconfig /usr/local/lib/
and added udev rules to /etc/udev/rules.d/11-ftdi.rules
```
SUBSYSTEM=="usb", ATTR{idVendor}=="0403", ATTR{idProduct}=="6001", GROUP="plugdev", MODE="0666"
SUBSYSTEM=="usb", ATTR{idVendor}=="0403", ATTR{idProduct}=="6011", GROUP="plugdev", MODE="0666"
SUBSYSTEM=="usb", ATTR{idVendor}=="0403", ATTR{idProduct}=="6010", GROUP="plugdev", MODE="0666"
SUBSYSTEM=="usb", ATTR{idVendor}=="0403", ATTR{idProduct}=="6014", GROUP="plugdev", MODE="0666"
SUBSYSTEM=="usb", ATTR{idVendor}=="0403", ATTR{idProduct}=="6015", GROUP="plugdev", MODE="0666"
```
After this I unload the wrong drivers:
```
sudo rmmod ftdi_sio
sudo rmmod usbserial
```
After this I started the programm and got:

[image]

I also tried the older driver for linux 1.4.22 and tried dotnet 5.

 also used the example programm inlcuded in the driver folder:
 
[image]

**Versions used**

.NET SDK (reflecting any global.json):
 Version:   6.0.407
 Commit:    d8659ea151

Runtime Environment:
 OS Name:     ubuntu
 OS Version:  22.04
 OS Platform: Linux
 RID:         ubuntu.22.04-x64
 Base Path:   /home/XXX/.dotnet/sdk/6.0.407/

global.json file:
  Not found

Host:
  Version:      6.0.15
  Architecture: x64
  Commit:       5edef4b20b

.NET SDKs installed:
  6.0.407 [/home/XXX/.dotnet/sdk]

.NET runtimes installed:
  Microsoft.AspNetCore.App 6.0.15 [/home/XXX/.dotnet/shared/Microsoft.AspNetCore.App]
  Microsoft.NETCore.App 6.0.15 [/home/XXX/.dotnet/shared/Microsoft.NETCore.App]

krwq, 2023-05-04:
cc: @Ellerbach

Ellerbach, 2023-09-01:
@COssfeld I have been looking at the code and it is the exact same for Windows and Linux.
The GetDevices() function is here: https://github.com/dotnet/iot/blob/47e40bcde89aeeed3ceaea9e6463060cd42f63e7/src/devices/FtCommon/FtCommon.cs#L23
It first call the FT_CreateDeviceInfoList see: https://github.com/dotnet/iot/blob/47e40bcde89aeeed3ceaea9e6463060cd42f63e7/src/devices/FtCommon/FtCommon.cs#L30
The function is an ftd2xx standard function: https://github.com/dotnet/iot/blob/47e40bcde89aeeed3ceaea9e6463060cd42f63e7/src/devices/FtCommon/FtFunction.cs#L20
It then call the function FT_GetDeviceInfoDetail to get the details and it is the same, a standard ftd2xx function:
https://github.com/dotnet/iot/blob/47e40bcde89aeeed3ceaea9e6463060cd42f63e7/src/devices/FtCommon/FtFunction.cs#L35C3-L35C3

See the application note https://ftdichip.com/wp-content/uploads/2020/08/D2XX_Programmers_GuideFT_000071.pdf and look at sections 3.5 and 3.6. Only loc ID should not show up on Linux.

I've been testing this on Windows, Mac and Linux and never had this issue. So, I don't think it's an issue with the code itself, I would be more something about the driver configuration.

Ellerbach, 2023-11-03:
@COssfeld I just checked and there is nothing I can fix as explained, it's related to the driver itself. The code is correct and follows the guidelines. So, closing this issue as it unfortunately works as expected with the limitations mentioned in the official FT guidelines.

closed 2023-11-03 by Ellerbach