dotnet/iot #2066: RK3588S / Orange PI 5 GPIO support

radderz, 2023-04-07:
It'd be great if there was device driver support for GPIO using .NET for the new Orange PI 5 which is using the new RK3588S SoC.

Quite a few boards are coming out with RK3588 and RK3588S chips, like the Rock Pi 5 and Orange PI 5. The chip is quite powerful and can run .NET on a few Linux distributions but I am unsure about the GPIO support for the board.

At the moment I have a Raspberry PI 3 doing the GPIO and Orange PI 5 running which adds a bit of complexity in one of my projects so it would be great to not need to have both.

Ellerbach, 2023-04-07:
All the available Orange Pi and more genrally Sunxi drivers are available here: https://github.com/dotnet/iot/tree/main/src/devices/Gpio/Drivers

If the new RK3588S is well documented and follows the same logic as the existing drivers, you should be able to implment it quite easilly. And we ❤️ PR ;-)

So feel free to play with this, happy to support!

slyshykO, 2023-04-17:
Isn't the Sunxi drivers connected to Allwinner SoCs?

KirillAldashkin, 2023-10-16:
I found [documentation](https://github.com/FanX-Tek/rk3588-TRM-and-Datasheet/blob/master/Rockchip%20RK3588%20TRM%20V1.0-Part1-20220309.pdf) for RK3588[S] and wrote some minimally working [driver](https://gist.github.com/KirillAldashkin/406fd8f9ec2c6051ba7c02cf5bac4b05)
Don't have time to contribute in unfortunately

Ellerbach, 2023-10-16:
> I found [documentation](https://github.com/FanX-Tek/rk3588-TRM-and-Datasheet/blob/master/Rockchip%20RK3588%20TRM%20V1.0-Part1-20220309.pdf) for RK3588[S] and wrote some minimally working (tested on Orange Pi 5) [driver](https://gist.github.com/KirillAldashkin/406fd8f9ec2c6051ba7c02cf5bac4b05) Don't have time to contribute it unfortunately

Thanks a lot for sharing. Feel free to contribute back once you'll have a bit of time!

redarccoder, 2024-01-05:
Was the driver released? I wanna use the OrangePI 5 for a new project with .Net and Avalonia for UI rendering on the FrameBuffer, but, having support for GPIO is a must* (I could create a python service for reading the inputs but that would complicate the project a LOT)

pgrawehr, 2024-01-07:
@redarccoder No, unfortunately no further work was done on this. You're welcome to take the above code and complete the driver, though.

GreenShadeZhang, 2025-05-13:
Do you have any new adaptation plans?

yaqi825, 2025-05-13:
您好，您的邮件已经收到，我会尽快回复，谢谢。祝工作愉快

raffaeler, 2025-05-13:
@radderz 
> Writing the driver side is a bit beyond me unfortunately.

In this repo, we call 'driver' a wrapper for specific boards. They are not OS drivers.
Our drivers basically maps the GPIOs pin numbers, the GPIO chips, etc.
Laurent already commented with the folder containing some great examples: https://github.com/dotnet/iot/tree/main/src/devices/Gpio/Drivers