dotnet/iot #2072: Adding Source code to Project Errors

Smurfj3, 2023-04-26:
If I add the source code as a new project to my WPF application, I get the following errors:

[image]

pgrawehr, 2023-04-29:
Can you explain why you would want to include the _source code_ directly in your project? Just reference the nuget file. 

On the other hand, the error indicates that you have some problems with accessing the nuget server. Do you have any strange `Nuget.config` files in your project?

Smurfj3, 2023-05-03:
I was trying to get it to work in Xamarin Forms, but with the Nuget file there are a few Windows related errors, I was trying to see if those things were necessary to run in Xamarin Forms.

Smurfj3, 2023-05-04:
[image]

Those are the errors I get as soon as I reference the Nuget File. 

This is not going to be used on a phone, it's actually Android installed on an Orange Pi 5, which I am aware there is no driver support for the Orange Pi 5 yet, I see someone else actually requested it. I am hoping it will be added soon or I am going to have to look at how it works exactly, so I can add it myself. As far as I am aware there is already Rockchip driver and just need to find the GPIO pin offset and do the mapping, I just haven't really looked into how that works and what it means if you make mistakes, not planning on frying my Orange Pi 5.

closed 2023-05-29 by Smurfj3