dotnet/iot #2075: The PN532 message factory produces bugged extended frames

pianca, 2023-05-08:
**Describe the bug**

The PN532 message factory produces extended frames that are not compliant with 1.6 firmware specifications as reported at page 29 of the official manual https://www.nxp.com/docs/en/user-guide/141520.pdf
After sending a bugged  extended frame there is no response.

**Steps to reproduce**
Checkout 02437dfa295cf8b627fb734236a79c16881d0cb7
Force the use of extended frames replacing line 1571 of Pn532.cs
```csharp
            int correctionLArgeSizeBuffer = writeData.Length < 250 ? 0 : 2;
```
with 
```csharp
            int correctionLArgeSizeBuffer = 2;
```
Run mifare test of the Pn532sample to verify that is broken.
Tested the sample on pn532 connected via serial port.

**Expected behavior**

The extended frame should work as the normal frame does, as reported in the manual at page 29.

**Actual behavior**

Every command does not receive a response in the default time (89 ms).

**Versions used**
.NET SDK:
 Version:   7.0.203
 Commit:    5b005c19f5

Runtime Environment:
 OS Name:     Windows
 OS Version:  10.0.19045
 OS Platform: Windows
 RID:         win10-x64
 Base Path:   C:\Program Files\dotnet\sdk\7.0.203\

Host:
  Version:      7.0.5
  Architecture: x64
  Commit:       8042d61b17

.NET SDKs installed:
  5.0.416 [C:\Program Files\dotnet\sdk]
  7.0.203 [C:\Program Files\dotnet\sdk]

.NET runtimes installed:
  Microsoft.AspNetCore.App 3.1.32 [C:\Program Files\dotnet\shared\Microsoft.AspNetCore.App]
  Microsoft.AspNetCore.App 5.0.17 [C:\Program Files\dotnet\shared\Microsoft.AspNetCore.App]
  Microsoft.AspNetCore.App 6.0.16 [C:\Program Files\dotnet\shared\Microsoft.AspNetCore.App]
  Microsoft.AspNetCore.App 7.0.5 [C:\Program Files\dotnet\shared\Microsoft.AspNetCore.App]
  Microsoft.NETCore.App 3.1.13 [C:\Program Files\dotnet\shared\Microsoft.NETCore.App]
  Microsoft.NETCore.App 3.1.15 [C:\Program Files\dotnet\shared\Microsoft.NETCore.App]
  Microsoft.NETCore.App 3.1.32 [C:\Program Files\dotnet\shared\Microsoft.NETCore.App]
  Microsoft.NETCore.App 5.0.17 [C:\Program Files\dotnet\shared\Microsoft.NETCore.App]
  Microsoft.NETCore.App 6.0.16 [C:\Program Files\dotnet\shared\Microsoft.NETCore.App]
  Microsoft.NETCore.App 7.0.5 [C:\Program Files\dotnet\shared\Microsoft.NETCore.App]
  Microsoft.WindowsDesktop.App 3.1.13 [C:\Program Files\dotnet\shared\Microsoft.WindowsDesktop.App]
  Microsoft.WindowsDesktop.App 3.1.15 [C:\Program Files\dotnet\shared\Microsoft.WindowsDesktop.App]
  Microsoft.WindowsDesktop.App 3.1.22 [C:\Program Files\dotnet\shared\Microsoft.WindowsDesktop.App]
  Microsoft.WindowsDesktop.App 3.1.32 [C:\Program Files\dotnet\shared\Microsoft.WindowsDesktop.App]
  Microsoft.WindowsDesktop.App 5.0.17 [C:\Program Files\dotnet\shared\Microsoft.WindowsDesktop.App]
  Microsoft.WindowsDesktop.App 6.0.16 [C:\Program Files\dotnet\shared\Microsoft.WindowsDesktop.App]
  Microsoft.WindowsDesktop.App 7.0.5 [C:\Program Files\dotnet\shared\Microsoft.WindowsDesktop.App]

- Version of `System.Device.Gpio`: dont know, is not released yet
- Version of `Iot.Device.Bindings` package: dont know, is not released yet

Ellerbach, 2023-05-09:
Thanks for reporting, you are correct. Seems like an additional line in the buffer management has been added.

Please adjust the lines 1597 to 1601 =>

```csharp
                // Large frame size
                buff[3 - correctionIndex] = 0xFF;
                buff[4 - correctionIndex] = (byte)(writeData.Length >> 8);
                buff[5 - correctionIndex] = (byte)(writeData.Length);
```

That should fix it, if you can report it's fixed, I'll be happy to review and approve your PR! I don't have anything to test this for quite some time.

pianca, 2023-05-09:
I think it's a little more complex...check the v. 1.6 specs:
[image]

I've pushed a PR with the fix from my fork https://github.com/dotnet/iot/pull/2076
I tested it by forcing the extended frame for every message creation and it works 

I'm trying to build an application to read the Italian Electronic Identity Card (ISO/IEC 14443, ISO/IEC 7816-4:2005, ISO/IEC 7816-8:2004) and I need to send an APDU of  273 bytes so I stumbled in the bug.
Anyway I discovered that the chip handles only APDUs of 264 bytes or less...
but at least I've helped the community 😃

PR #2076 (2023-06-15): Fixed the CreateWriteMessage for the PN532 chip.
Fixes #2075 
Fixed the PN532 message factory that now produce extended frames compliant with 1.6 firmware specifications

###### Microsoft Reviewers: [Open in CodeFlow](https://portal.fabricbot.ms/api/codeflow?pullrequest=https://github.com/dotnet/iot/pull/2076)
closed 2023-06-15 by pgrawehr