dotnet/iot #2081: Raspberry 4.1 is not supported by GPIO

SteenHessLattec, 2023-05-25:
**Describe the bug**

Attempting to use System.Device,Gpio on a "Raspberry Pi Compute Module 4 Rev 1.1" fails with System.PlatformNotSupportedException. 

**Steps to reproduce**
The tail of the file /proc/cpuinfo (used for board identification)  contains 
Hardware        : BCM2835
Revision        : a03141
Serial          : 10000000cb48e487
Model           : Raspberry Pi Compute Module 4 Rev 1.1

**Expected behavior**
At least support this board same as a Raspberry Pi Compute Module 4 Rev 1.0 (firmware 0x3140)
I suppose there is potentially differences between the boards , but i dont know these

**Actual behavior**
The function Model GetBoardModel() found in  System.Device.Gpio.RaspberryBoardInfo
has no handling of firmware 0x3141 and therefore returns Model.Unknown

**Versions used**
I have tried withe release version 2.20
In addition I hvae tried with the prerelease version 2.3.0-prerelease.22520.10

.NET SDKs installed:
  6.0.202 [/home/lattec/dotnet/sdk]

.NET runtimes installed:
  Microsoft.AspNetCore.App 6.0.4 [/home/lattec/dotnet/shared/Microsoft.AspNetCore.App]
  Microsoft.NETCore.App 6.0.4 [/home/lattec/dotnet/shared/Microsoft.NETCore.App]

krwq, 2023-05-25:
Seems like this was already addressed in https://github.com/dotnet/iot/commit/4572bdf14b4b1c05e226beef473cfa6d9e4b64e3 - once 3.0 ships this should be fixed. We'll leave the issue open until then.

While you're waiting you can unblock yourself by using nightly packages:
https://github.com/dotnet/iot/#net-cli

```
dotnet add package Iot.Device.Bindings --source https://pkgs.dev.azure.com/dotnet/IoT/_packaging/nightly_iot_builds/nuget/v3/index.json
```

lm-dk, 2023-05-26:
Hi @krwq 
Any chance that you could give a guesstimate on when this will go into a stable release?

Br.

raffaeler, 2023-05-26:
Hi @lm-dk we discussed this yesterday during the Triage.
You can see the progress here: https://github.com/dotnet/iot/milestone/6
The biggest one is [this PR](https://github.com/dotnet/iot/pull/2069) which is close to be reviewed and approved.

The easiest immediate workaround I can think of, is to clone the repo, build according to the documentation (which is easy) and reference the compiled assemblies.

Ellerbach, 2023-07-20:
[Triage] As we just shipped 3.0, and this is integrated into the release, we are closing this issue. Feel free to reopen if any issue.

closed 2023-07-20 by Ellerbach