dotnet/iot #2086: Capture images from Raspberry Pi Camera Module

mg-yolo-enterprises, 2023-06-08:
I'm interested in building a simple C# application to capture single image frames on Raspberry Pi. Some details:
OS: Raspberry Pi OS, Bullseye, 64-bit
Device: Raspberry Pi 4B
Camera: Raspberry Pi Camera Module 3

It looks like @raffaeler is the resident expert on the topic, based on reading elsewhere. Has there been any progress made on supporting libcamera from within C#? 

The documentation states "The legacy camera stack is not available in the 64-bit version of Raspberry Pi OS" so I don't believe that's an appropriate path to consider. I've considered trying to reverse-engineer [libcamera_hello](https://github.com/raspberrypi/libcamera-apps/blob/main/apps/libcamera_hello.cpp) in hopes that I might get somewhere using the C++ example and write some interop to do the work, but I'm not well versed in that language (yet).

This is destined for production code, so the image frames should be received efficiently and reliably for further processing in whatever format is cleanest; therefore I would prefer to avoid using the Process class to rely on other exe's to generate image files or produce a binary output stream, etc.

Is there a reasonable path forward here yet, or should I be considering this problem from a different direction?

raffaeler, 2023-06-08:
Hi @mg-yolo-enterprises, the fastest and easiest solution is to use the legacy stack because all the code you would need to ship would be .NET (both the iot libraries and your app) and ready to go. Unfortunately I confirm that the [legacy stack cannot be used on the 64 bit versions of the OS](https://www.raspberrypi.com/documentation/computers/camera_software.html#re-enabling-the-legacy-stack).

We would love to ship a new binding for Libcamera but this would require to ship a new version of the interoperability code with C++ for each version of Libcamera which is definitely a nightmare. The technical reason behind this is because C++ does not have a standard binary format. This means that every time you recompile some C++ code with a different compiler or even just a different version, the binary signature exported for the functions/classes in the library may change.
The Raspberry foundation can afford this because they ship the Raspbian OS. Every time they ship it, they recompile the interoperability layer with Python provided in their libraries for the Libcamera version they embded in the OS image. For us it's harder because it would require to maintain multiple versions of the interoperability layer, including backporting patches, bug fixes, etc.
For completeness, if Libcamera was a C library (or at least shipped with a C wrapper around the C++ code), there would be no problem at all. In fact, C has a stable binary standard and works regardless the compiler who generated it.

I wrote all of this because this is also your scenario when you should decide to go for your own interoperability layer.
You would need to:
- Write a small C wrapper around the functionalities you need for Libcamera. This is not necessarily needed if you decide to PInvoke directly the C++ code, but it can be harder because of the mangled names (the weird names that C++ function signatures in the dll).
- Write a C# PInvoke layer to bridge .NET to your wrapper (or directly to the C++ code)
- Keep both of them aligned to specific Libcamera versions. Every time your users  will `apt-get install` a new version of Libcamera, those layers may need to be updated and recompiled

Some time ago I spoke with a Libcamera dev who said me they would love to ship a C wrapper but do not have enough time to do (low priority). Anyone willing to help them in doing that would be welcome.

### Important note
Another approach would be to create an interoperability layer with the Python code provided in the OS image by the Raspberry PI foundation. It is probably not the best approach from the performance perspective but it would solve the  versioning nightmare.

If you have questions, let me know.

raffaeler, 2023-06-08:
I forgot to mention another alternative I used in the past.
You can run the raspi* tools with the Process class to do the hard work, and capture the streaming bytes using stdio of the process.
Even if it looks slow, it isn't. I used this approach to stream videos and works pretty well in Linux (it won't fit that well in Windows).

govert, 2023-08-08:
I found a Rust binding to libcamera [here](https://github.com/lit-robotics/libcamera-rs). They seem to have a C API as intermediate layer (a library they call [libcamera-sys](https://github.com/lit-robotics/libcamera-rs/tree/main/libcamera-sys/c_api) ). Maybe it would be useful as a start for a .NET binding - I can't tell myself but wanted to post the reference?

raffaeler, 2023-08-08:
@govert Interesting! The intermediate C library has the potential, but I have to spend some time to look at it first.

The rust library could also be useful, but unfortunately the [rust bindings for .NET only works on Windows](https://github.com/microsoft/windows-rs). FYI they could easily be ported and maintained on Linux as well but they have no plans for it.

raffaeler, 2023-08-31:
@kbingham Do you think it is possible for libcamera to integrate the wrapper coming from the [Rust binding](https://github.com/lit-robotics/libcamera-rs)?
Shipping together both C++ and the C wrapper would solve the versioning problem and make everybody happy :-)

kbingham, 2023-08-31:
definitely possible - if someone works on it and submits it to the libcamera project and meets our coding style requirements.  ...
The rust bindings project has been entirely separate from us so far.

kbingham, 2023-08-31:
Note that the project states:

License

Licensed under either of

    Apache License, Version 2.0, ([LICENSE-APACHE](https://github.com/lit-robotics/libcamera-rs/blob/main/LICENSE-APACHE) or https://www.apache.org/licenses/LICENSE-2.0)
    MIT license ([LICENSE-MIT](https://github.com/lit-robotics/libcamera-rs/blob/main/LICENSE-MIT) or https://opensource.org/licenses/MIT)

at your option.

libcamera is L-GPL - so I don't know enough to know if the existing code is compatible with the libcamera licence requirements. That may have to be discussed with the libcamera-sys authors - or the implementation needs to be handled from scratch.

raffaeler, 2023-08-31:
> That may have to be discussed with the libcamera-sys authors

Thanks for the answer @kbingham 
Are you willing to do that? I believe it would be very helpful for them as well. This would completely remove the need to check for the version and keep the C++ and C in sync.

raffaeler, 2023-08-31:
> The rust bindings project has been entirely separate from us so far.

Yes, I am wondering if the C wrapper (and just that) can be somehow moved into libcamera.

kbingham, 2023-08-31:
The code is unlikely to be accepted directly by simply copy/paste into libcamera and submit. There will be cleanup / discussion and ensuring the code follows libcamera code style and practices and unit tests for ongoing maintenance.

It's not something I can do myself right now - I don't have sufficient bandwidth.

In terms of licencing, the authors would have to be willing for that to happen and for the code to be licenced under the L-GPL to be merged to libcamera.

raffaeler, 2023-08-31:
@kbingham Thank you, but I am asking whether you are willing to contact the rust guys and asking them to contribute.

We (in this repo) are not familiar with libcamera or the rust binding and it would not make sense for us to put our hands on that.

On the other side, if the rust guys are interested, they could solve the versioning problem for them and also for any other library using a programming language (including C++ itself which is not compatible over different versions/compilers).
Thanks again

raffaeler, 2023-08-31:
> The code is unlikely to be accepted directly by simply copy/paste into libcamera and submit. There will be cleanup / discussion and ensuring the code follows libcamera code style and practices and unit tests for ongoing maintenance.

Sure, I understand this

> It's not something I can do myself right now - I don't have sufficient bandwidth.

I didn't suggested that, but just to contact the wrapper authors and asking them

> In terms of licencing, the authors would have to be willing for that to happen and for the code to be licenced under the L-GPL to be merged to libcamera.

I am not familiar with the licensing mismatch. But if they contribute, they can decide anything.

kbingham, 2023-08-31:
Sorry for the misunderstanding. I've already been in contact with the rust guys in the past. 

See https://github.com/lit-robotics/libcamera-rs/issues/3

raffaeler, 2023-08-31:
Thank you @kbingham but I don't see a practical outcome from that thread.
If instead something will ever happen, please let us/me know so that we can start working on the interoperability.

It's my personal opinion that dealing with a version mismatch between the interoperability wrapper and the C++ binary for a public project is a mission impossible. C++ was never meant for distributing binary libraries.
For your info, I made a proposal to the new `cppfront` proposal to automatically generate the `C` wrapper during transpilation. If this project will ever land to something real, it could totally resolve the problem, but I don't think it can happen in the short period.

In the next triage I will propose to close all the libcamera-related threads here, as I don't see a solution that is viable for this repository.

kbingham, 2023-08-31:
It's viable if someone will work on it or sponsor the work to be done. If no one works on it ... Then ?

No issue with you closing the issues. It's your project here.

raffaeler, 2023-08-31:
> It's viable if someone will work on it or sponsor the work to be done. If no one works on it ... Then ?

I understand that you need contributions, but the same is true for every other project, including the current repo.
I am one of the many volunteers that tries to do his best in his free time. This is how this repository (owned by the .NET Foundation) works.
Contributions are possible when someone has a very specific problem and is willing to share his/her code to the repository. This is more complicated with complex libraries like libcamera.

On the contrary, I suppose libcamera needs adoption and I don't see how this can grow with the versioning mismatch issue.

Should I need libcamera, it would be easy for me to create such interop for my own project. But keeping in sync and upgrading for every library that you deploy publicly it's a totally different problem which impacts backward compatibility, security issues, tests, CI, etc.

> No issue with you closing the issues. It's your project here.

Not mine, but the solution should be pushed and driven by libcamera, even if the contribution comes from external teams, as I suggested in this thread.
I may be wrong, but my impression is that you don't see the value in having an interoperability layer available for all the languages. This is the only reason why I am going to propose closing the issues.

To be very clear, I am speaking for myself.

kbingham, 2023-09-11:
I'm sorry if I gave you the impression that I don't see value in supporting other languages, that's not the case at all. Infinite resources would surely solve every technical problem - but alas we are doing what we can with the resources available to us ...

I do want to see libcamera being used by all who need it. That said - users shouldn't care about libcamera - it's intended to be a system implementation detail.

Even if we provide a C API /now/ this will still not be considered stable. We are /explicitly/ at v0.x to state this. Controls are being added/removed/renamed. Configuration structures around the public API are being changed. We will very soon be changing how the Sensor can be reconfigured for instance. This will cause ABI breakage in both C++ and it would also cause breakage if it were a C API.

Sure - libcamera needs adoption - but we also need to experiment to get the implementation right. I do look forward to when we can settle to a 1.x though.

And I think I understand what you are trying to say about the difficulties of binary stability of C++ - for that we are already using the [PIMPL](https://en.cppreference.com/w/cpp/language/pimpl) design pattern to reduce the public ABI from any implementation details.

In the future I would expect to see us implement the [hourglass](https://www.google.com/url?sa=t&rct=j&q=&esrc=s&source=web&cd=&cad=rja&uact=8&ved=2ahUKEwjprL3c26OBAxW4TUEAHcFNACoQwqsBegQIDxAG&url=https%3A%2F%2Fwww.youtube.com%2Fwatch%3Fv%3DPVYdHDm0q6Y&usg=AOvVaw0wb7Fr0Dwx37vu8Mr95Sw5&opi=89978449) C++ pattern to continue our C++ internally, with a C API built on top, and a C++ layer on top of that to continue the C++ API. Back in February I took some notes on this - https://github.com/kbingham/linux-cameras/issues/5. Perhaps a Google-summer-of-code project could help progress this ?

Do you have other ideas how libcamera can drive this forward specifically?

Finally - it may be that perhaps using the Pipewire API would be more suited to your use case. Pipewire already supports libcamera devices, and it can provide access to libcamera devices through a C API.

raffaeler, 2023-09-14:
> users shouldn't care about libcamera - it's intended to be a system implementation detail.

Users are trying to access the camera to acquire still images or videos. In this repository we have a binding for the `legacy` Raspberry PI stack that the Raspberry Pi foundation deprecated in favor of `libcamera`.
This is not an implementation detail but a very practical problem that users have at this time.

> I do look forward to when we can settle to a 1.x though.

From our perspective this translates that it does not make any sense to make an investment now.

As a personal effort, I'll try to figure out how to access the hardware directly and acquire images/streams directly.
In the worst case I can use the code I already wrote a long time ago that reads the stdout provided by the RPi tools.

When libcamera will reach a `1.0` and there will be interest in offering the other languages a stable ABI, it will be straightforward for me/us to create the interoperability layer. Most probably it can even be automatically using the (already available) C# code generators.

raffaeler, 2023-11-10:
My new Camera binding has been merged yesterday and will soon be packaged in version 3.1 as soon as the milestone is reached: https://github.com/dotnet/iot/milestone/7

This new Camera binding works with both the old legacy stack and the new libcamera stack.
Anyway, it does **not** use the libcamera APIs for the reasons mentioned in this thread.
Instead, it leverages the linux pipes to get the binary data using the apps shipped with the Raspbian OS. I do not expect significant performance hit deriving from this choice. In any case this was the one and only way to support everything, rapidly while still avoiding the problems with using the `libcamera` binaries.

I invite @mg-yolo-enterprises to test the binding and provide feedback.
I will now close this issue but feel free to open a new one in case of bugs or any other problem related with the new camera binding.

closed 2023-11-10 by raffaeler