dotnet/iot #2087: DHT22 Fails to read below humidity threashold

tylerhartwig, 2023-06-11:
I'm using the DHT22 class on a Raspberry Pi Zero W (running mono). Reading temp and humidity work great until the humidity gets too low. I verified this by running an alternate set of code (the AdaFruit Python lib for DHT22) which does work in comparison to reading with this library. The humidities I'm working with are under 25%, my guess is it starts to fail at 25.5%. 

To reproduce: Attempt to use a DHT22 Sensor to read a value below 25%. 

My code to read: 
```fsharp
let main argv =
    let pin = 17
    use sensor = new Dht22(pin)
    sensor.MinTimeBetweenReads <- TimeSpan.FromSeconds(5.)
   
    let readValues () =
        let mutable isRead = false
        let mutable temp = None
        let mutable humidity = None
        while not isRead do
            printfn "Attempting temp read"
            match sensor.TryReadTemperature() with
            | (true, t)  ->
                temp <- Some t.DegreesFahrenheit
                printfn "Temp read as: %f" (Option.get temp)
                isRead <- true
            | _ ->
                printfn "Temp read failed, retrying..."
                Thread.Sleep(5000)
            
        
        isRead <- false
        
        while not isRead do
            printfn "Attempting humidity read"
            match sensor.TryReadHumidity() with
            | (true, h)  ->
                humidity <- Some h.Percent
                printfn "Humidity read as: %f" (Option.get humidity)
                isRead <- true
            | _ ->
                printfn "Humidity read failed, retrying..."
                Thread.Sleep(5000)
        (temp, humidity)

    while true do
        printfn "Starting read loop"
        let (temp, humidity) = readValues()
 
        printfn "Read finished, sleeping 1 minutes"
        Thread.Sleep(TimeSpan.FromMinutes(1.0))

```

(I've omitted lines where I actually use the values)

**Versions used**

Iot.Device.Bindings 2.2.0
System.Device.Gpio 2.2.0

- `dotnet --info` on the machine being used to build: .NET 7.0
- `dotnet --info` on the machine where app is being run (not applicable for self-contained apps): Running on Mono

tylerhartwig, 2023-06-12:
I did some additional debugging, utilized reflection to access the read buffer, and this was the result (note I re-interpereted the humidity fields as it is done in Dht22)

```
Starting read loop
Attempting temp read
Read Buff was: [|0uy; 0uy; 0uy; 0uy; 0uy|]
As Humidity: 0y
Temp read failed, retrying...
Attempting temp read
Read Buff was: [|0uy; 0uy; 0uy; 0uy; 0uy|]
As Humidity: 0y
Temp read failed, retrying...
Attempting temp read
Read Buff was: [|0uy; 182uy; 1uy; 251uy; 0uy|]
As Humidity: 182y
Temp read failed, retrying...
Attempting temp read
Read Buff was: [|0uy; 179uy; 0uy; 251uy; 0uy|]
As Humidity: 179y
Temp read failed, retrying...
Attempting temp read
Read Buff was: [|0uy; 179uy; 0uy; 253uy; 176uy|]
As Humidity: 179y
```

The sensor settles at 17.9% (which is actually correct!). It seems this in incorrectly getting flagged as an incorrect reading.

An example of it working:
```
Starting read loop
Attempting temp read
Read Buff was: [|0uy; 0uy; 0uy; 0uy; 0uy|]
As Humidity: 0y
Temp read failed, retrying...
Attempting temp read
Read Buff was: [|0uy; 0uy; 0uy; 0uy; 0uy|]
As Humidity: 0y
Temp read failed, retrying...
Attempting temp read
Read Buff was: [|1uy; 180uy; 0uy; 126uy; 179uy|]
As Humidity: 181y
Temp read failed, retrying...
Attempting temp read
Read Buff was: [|1uy; 184uy; 0uy; 254uy; 183uy|]
Temp read as: 77.720000
Attempting humidity read
Read Buff was: [|1uy; 187uy; 0uy; 254uy; 0uy|]
Humidity read failed, retrying...
Attempting humidity read
Read Buff was: [|1uy; 187uy; 0uy; 253uy; 0uy|]
Humidity read failed, retrying...
Attempting humidity read
Read Buff was: [|1uy; 190uy; 0uy; 253uy; 188uy|]
Humidity read as: 44.600000
Read finished, sleeping 1 minutes
``` 

 If I'm not mistaken, the problem lies here:
 ```            
            if ((newBuf[4] == ((newBuf[0] + newBuf[1] + newBuf[2] + newBuf[3]) & 0xFF)))
            {
                _isLastReadSuccessful = (newBuf[0] != 0) || (newBuf[2] != 0);
            }
            else
            {
                _isLastReadSuccessful = false;
            }
```
byte 0 and 2 are both 0 when the humidity is below 25.5%. I'm not sure what the purpose of that check is, but I think that's the root issue here!

tylerhartwig, 2023-08-04:
@krwq sorry never saw this comment, which spec are you referring to? just the DHT Spec?

krwq, 2023-08-24:
yes, DHT22 spec - did you verify that fixing that fixed your issue?

Ellerbach, 2023-11-03:
@tylerhartwig I have been looking at this issue and setup a DHT22 on a RaspberryPI 4.
After few non correct reads, I manage to get properly readings:
[image]

Ellerbach, 2023-11-03:
@tylerhartwig ok, found your issue with lowering both the temperature and the humidity. Fix on its way.

closed 2023-11-06 by Ellerbach

tylerhartwig, 2023-11-16:
Awesome thanks for the fix! I made a fix locally but never had a chance to verify it against the spec, glad to see it resolved

PR #2155 (2023-11-06): Fix DHT22 reading errors when temperature and humidity are too low

- Fixes #2087
- Fix DHT22 reading errors when temperature and humidity are too low

###### Microsoft Reviewers: [Open in CodeFlow](https://portal.fabricbot.ms/api/codeflow?pullrequest=https://github.com/dotnet/iot/pull/2155)