dotnet/iot #2088: API support for SenseHat V2 Color Sensor TCS34725

Ducki, 2023-06-14:
The revision 2 of the Raspberry Pi SenseHat has a color sensor TCS34725, which is currently unsupported in the SenseHat bindings.

The sensor itself is [already implemented](https://github.com/dotnet/iot/tree/main/src/devices/Tcs3472x), but it would be nice, if it was usable via the SenseHat binding.

Ellerbach, 2023-06-14:
Feel free to raise a PR to add it. We're not tracking the binding requests. So I'm leaving this one open a little bit but it will be closed in one of the next triage we'll collectively have.

Ducki, 2023-06-15:
So while investigating, I found out that I was wrong about the sensor. The color sensor on the SenseHat seems to have changed recently to a different model: TSC3400 (which is not covered in the device bindings as of yet).

Source: https://github.com/astro-pi/python-sense-hat/issues/126#issuecomment-1319782198

I'm sorry about that, but my low level skills aren't good enough to implement the other sensor. :)

Ellerbach, 2023-06-18:
>  TSC3400 (which is not covered in the device bindings as of yet).

I just checked and indeed, the sensors are different. The excellent news is that they are vey close. So it should be a matter of hours to adjust/adapt/extend the existing one to support the 3400 ones. 3400 seems to have Infra red as well. I don't have this sensor at all but I'm happy to support and help.