dotnet/iot #2114: Segfault error again.

Awire9966, 2023-07-22:
**Describe the bug**

Library seg faulting

**Steps to reproduce**
Run this code on the "le potato" with ```libgpiod-dev``` installed:
```csharp
using System;
using System.Device.Gpio;
using System.Threading;

Console.WriteLine("Blinking LED. Press Ctrl+C to end.");
int pin = 18;
using var controller = new GpioController();
controller.OpenPin(pin, PinMode.Output);
bool ledOn = true;
while (true)
{
    controller.Write(pin, ((ledOn) ? PinValue.High : PinValue.Low));
    Thread.Sleep(1000);
    ledOn = !ledOn;
}
```

**Expected behavior**

My LED to flash

**Actual behavior**

Gave me ```Segmentation fault```

**Versions used**

Building Machine: https://pastebin.com/raw/mUmyK58q
Running Machine: https://pastebin.com/raw/NibVchiN
System.Device.Gpio version is 3.0.0

pgrawehr, 2023-07-23:
I don't think that's related. The other error happens only if you create the GpioController multiple times without disposing it. But this simple program should clearly not cause any problems. What version of libgpiod are you using? And can you say where it fails? Does it fail right away or only after some time?

Awire9966, 2023-07-24:
fails right away

Ellerbach, 2023-07-25:
Is this the hardware you are using? https://libre.computer/products/aml-s905x-cc/
And if yes, which version of debian are you running?
Which version of libgpiod is installed?
And is a simple "hello world" application working properly?

Awire9966, 2023-08-03:
> Is this the hardware you are using? https://libre.computer/products/aml-s905x-cc/ And if yes, which version of debian are you running? Which version of libgpiod is installed? And is a simple "hello world" application working properly?
Yes I am using that with the 5/3/2023 version and no i dont know how to install libgpiod 
this is the image:
https://distro.libre.computer/ci/raspbian/11/2023-05-03-raspbian-bullseye-arm64%2Baml-s905x-cc.img.xz

joperezr, 2023-08-04:
@Ellerbach @pgrawehr Are there any ideas on how we can try to repro this without the hardware to try to understand what is going on?

@Awire9966 one idea that I have is, can you try installing the `libgpiod-tools` package in your distro and then try running the command-line utilities to make sure those work as expected? The main ones to try out woud be `gpiodetect`, `gpioinfo [chip]`, `gpioget [chip] [line]` and `gpioset [chip] [line]=1`. In general those should use the same mechanism that we do in the library, so if they work but you still see issues  in the command line then the bug would be in our libraries and we should dig deeper. If those tests fail on the other hand, then there must be something wrong with your installation. I'll mark the issue as "needs author feedback" to track that you are getting that info for us.

Awire9966, 2023-08-06:
Update: I got it working, but it can only control a few pins

raffaeler, 2023-08-06:
> Update: I got it working, but it can only control a few pins

Could you please verify that this library can access the same pins of `libgpiod-tools`?
Can you also try a different OS like Ubuntu?

Ellerbach, 2023-08-06:
> Update: I got it working, but it can only control a few pins

Pin mapping is different for sure, please refer to the hardware vendor to understand how they are mapped.

Awire9966, 2023-08-07:
> > Update: I got it working, but it can only control a few pins
> 
> Pin mapping is different for sure, please refer to the hardware vendor to understand how they are mapped.

The vendor's map was "not good" putting it in a nice way.

joperezr, 2023-08-07:
Correct me if I'm wrong, but seems like there is nothing really to fix then in our library, as this seems to just be some issue with some vendor docs. I'll mark this issue to auto close but @Awire9966 let me know if there is anything that you still think is wrong with the libraries in this repo.

Awire9966, 2023-08-07:
I think its the driver's maintainer. The thing got fried while trying to fix it.

closed 2023-08-07 by Awire9966