dotnet/iot #2115: Move off from VSTest so we can get test console log failures

joperezr, 2023-08-03:
Due to a bug in vstest, when a test fails and we are running them in parallel we may lose the console output making it impossible to debug the test failure, especially when we have flaky tests. This issue is to track the work of move away from vstest and use the runner directly to avoid this issue.

ViktorHofer, 2023-08-31:
Yes, but mostly for historic reasons and because VSTest is too heavy weight for runtime's usecase. xunit.console is no longer maintained by Brad.

ViktorHofer, 2023-09-01:
> Isn't better migrating to the same infrastructure used by the runtime?

Definitely not. Our infrastructure only exists because of corner case requirements (we build the runtime ourselves) that other repositories shouldn't need to consider. I would recommend to stick using VSTest and work with their team on a path that resolves your issues so that customers would benefit from that as well.

@Evangelink @nohwnd who would be the best person to talk to about https://github.com/microsoft/vstest/pull/2702 to prevent iot moving away from VSTest?

Evangelink, 2023-09-01:
Thanks for the ping @ViktorHofer! @joperezr Sending you an invite for a call (will cc you @ViktorHofer).

krwq, 2025-07-31:
[Triage] Closing this as we don't think this still repros. We can reiterate on this if needed