dotnet/iot #2123: a few bugs in BuildHat code implementation

jeansfish, 2023-08-25:
1. ColorLightMatrix should be a passive sensor
2. ButtonSensor should be an active sensor
3. IsActiveSensor method in Brick.cs only checks ColourAndDistanceSensor, it should include all other active sensors.

raffaeler, 2023-08-27:
I'm not familiar with the BuildHat binding. Could you please expand a bit the problems you found?
It would be helpful if you also point to the sensors source code that you believe does not work as it should.

And of course, if you are willing to help with a PR fixing the issues, you would be more than welcome :)

Ellerbach, 2023-08-27:
@jeansfish it's been a long time since I implemented this binding and I don't remember everything. Now, firmware may have change and some elements may have change as well from the specifications. Happy to help.
* Looking at the code of the button and the definition,, it is not a passive sensor, so it seems properly set at a Sensor (not active, see below comment as well)
* The color matrix seems to be an active (see also below)
* Any sensor more than WeDoTilt seems to be active. The code checks any that is higher than the ColourAndDistanceSensor See: https://github.com/dotnet/iot/blob/29d63474bf507b11f95878e5eccc8d46663da0c3/src/devices/BuildHat/Models/SensorType.cs#L52  and here:
https://github.com/dotnet/iot/blob/29d63474bf507b11f95878e5eccc8d46663da0c3/src/devices/BuildHat/Brick.cs#L78

The only potential issue I see is for the 2 WeDo sensors before the color one.

jeansfish, 2023-08-28:
@Ellerbach Thanks for your amazing job. Yes, when I digging into it, as the BuildHat protocol defines, the button is not an active sensor and the color matrix is an active one. 
for the IsActiveSensor function, it should be >=SensorType.WeDoTiltSensor. I have done some tests and after code changes, both sensors work as expected.
There's another issue is MediumLinearMotor, it has only 2 modes, so active motor codes need to change.
I'm still testing other motors and sensors and try to find the best PID for the medium linear motor.

Ellerbach, 2023-08-28:
> for the IsActiveSensor function, it should be >=SensorType.WeDoTiltSensor. I have done some tests and after code changes, both sensors work as expected.

I did not have all the sensors to test but most of them, so indeed, some may not fully work as expected.

We love PR ;-) So once you have everything working, please submit a PR! fork the repo first then create a branch in your repo, make all your changes, push on your repo and then make a PR. And the git magic will happen :-)