dotnet/iot #2130: This driver is generic so it can not perform conversions between pin numbering schemes on Orange pi

blackmoonstudio, 2023-09-05:
using the System.Device.Gpio assembly fails on OrangePI lite

When trying to open a pin from the GpioController the following exception is thrown:
Unhandled exception. System.PlatformNotSupportedException: This driver is generic so it can not perform conversions between pin numbering schemes.

I try this simple code:

```
using System;
using System.Device.Gpio;

class Program
{
    static void Main(string[] args)
    {
        Console.WriteLine("hello");
        // Set up GPIO pin 4 as output
        using (var controller = new GpioController(PinNumberingScheme.Board))
        {
            int ledPin = 8;
            controller.OpenPin(ledPin, PinMode.Output);

            // Turn on the LED
            controller.Write(ledPin, PinValue.High);
            Console.WriteLine("LED is ON. Press any key to turn it off.");

            // Wait for user input
            Console.ReadKey(); ;

            // Turn off the LED
            controller.Write(ledPin, PinValue.Low);
            Console.WriteLine("LED is OFF. Press any key to exit.");

            // Wait for user input
            Console.ReadKey();
        }
    }
}
```

**Expected behavior**

I expected to turn on and off an LED with Orange Pi Lite

**Actual behavior**

at System.Device.Gpio.Drivers.SysFsDriver.ConvertPinNumberToLogicalNumberingScheme(Int32 pin number)
at System.Device.Gpio.GpioController.GetLogicalPinNumber(Int32 pin number)
at System.Device.Gpio.GpioController.OpenPinCore(Int32 pin number)
at System.Device.Gpio.GpioController.OpenPin(Int32 pinNumber)
at System.Device.Gpio.GpioController.OpenPin(Int32 pinNumber, PinMode mode)

**Versions used**

bot dotnet SDK 7.0.100
on debian linux arm (armbian)

Ellerbach, 2023-09-05:
Hi, please use the appropriate OrangePi driver: https://github.com/dotnet/iot/tree/main/src/devices/Gpio/Drivers

And here on how to use those: https://github.com/dotnet/iot/tree/main/src/devices/Gpio

Please confirm that it works properly so we can close this issue. Thanks!

blackmoonstudio, 2023-09-09:
thank you very much yeah it worked for me

raffaeler, 2023-09-10:
> thank you very much yeah it worked for me

Can we close the issue then?

blackmoonstudio, 2023-09-11:
yeah and can you help me with how I can send data with the i2c protocol?

raffaeler, 2023-09-11:
> yeah and can you help me with how I can send data with the i2c protocol?

I am currenty OOF, but the best place to ask this kind of questions is our [discord channel](https://discord.gg/kqWhBbj).

In any case, if you look at other "bindings" in this repo, you will find plenty of devices using I2C. I strongly suggest to look at those examples and just change the bytes you send to the device. This should be a good (and easy) starting point.

blackmoonstudio, 2023-09-12:
thank you very much

closed 2023-09-12 by blackmoonstudio